evcc-io/evcc #10500: Porsche acoount blocked

MarkusGH, 2023-10-26:
### Describe the bug

Heute Nachricht bekommen, dass mein Porsche Account geblockt ist. Log anbei.

### Steps to reproduce

Kam aus heiterem Himmel. Fahrzeug angesteckt und PV laden aktiv.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: localhost
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 3s # control cycle interval

log: info
levels:
  porsche: trace
  lp-1: debug
  lp-2: debug
  openwb: trace

mqtt:
  broker: mqtt.xxx.de:1883
  topic: evcc

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.1570 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.0816 # [currency]/kWh

meters:
- name: grid1
  type: movingaverage
  decay: 1 # 0.5
  meter:
    type: custom
    power:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Power
      scale: -1  # Die Leistung wird in Watt erwartet. Falls z.b. kW geliefert wird, kann dies hier mit dem Wert 1000 angepasst werden
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    energy:
      source: mqtt
      topic: mbmd/solaredge1-1-1/Import
      timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
    currents:
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L1
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L2
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
      - source: mqtt
        topic: mbmd/solaredge1-1-1/Current/L3
        timeout: 30s  # Akzeptiere keine Daten die älter als dieser Wert ist
[... 237 lines cut ...]
Oct 26 10:17:31 homebridge evcc[1606536]: --
Oct 26 10:17:31 homebridge evcc[1606536]: {"error":"invalid_grant","error_description":"Unknown or invalid refresh token."}
Oct 26 10:17:31 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:17:31 GET https://identity.porsche.com/authorize?audience=https%3A%2F%2Fapi.porsche.com&client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&code_challenge=TAeisJwAujo3GeXFZftbPOx9ySatj7uFwMJn6T6x8Qs&code_challenge_method=S256&redirect_uri=https%3A%2F%2Fmy.porsche.com%2F&response_type=code&scope=openid+offline_access&state=38CY7toMhfp2O8gL&ui_locales=de-DE
Oct 26 10:17:31 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:17:31 Found. Redirecting to /u/login/identifier?state=hKFo2SBTSWZBVG5tdU0zN1ZRVGk3cXdGd19qLXgxZHBGZHpVQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIFVHZVc0bmVreHZlZ0JWclZBTmJ0OGJiTXhydHl6N1Fzo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
Oct 26 10:17:31 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:17:31 POST https://identity.porsche.com/usernamepassword/login
Oct 26 10:17:31 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:17:31 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&connection=Username-Password-Authentication&password=***&sec=high&state=hKFo2SBTSWZBVG5tdU0zN1ZRVGk3cXdGd19qLXgxZHBGZHpVQaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIFVHZVc0bmVreHZlZ0JWclZBTmJ0OGJiTXhydHl6N1Fzo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&tenant=porsche-production&ui_locales=de&username=***
Oct 26 10:17:31 homebridge evcc[1606536]: --
Oct 26 10:17:31 homebridge evcc[1606536]: {"statusCode":429,"description":"Your account has been blocked after multiple consecutive login attempts. We've sent you an email with instructions on how to unblock it.","name":"AnomalyDetected","code":"too_many_attempts"}
Oct 26 10:22:54 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:22:54 POST https://identity.porsche.com/oauth/token
Oct 26 10:22:54 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:22:54 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&grant_type=refresh_token&refresh_token=v1.MfQ9VPysztRJ3AUi_mx-Ot125P6twROoeSF3WvyUhbobipXcCVUUL8IO3f3w393ML-auLZRy_jHVVccLj48eE0w
Oct 26 10:22:54 homebridge evcc[1606536]: --
Oct 26 10:22:54 homebridge evcc[1606536]: {"error":"invalid_grant","error_description":"Unknown or invalid refresh token."}
Oct 26 10:22:54 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:22:54 GET https://identity.porsche.com/authorize?audience=https%3A%2F%2Fapi.porsche.com&client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&code_challenge=4icAtK4kYLiXyrwP0GLCclrEI71UXdSyrcZCqKf2SVo&code_challenge_method=S256&redirect_uri=https%3A%2F%2Fmy.porsche.com%2F&response_type=code&scope=openid+offline_access&state=hCwktP1dOPEPDFlG&ui_locales=de-DE
Oct 26 10:22:54 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:22:54 Found. Redirecting to /u/login/identifier?state=hKFo2SBTdWtRWmlwYXJUM1Yza3NjNFZNUXhLbFpGUWw4X210UaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGZNOF9ROVYxN0lTZG9BTGRacFpMU3hHZmE1aG90WGVBo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&ui_locales=de
Oct 26 10:22:54 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:22:54 POST https://identity.porsche.com/usernamepassword/login
Oct 26 10:22:55 homebridge evcc[1606536]: [porsche] TRACE 2023/10/26 10:22:55 client_id=UYsK00My6bCqJdbQhTQ0PbWmcSdIAMig&connection=Username-Password-Authentication&password=***&sec=high&state=hKFo2SBTdWtRWmlwYXJUM1Yza3NjNFZNUXhLbFpGUWw4X210UaFur3VuaXZlcnNhbC1sb2dpbqN0aWTZIGZNOF9ROVYxN0lTZG9BTGRacFpMU3hHZmE1aG90WGVBo2NpZNkgVVlzSzAwTXk2YkNxSmRiUWhUUTBQYldtY1NkSUFNaWc&tenant=porsche-production&ui_locales=de&username=***
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.1 (090ca2c5)

RTTTC, 2023-10-26:
I know you love quick interval.. 🏎️ 🏁 
But this MUST be the problem here `interval: 3s`
Even Porsche has had enough 😄

MarkusGH, 2023-10-26:
Hahaha - very funny.
I would think that the frequency of login attempts is not linked with this interval.