evcc-io/evcc #10505: Climater error

MackKaufmann, 2023-10-26:
### Describe the bug

Multiple error notifications in UI:

Carport: climater: Get \"https://emea.bff.cariad.digital/vehicle/v1/vehicles/XXXXXXXXXXXXXXXX/selectivestatus?jobs=all\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

### Steps to reproduce

Error occurs when evcc is started in UI. Every 10 seconds (control cycle interval).

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

interval: 10s # control cycle interval

tariffs:
  grid:
    type: tibber
    token: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX" # access token
    homeid: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX" # optional if multiple homes associated to account

meters:
- type: template
  template: shelly-1pm
  usage: grid
  host: 192.168.XXX.111
  user: XXXXXXXXXX #
  password: XXXXXXXXXXXXXXXXXXXXXXX #
  name: grid1

chargers:
- type: template
  template: shelly
  host: 192.168.XXX.111
  user: XXXXXXXX #
  password: XXXXXXXXXXXXXXXXXXXXXXXX #
  channel: 0
  standbypower: 15
  name: wallbox3

vehicles:
- type: template
  template: audi
  title: Audi Q3
  user: XXXXX@XXXXXXXXXXXXX.XX
  password: XXXXXXXXXXXXXXXXXXXXX
  vin: XXXXXXXXXXXXXXXXXXXXXXX
  capacity: 11.5
  name: ev2

loadpoints:
- title: Carport
  charger: wallbox3
  vehicle: ev2
  mode: pv
  phases: 1
  mincurrent: 6
```

### Log details

```text
[lp-1  ] DEBUG 2023/10/26 18:42:16 charger status: B
[lp-1  ] DEBUG 2023/10/26 18:42:16 next soc poll remaining time: 29m59s
[lp-1  ] DEBUG 2023/10/26 18:42:16 pv charge current: 0A = 0A + -0A (0W @ 1p)
[lp-1  ] ERROR 2023/10/26 18:42:16 climater: Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/WAUZZZF38M1136229/selectivestatus?jobs=all": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[site  ] DEBUG 2023/10/26 18:42:25 ----
[lp-1  ] DEBUG 2023/10/26 18:42:25 charge power: 0W
[site  ] DEBUG 2023/10/26 18:42:25 grid power: 0W
[site  ] DEBUG 2023/10/26 18:42:25 pv power: 0W
[site  ] DEBUG 2023/10/26 18:42:25 site power: 0W
[lp-1  ] DEBUG 2023/10/26 18:42:25 charge total import: 190.349kWh
[lp-1  ] DEBUG 2023/10/26 18:42:25 charger status: B
[lp-1  ] DEBUG 2023/10/26 18:42:25 next soc poll remaining time: 29m50s
[lp-1  ] DEBUG 2023/10/26 18:42:25 pv charge current: 0A = 0A + -0A (0W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.2

andig, 2023-10-26:
Server doesn't answer- nothing we can do.

closed 2023-10-26 by andig

MackKaufmann, 2023-10-26:
Yes, I saw that too. However, I don't recall this problem ever occurring before I updated to 0.121.2. Since then it has been persistent. Before I believe it's a coincidence, I'd better ask the professionals.