evcc-io/evcc #10513: MQTT savings werden nicht mehr übertragen

zwiebelxxl, 2023-10-27:
### Describe the bug

In der Version 0.121.2 (vielleicht auch schon früher) vermisse ich die savings* Topics wie
savingsTotalCharged
savingsGridCharged
savingsSelfConsumptionCharged
savingsSelfConsumptionPercent
savingsEffectivePrice
savingsAmount

Ich weiß das an der Berechnung der savings sich etwas verändert hat. Aber es wäre schön wenn die absoluten Werte weiterhin auch über MQTT zu bekommen sind.

### Steps to reproduce

Wenn ich mit dem MQTT Explorer auf die Topics des evcc schaue fehlen die savings*.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token: *****
sponsortoken: *****

# log settings
#log: error
log: debug
#log: trace
levels:
  core: error
  lp-1: error
  lp-2: error
  #core: trace
  #lp-1: trace
  #lp-2: trace

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- name: pv
  #type: custom
  #type: default
  type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.10.11 # IP-Adresse oder Hostname
    #power:
    #source: http
    #type: http
    #uri: http://192.168.10.11/solar_api/v1/GetPowerFlowRealtimeData.fcgi
    #jq: if .Body.Data.Site.P_PV == null then 0 else .Body.Data.Site.P_PV end
- name: pv2
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.10.126 # IP-Adresse oder Hostname

[... 245 lines cut ...]
  #   - # list of recipient ids
  # - type: telegram
  #   token: *****
  #   chats: *****
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user: *****

</details>
Version: `0.121.2`
```

### Log details

```text
evcc startet normal und logs bringen hier denk ich nichts.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.2

andig, 2023-10-27:
Duplikat zu https://github.com/evcc-io/evcc/issues/10515

closed 2023-10-27 by andig

emcepe, 2023-10-27:
das ist ja witzig. Bei mir sehe ich via MQTT die savings.* :)

zwiebelxxl, 2023-10-27:
hier geht es darum das die Werte garnicht mit MQTT übertragen werden.

Bei dem issue #10515 geht es um Formatierung von Werten.