evcc-io/evcc #10515: evcc >= v1.121.0: falsch formatierte Werte via MQTT

emcepe, 2023-10-27:
### Describe the bug

evcc übermittelt via MQTT für site/log das hier:

```
{vehicle soc: unexpected status: 509 () error 1}
```

und für site/stats das hier:

```
map[30d:map[avgCo2:0 avgPrice:0.1087020843790553 chargedKWh:27.044999999999163 solarPercentage:79.76066406008634] 365d:map[avgCo2:5.895004296353555 avgPrice:0.09052521102331224 chargedKWh:64.18799999999919 solarPercentage:90.58023153374278] total:map[avgCo2:5.895004296353555 avgPrice:0.09052521102331224 chargedKWh:64.18799999999919 solarPercentage:90.58023153374278]]
```

das ist jeweils kein valides JSON.

Entweder als korrektes JSON übermitteln oder ich wäre für einen Tipp dankbar, wie man das z.B. via FHEM sinnvoll parsen sollte. json2nameValue fällt mit dem Output natürlich auf die Nase.

Danke.

### Steps to reproduce

n/a

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

0.121.2

andig, 2023-10-27:
@naltatis site/log muss raus (mache ich). Was machen wir mit den Stats? Die wiederrum müssten ja Einzelwerte sein? Wie würden wir das für InfluxDB machen?

emcepe, 2023-10-27:
im Prinzip finde ich site/log überhaupt nicht verkehrt, wäre halt nur toll, wenn es richtiges JSON wäre ;-)

site/stats als JSON wäre auch super, in etwa so:

```
{"30d":{"avgCo2":0,"avgPrice":0.1087020843790553,"chargedKWh":27.044999999999163,"solarPercentage":79.76066406008634},"365d":{"avgCo2":5.895004296353555,"avgPrice":0.09052521102331224,"chargedKWh":64.18799999999919,"solarPercentage":90.58023153374278},"total":{"avgCo2":5.895004296353555,"avgPrice":0.09052521102331224,"chargedKWh":64.18799999999919,"solarPercentage":90.58023153374278}}
```

und in human readable:

```
{
  "30d": {
    "avgCo2": 0,
    "avgPrice": 0.1087020843790553,
    "chargedKWh": 27.044999999999163,
    "solarPercentage": 79.76066406008634
  },
  "365d": {
    "avgCo2": 5.895004296353555,
    "avgPrice": 0.09052521102331224,
    "chargedKWh": 64.18799999999919,
    "solarPercentage": 90.58023153374278
  },
  "total": {
    "avgCo2": 5.895004296353555,
    "avgPrice": 0.09052521102331224,
    "chargedKWh": 64.18799999999919,
    "solarPercentage": 90.58023153374278
  }
}
```

das wäre portabel und man kann damit eigentlich überall und mit allem etwas anfangen was halbwegs aus der Neuzeit kommt ;-)

naltatis, 2023-10-28:
@andig Bei mqtt haben wir ja heute bereits mit `vehicle` und `pv` komplexere Unterstrukturen in `site`. Für `vehicle` gibts ne Sonderbehandlung. Schön wäre, wenn wir hier ne generische Lösung hätten, die Structs/Slices in MQTT Pfade verwandelt.

https://github.com/evcc-io/evcc/blob/d92d3875017b30f0c8a9453d2564285ee5f99978/server/mqtt.go#L84-L115

andig, 2023-10-28:
Das sollte mit ein bisschen Reflect-Magie gehen, bitte mal ausprobieren.

closed 2023-10-28 by andig

emcepe, 2023-10-28:
Bin gespannt. Werde ich beim nächsten Release testen.

Danke!

emcepe, 2023-10-28:
0.121.3 crashed bei mir permanent:

```
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: [lp-1  ] INFO 2023/10/28 18:12:09 vehicle updated: unknown -> Hyundai KONA
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: panic: reflect: call of reflect.Value.MapRange on ptr Value
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: goroutine 154 [running]:
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: reflect.flag.mustBe(...)
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: #011reflect/value.go:233
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: reflect.flag.panicNotMap(0x16)
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: #011reflect/value.go:1989 +0x88
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: reflect.Value.MapRange(...)
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: #011reflect/value.go:1978
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x543f9c0, {0x5822450, 0xf}, 0x1, {0x1f1ebd0, 0x5e94b20})
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: #011github.com/evcc-io/evcc/server/mqtt.go:85 +0xc8
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: github.com/evcc-io/evcc/server.(*MQTT).publish(0x543f9c0, {0x5822450, 0xf}, 0x1, {0x1f1ebd0, 0x5e94b20})
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: #011github.com/evcc-io/evcc/server/mqtt.go:123 +0x8c
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: github.com/evcc-io/evcc/server.(*MQTT).Run(0x543f9c0, {0x2c14c48, 0x5165040}, 0x5725340)
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: #011github.com/evcc-io/evcc/server/mqtt.go:306 +0xdcc
Oct 28 18:12:09 raspberry-pi-4b evcc[26235]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Oct 28 18:12:10 raspberry-pi-4b evcc[26235]: #011github.com/evcc-io/evcc/cmd/root.go:202 +0xf44
...
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [site  ] INFO 2023/10/28 18:12:21 site config:
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [site  ] INFO 2023/10/28 18:12:21   meters:      grid ✓ pv ✓ battery ✓
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [site  ] INFO 2023/10/28 18:12:21     grid:      power ✓ energy ✗ currents ✗
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [site  ] INFO 2023/10/28 18:12:21     pv 1:      power ✓ energy ✗ currents ✗
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [site  ] INFO 2023/10/28 18:12:21     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [site  ] INFO 2023/10/28 18:12:21   vehicles:
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [site  ] INFO 2023/10/28 18:12:21     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [lp-1  ] INFO 2023/10/28 18:12:21 loadpoint 1:
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [lp-1  ] INFO 2023/10/28 18:12:21   mode:        off
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [lp-1  ] INFO 2023/10/28 18:12:21   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✓
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [lp-1  ] INFO 2023/10/28 18:12:21   meters:      charge ✓
Oct 28 18:12:21 raspberry-pi-4b evcc[26852]: [lp-1  ] INFO 2023/10/28 18:12:21     charge:    power ✓ energy ✓ currents ✓
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: [lp-1  ] INFO 2023/10/28 18:12:22 car connected
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: [lp-1  ] INFO 2023/10/28 18:12:22 vehicle updated: unknown -> Hyundai KONA
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: panic: reflect: call of reflect.Value.MapRange on ptr Value
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: goroutine 166 [running]:
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: reflect.flag.mustBe(...)
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: #011reflect/value.go:233
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: reflect.flag.panicNotMap(0x16)
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: #011reflect/value.go:1989 +0x88
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: reflect.Value.MapRange(...)
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: #011reflect/value.go:1978
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x67ae910, {0x6cb5150, 0xf}, 0x1, {0x1f1ebd0, 0x6d3dd20})
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: #011github.com/evcc-io/evcc/server/mqtt.go:85 +0xc8
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: github.com/evcc-io/evcc/server.(*MQTT).publish(0x67ae910, {0x6cb5150, 0xf}, 0x1, {0x1f1ebd0, 0x6d3dd20})
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: #011github.com/evcc-io/evcc/server/mqtt.go:123 +0x8c
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: github.com/evcc-io/evcc/server.(*MQTT).Run(0x67ae910, {0x2c14c48, 0x679ca00}, 0x6e03280)
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: #011github.com/evcc-io/evcc/server/mqtt.go:306 +0xdcc
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Oct 28 18:12:22 raspberry-pi-4b evcc[26852]: #011github.com/evcc-io/evcc/cmd/root.go:202 +0xf44
...
```

andig, 2023-10-28:
Was für ein blöder Fehler. Ich publishe es nochmal- DANKE!

emcepe, 2023-10-28:
ok, nun siehts besser aus.

hmm ... FHEM fügt das hier in die readinglist hinzu:

```
evcc/site/stats/30d/solarPercentage:.* solarPercentage
evcc/site/stats/30d/chargedKWh:.* chargedKWh
evcc/site/stats/30d/avgPrice:.* avgPrice
evcc/site/stats/30d/avgCo2:.* avgCo2
evcc/site/stats/365d/solarPercentage:.* solarPercentage
evcc/site/stats/365d/chargedKWh:.* chargedKWh
evcc/site/stats/365d/avgPrice:.* avgPrice
evcc/site/stats/365d/avgCo2:.* avgCo2
evcc/site/stats/total/solarPercentage:.* solarPercentage
evcc/site/stats/total/chargedKWh:.* chargedKWh
evcc/site/stats/total/avgPrice:.* avgPrice
evcc/site/stats/total/avgCo2:.* avgCo2
```

ich hätte es als reines JSON wie vorgeschlagen sinniger & schöner gefunden, json2nameValue hätte dann automagisch das daraus gemacht:
```
evcc_stats_30d_avgCo2
evcc_stats_30d_avgPrice
evcc_stats_30d_chargedKWh
evcc_stats_30d_solarPercentage
evcc_stats_365d_avgCo2
evcc_stats_365d_avgPrice
evcc_stats_365d_chargedKWh
evcc_stats_365d_solarPercentage
evcc_stats_total_avgCo2
evcc_stats_total_avgPrice
evcc_stats_total_chargedKWh
evcc_stats_total_solarPercentage
```
mit ```json2nameValue($EVENT, 'evcc_stats_', $JSONMAP)``` - 1 Eintrag, große Wirkung.

nun gibt's pro Wert einen Readinglisteintrag, den man dann wieder manuell abändern muss damit die Werte sich nicht alle gegenseitig überschreiben.

Whatever, es funktioniert :)

andig, 2023-10-28:
Yep. nicht jeder hat `json2nameValue` und genau dafür gibts mqtt Topics ja eigentlich- sonst kannst Du auch gleich /api/state verwenden.

docolli, 2023-10-28:
Bin eben auch auf 0121.3. Leider steigt auch bei mir evcc noch regelmäßig aus. Etwas anderere Fehlermeldung als bei @emcepe 

```
Okt 28 19:27:16 fhem-heizung systemd[1]: Started evcc.
Okt 28 19:27:17 fhem-heizung evcc[6459]: [main  ] INFO 2023/10/28 19:27:17 evcc 0.121.3
Okt 28 19:27:17 fhem-heizung evcc[6459]: [main  ] INFO 2023/10/28 19:27:17 using config file: /etc/evcc.yaml
Okt 28 19:27:17 fhem-heizung evcc[6459]: [main  ] INFO 2023/10/28 19:27:17 starting ui and api at :7070
Okt 28 19:27:18 fhem-heizung evcc[6459]: [db    ] INFO 2023/10/28 19:27:18 using sqlite database: /var/lib/evcc/evcc.db
Okt 28 19:27:18 fhem-heizung evcc[6459]: [mqtt  ] INFO 2023/10/28 19:27:18 connecting 653555047 at tcp://localhost:1883
Okt 28 19:27:26 fhem-heizung evcc[6459]: [lp-1  ] WARN 2023/10/28 19:27:26 poll mode '{always 12h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] INFO 2023/10/28 19:27:27 site config:
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] INFO 2023/10/28 19:27:27   meters:      grid ✓ pv ✓ battery ✓
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] INFO 2023/10/28 19:27:27     grid:      power ✓ energy ✗ currents ✗
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] INFO 2023/10/28 19:27:27     pv 1:      power ✓ energy ✗ currents ✗
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] INFO 2023/10/28 19:27:27     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] INFO 2023/10/28 19:27:27   vehicles:
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] INFO 2023/10/28 19:27:27     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
Okt 28 19:27:27 fhem-heizung evcc[6459]: [lp-1  ] INFO 2023/10/28 19:27:27 loadpoint 1:
Okt 28 19:27:27 fhem-heizung evcc[6459]: [lp-1  ] INFO 2023/10/28 19:27:27   mode:        minpv
Okt 28 19:27:27 fhem-heizung evcc[6459]: [lp-1  ] INFO 2023/10/28 19:27:27   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Okt 28 19:27:27 fhem-heizung evcc[6459]: [lp-1  ] INFO 2023/10/28 19:27:27   meters:      charge ✓
Okt 28 19:27:27 fhem-heizung evcc[6459]: [lp-1  ] INFO 2023/10/28 19:27:27     charge:    power ✓ energy ✓ currents ✓
Okt 28 19:27:27 fhem-heizung evcc[6459]: [lp-1  ] INFO 2023/10/28 19:27:27 vehicle updated: unknown -> Nissan Leaf
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] WARN 2023/10/28 19:27:27 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Okt 28 19:27:27 fhem-heizung evcc[6459]: [site  ] ERROR 2023/10/28 19:27:27 tariff: no matching rate
Okt 28 19:27:27 fhem-heizung evcc[6459]: [lp-1  ] INFO 2023/10/28 19:27:27 car disconnected
Okt 28 19:27:27 fhem-heizung evcc[6459]: panic: reflect.Value.Interface: cannot return value obtained from unexported field or method
Okt 28 19:27:27 fhem-heizung evcc[6459]: goroutine 89 [running]:
Okt 28 19:27:27 fhem-heizung evcc[6459]: reflect.valueInterface({0x1ee64b0, 0x5ca17e8, 0xab}, 0x1)
Okt 28 19:27:27 fhem-heizung evcc[6459]:        reflect/value.go:1501 +0x11c
Okt 28 19:27:27 fhem-heizung evcc[6459]: reflect.Value.Interface(...)
Okt 28 19:27:27 fhem-heizung evcc[6459]:        reflect/value.go:1490
Okt 28 19:27:27 fhem-heizung evcc[6459]: github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x60660e0, {0x63cb080, 0x1c}, 0x1, {0x222ab30, 0x5ca17e8})
Okt 28 19:27:27 fhem-heizung evcc[6459]:        github.com/evcc-io/evcc/server/mqtt.go:97 +0x440
Okt 28 19:27:27 fhem-heizung evcc[6459]: github.com/evcc-io/evcc/server.(*MQTT).publish(0x60660e0, {0x63cb080, 0x1c}, 0x1, {0x222ab30, 0x5ca17e8})
Okt 28 19:27:27 fhem-heizung evcc[6459]:        github.com/evcc-io/evcc/server/mqtt.go:123 +0x8c
Okt 28 19:27:27 fhem-heizung evcc[6459]: github.com/evcc-io/evcc/server.(*MQTT).Run(0x60660e0, {0x2c14bf0, 0x54e2dc0}, 0x63cc100)
Okt 28 19:27:27 fhem-heizung evcc[6459]:        github.com/evcc-io/evcc/server/mqtt.go:306 +0xdcc
Okt 28 19:27:27 fhem-heizung evcc[6459]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Okt 28 19:27:27 fhem-heizung evcc[6459]:        github.com/evcc-io/evcc/cmd/root.go:202 +0xf44
Okt 28 19:27:27 fhem-heizung systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Okt 28 19:27:27 fhem-heizung systemd[1]: evcc.service: Failed with result 'exit-code'.
Okt 28 19:27:37 fhem-heizung systemd[1]: evcc.service: Service RestartSec=10s expired, scheduling restart.
Okt 28 19:27:37 fhem-heizung systemd[1]: evcc.service: Scheduled restart job, restart counter is at 41.
Okt 28 19:27:37 fhem-heizung systemd[1]: Stopped evcc.

```

emcepe, 2023-10-28:
Du musst die erneut downloaden und installieren.
Die Version ist leider gleich geblieben weshalb man auf den ersten Blick kein Update sieht.

docolli, 2023-10-28:
Ich hole mir evcc sonst aus dem APT Repository. Kann gut sein, dass durch die gleiche Version das Repository (noch) nicht aktualisiert wurde. Ich schau morgen vormittag nochmals, sonst hole ich mir das Paket manuell.

emcepe, 2023-10-28:
apt wird keinen Anlass dazu haben das neu zu installieren, weil wie gesagt die Version gleich geblieben ist.

Genau wegen sowas wie hier macht man sowas eigentlich nicht ;)

```
apt-get clean
apt-get --reinstall install evcc
```

sollte helfen. ansonsten per wget/curl etc. holen und per dpkg installieren.

docolli, 2023-10-28:
Ich glaube bei mir ist was anderes das Problem.
Habe das Paket jetzt geholt mit `wget https://github.com/evcc-io/evcc/releases/download/0.121.3/evcc_0.121.3_armhf.deb`
Dann
```
root@fhem:/home/pi# dpkg -i evcc_0.121.3_armhf.deb
(Lese Datenbank ... 49662 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von evcc_0.121.3_armhf.deb ...
Entpacken von evcc (0.121.3) über (0.121.3) ...
evcc (0.121.3) wird eingerichtet ...
root@fhem:/home/pi# sudo journalctl -fau evcc
-- Logs begin at Sun 2023-10-01 15:28:30 CEST. --
Okt 28 20:01:24 fhem-heizung evcc[7796]: [db    ] INFO 2023/10/28 20:01:24 using sqlite database: /var/lib/evcc/evcc.db
Okt 28 20:01:24 fhem-heizung evcc[7796]: [mqtt  ] INFO 2023/10/28 20:01:24 connecting 653555047 at tcp://localhost:1883
Okt 28 20:01:30 fhem-heizung systemd[1]: Stopping evcc...
Okt 28 20:01:30 fhem-heizung systemd[1]: evcc.service: Main process exited, code=killed, status=15/TERM
Okt 28 20:01:30 fhem-heizung systemd[1]: evcc.service: Succeeded.
Okt 28 20:01:30 fhem-heizung systemd[1]: Stopped evcc.
Okt 28 20:01:30 fhem-heizung systemd[1]: Started evcc.
Okt 28 20:01:32 fhem-heizung evcc[7839]: [main  ] INFO 2023/10/28 20:01:32 evcc 0.121.3
Okt 28 20:01:32 fhem-heizung evcc[7839]: [main  ] INFO 2023/10/28 20:01:32 using config file: /etc/evcc.yaml
Okt 28 20:01:32 fhem-heizung evcc[7839]: [main  ] INFO 2023/10/28 20:01:32 starting ui and api at :7070
Okt 28 20:01:33 fhem-heizung evcc[7839]: [db    ] INFO 2023/10/28 20:01:33 using sqlite database: /var/lib/evcc/evcc.db
Okt 28 20:01:33 fhem-heizung evcc[7839]: [mqtt  ] INFO 2023/10/28 20:01:33 connecting 653555047 at tcp://localhost:1883
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] WARN 2023/10/28 20:01:43 poll mode '{always 12h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] INFO 2023/10/28 20:01:43 site config:
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] INFO 2023/10/28 20:01:43   meters:      grid ✓ pv ✓ battery ✓
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] INFO 2023/10/28 20:01:43     grid:      power ✓ energy ✗ currents ✗
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] INFO 2023/10/28 20:01:43     pv 1:      power ✓ energy ✗ currents ✗
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] INFO 2023/10/28 20:01:43     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] INFO 2023/10/28 20:01:43   vehicles:
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] INFO 2023/10/28 20:01:43     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] INFO 2023/10/28 20:01:43 loadpoint 1:
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] INFO 2023/10/28 20:01:43   mode:        minpv
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] INFO 2023/10/28 20:01:43   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] INFO 2023/10/28 20:01:43   meters:      charge ✓
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] INFO 2023/10/28 20:01:43     charge:    power ✓ energy ✓ currents ✓
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] INFO 2023/10/28 20:01:43 vehicle updated: unknown -> Nissan Leaf
Okt 28 20:01:43 fhem-heizung evcc[7839]: [site  ] WARN 2023/10/28 20:01:43 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Okt 28 20:01:43 fhem-heizung evcc[7839]: [lp-1  ] INFO 2023/10/28 20:01:43 car disconnected
Okt 28 20:01:44 fhem-heizung evcc[7839]: panic: reflect.Value.Interface: cannot return value obtained from unexported field or method
Okt 28 20:01:44 fhem-heizung evcc[7839]: goroutine 101 [running]:
Okt 28 20:01:44 fhem-heizung evcc[7839]: reflect.valueInterface({0x1ee64b0, 0x59f2b88, 0xab}, 0x1)
Okt 28 20:01:44 fhem-heizung evcc[7839]:        reflect/value.go:1501 +0x11c
Okt 28 20:01:44 fhem-heizung evcc[7839]: reflect.Value.Interface(...)
Okt 28 20:01:44 fhem-heizung evcc[7839]:        reflect/value.go:1490
Okt 28 20:01:44 fhem-heizung evcc[7839]: github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x5be20f0, {0x60a6ca0, 0x1c}, 0x1, {0x222ab30, 0x59f2b88})
Okt 28 20:01:44 fhem-heizung evcc[7839]:        github.com/evcc-io/evcc/server/mqtt.go:97 +0x440
Okt 28 20:01:44 fhem-heizung evcc[7839]: github.com/evcc-io/evcc/server.(*MQTT).publish(0x5be20f0, {0x60a6ca0, 0x1c}, 0x1, {0x222ab30, 0x59f2b88})
Okt 28 20:01:44 fhem-heizung evcc[7839]:        github.com/evcc-io/evcc/server/mqtt.go:123 +0x8c
Okt 28 20:01:44 fhem-heizung evcc[7839]: github.com/evcc-io/evcc/server.(*MQTT).Run(0x5be20f0, {0x2c14bf0, 0x5b64b40}, 0x605a100)
Okt 28 20:01:44 fhem-heizung evcc[7839]:        github.com/evcc-io/evcc/server/mqtt.go:306 +0xdcc
Okt 28 20:01:44 fhem-heizung evcc[7839]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Okt 28 20:01:44 fhem-heizung evcc[7839]:        github.com/evcc-io/evcc/cmd/root.go:202 +0xf44
Okt 28 20:01:44 fhem-heizung systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Okt 28 20:01:44 fhem-heizung systemd[1]: evcc.service: Failed with result 'exit-code'.

```

andig, 2023-10-28:
Danke für den Hinweis! Hoffnung: es war nur ein 20min Fenster…

andig, 2023-10-28:
Siehe 1 Kommentar deüber wie das zu lösen ist.

docolli, 2023-10-28:
Hmm, habe das vor dem manuellen Download auch bereits versucht. Leider ohne Erfolg!

```
root@fhem:/home/pi# apt-get clean
root@fhem:/home/pi# apt-get --reinstall install evcc
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
0 aktualisiert, 0 neu installiert, 1 erneut installiert, 0 zu entfernen und 79 nicht aktualisiert.
Es müssen 20,9 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 0 B Plattenplatz zusätzlich benutzt.
Holen:1 https://dl.cloudsmith.io/public/evcc/stable/deb/raspbian buster/main armhf evcc armhf 0.121.3 [20,9 MB]
Es wurden 20,9 MB in 6 s geholt (3.611 kB/s).
(Lese Datenbank ... 49662 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../evcc_0.121.3_armhf.deb ...
Entpacken von evcc (0.121.3) über (0.121.3) ...
evcc (0.121.3) wird eingerichtet ...
root@fhem:/home/pi# sudo journalctl -fau evcc
-- Logs begin at Sun 2023-10-01 15:28:30 CEST. --
Okt 28 20:00:26 fhem-heizung systemd[1]: Stopping evcc...
Okt 28 20:00:26 fhem-heizung systemd[1]: evcc.service: Main process exited, code=killed, status=15/TERM
Okt 28 20:00:26 fhem-heizung systemd[1]: evcc.service: Succeeded.
Okt 28 20:00:26 fhem-heizung systemd[1]: Stopped evcc.
Okt 28 20:00:40 fhem-heizung systemd[1]: Started evcc.
Okt 28 20:00:41 fhem-heizung evcc[7732]: [main  ] INFO 2023/10/28 20:00:41 evcc 0.121.3
Okt 28 20:00:41 fhem-heizung evcc[7732]: [main  ] INFO 2023/10/28 20:00:41 using config file: /etc/evcc.yaml
Okt 28 20:00:41 fhem-heizung evcc[7732]: [main  ] INFO 2023/10/28 20:00:41 starting ui and api at :7070
Okt 28 20:00:42 fhem-heizung evcc[7732]: [db    ] INFO 2023/10/28 20:00:42 using sqlite database: /var/lib/evcc/evcc.db
Okt 28 20:00:43 fhem-heizung evcc[7732]: [mqtt  ] INFO 2023/10/28 20:00:43 connecting 653555047 at tcp://localhost:1883
Okt 28 20:00:50 fhem-heizung evcc[7732]: [lp-1  ] WARN 2023/10/28 20:00:50 poll mode '{always 12h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] INFO 2023/10/28 20:00:51 site config:
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] INFO 2023/10/28 20:00:51   meters:      grid ✓ pv ✓ battery ✓
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] INFO 2023/10/28 20:00:51     grid:      power ✓ energy ✗ currents ✗
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] INFO 2023/10/28 20:00:51     pv 1:      power ✓ energy ✗ currents ✗
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] INFO 2023/10/28 20:00:51     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] INFO 2023/10/28 20:00:51   vehicles:
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] INFO 2023/10/28 20:00:51     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
Okt 28 20:00:51 fhem-heizung evcc[7732]: [lp-1  ] INFO 2023/10/28 20:00:51 loadpoint 1:
Okt 28 20:00:51 fhem-heizung evcc[7732]: [lp-1  ] INFO 2023/10/28 20:00:51   mode:        minpv
Okt 28 20:00:51 fhem-heizung evcc[7732]: [lp-1  ] INFO 2023/10/28 20:00:51   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Okt 28 20:00:51 fhem-heizung evcc[7732]: [lp-1  ] INFO 2023/10/28 20:00:51   meters:      charge ✓
Okt 28 20:00:51 fhem-heizung evcc[7732]: [lp-1  ] INFO 2023/10/28 20:00:51     charge:    power ✓ energy ✓ currents ✓
Okt 28 20:00:51 fhem-heizung evcc[7732]: [lp-1  ] INFO 2023/10/28 20:00:51 vehicle updated: unknown -> Nissan Leaf
Okt 28 20:00:51 fhem-heizung evcc[7732]: [site  ] WARN 2023/10/28 20:00:51 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Okt 28 20:00:51 fhem-heizung evcc[7732]: [lp-1  ] INFO 2023/10/28 20:00:51 car disconnected
Okt 28 20:00:51 fhem-heizung evcc[7732]: panic: reflect.Value.Interface: cannot return value obtained from unexported field or method
Okt 28 20:00:51 fhem-heizung evcc[7732]: goroutine 96 [running]:
Okt 28 20:00:51 fhem-heizung evcc[7732]: reflect.valueInterface({0x1ee64b0, 0x64164b0, 0xab}, 0x1)
Okt 28 20:00:51 fhem-heizung evcc[7732]:        reflect/value.go:1501 +0x11c
Okt 28 20:00:51 fhem-heizung evcc[7732]: reflect.Value.Interface(...)
Okt 28 20:00:51 fhem-heizung evcc[7732]:        reflect/value.go:1490
Okt 28 20:00:51 fhem-heizung evcc[7732]: github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x565b0f0, {0x5cca3e0, 0x1c}, 0x1, {0x222ab30, 0x64164b0})
Okt 28 20:00:51 fhem-heizung evcc[7732]:        github.com/evcc-io/evcc/server/mqtt.go:97 +0x440
Okt 28 20:00:51 fhem-heizung evcc[7732]: github.com/evcc-io/evcc/server.(*MQTT).publish(0x565b0f0, {0x5cca3e0, 0x1c}, 0x1, {0x222ab30, 0x64164b0})
Okt 28 20:00:51 fhem-heizung evcc[7732]:        github.com/evcc-io/evcc/server/mqtt.go:123 +0x8c
Okt 28 20:00:51 fhem-heizung evcc[7732]: github.com/evcc-io/evcc/server.(*MQTT).Run(0x565b0f0, {0x2c14bf0, 0x6180500}, 0x5fab4c0)
Okt 28 20:00:51 fhem-heizung evcc[7732]:        github.com/evcc-io/evcc/server/mqtt.go:306 +0xdcc
Okt 28 20:00:51 fhem-heizung evcc[7732]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Okt 28 20:00:51 fhem-heizung evcc[7732]:        github.com/evcc-io/evcc/cmd/root.go:202 +0xf44
Okt 28 20:00:51 fhem-heizung systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Okt 28 20:00:51 fhem-heizung systemd[1]: evcc.service: Failed with result 'exit-code'.
```

docolli, 2023-10-28:
root@fhem:/home/pi# sha1sum evcc_0.121.3_armhf.deb
aec1971446c84e52dd4aada8844cbe4e3253baf8  evcc_0.121.3_armhf.deb

flushbug, 2023-10-28:
Verdammt, genau rein gerannt. Hat jemand ne Idee wie ich dem Home Assistant beibringe die Version nochmal zu installieren ? 😞

docolli, 2023-10-28:
Wird wohl besser auf ein Release von 0.121.4 rauslaufen. 😉

emcepe, 2023-10-28:
> ```
> root@fhem:/home/pi# sha1sum evcc_0.121.3_armhf.deb
> aec1971446c84e52dd4aada8844cbe4e3253baf8  evcc_0.121.3_armhf.deb
> ```
> 
> Und hier die Werte für die evcc binary:
> 
> ```
> root@fhem:/# sha1sum /usr/bin/evcc
> 90351fb90ce4889a805076a16df66112407862c1  /usr/bin/evcc
> root@fhem:/# ls -l /usr/bin/evcc
> -rwxr-xr-x 1 root root 73990144 Okt 28 18:33 /usr/bin/evcc
> ```

ok, das ist schon die richtige Version bzw. die, welche bei mir läuft.

Evtl. hast Du noch einen Bug gefunden :)

docolli, 2023-10-28:
Bin jetzt wieder auf 0121.2 runter. Hat wegen mir keine Eile mit einer neuen Version. Helfe aber gerne den Bug zu finden.

Andrenato, 2023-10-28:
Habe im Home Assistent mit der neuen Version 0121.4 leider die gleichen Probleme wie oben in der Logfile

emcepe, 2023-10-28:
Hi @andig,

eines verstehe ich immer noch nicht, das Problem hatte ich schon des öfteren mitgeteilt.

laut Changelog von 0.121.4:

```Mqtt: remove deprecated savings```

laut Code:

```
var deprecatedTopics = []string{
	"activePhases", "range", "socCharge", "vehicleSoC",
	"batterySoC", "bufferSoC", "minSoC", "prioritySoC",
	"savingsAmount", "savingsEffectivePrice", "savingsGridCharged", "savingsSelfConsumptionCharged", "savingsSelfConsumptionPercent", "savingsTotalCharged",
	"targetSoC", "vehicleTargetSoC",
}
```

laut Changelog 0.121.3:

```Mqtt: remove log struct```

laut Code:

```
	ignoreEmpty = ""                                      // ignore empty keys
	ignoreLogs  = []string{"log"}                         // ignore log messages, including warn/error
	ignoreMqtt  = []string{"log", "auth", "releaseNotes"} // excessive size may crash certain brokers
```

und evcc diese Topics und Werte dennoch via MQTT sendet? Den Sinn verstehe ich nicht.
Hatte vorhin aus FHEM die savings.* und site/log und site/stats rausgenommen, Update auf 0.121.4 gemacht, siehe da, MQTT bekommt diese Topics wieder und erstellt sie mir natürlich dann auch wieder in FHEM:

```
evcc/site/log:.* { json2nameValue($EVENT, 'log_', $JSONMAP) }
evcc/site/stats:.* stats
evcc/loadpoints/1/savingsAmount:.* loadpoints_1_savingsAmount
evcc/loadpoints/1/savingsEffectivePrice:.* loadpoints_1_savingsEffectivePrice
evcc/loadpoints/1/savingsGridCharged:.* loadpoints_1_savingsGridCharged
evcc/loadpoints/1/savingsSelfConsumptionCharged:.* loadpoints_1_savingsSelfConsumptionCharged
evcc/loadpoints/1/savingsSelfConsumptionPercent:.* loadpoints_1_savingsSelfConsumptionPercent
evcc/loadpoints/1/savingsTotalCharged:.* loadpoints_1_savingsTotalCharged
```

diesmal jedoch nicht unter site/savings.* sondern unterhalb loadpoints/1 ?! :-)

Klar, ich kann die via {} deaktivieren, jedoch die Frage: welchen Sinn ergibt es, daß die Topics nach wie vor gesendet werden? :)

Das Gleiche habe ich mit den ganzen SoC Sachen, die nach wie vor alle noch da sind und auch erstellt werden, wenn ich sie nicht via {} deaktiviere:

```
$DEVICETOPIC/site/batterySoC:.* {}
$DEVICETOPIC/site/bufferSoC:.* {}
$DEVICETOPIC/site/minSoC:.* {}
$DEVICETOPIC/site/prioritySoC:.* {}
$DEVICETOPIC/site/targetSoC:.* {}
$DEVICETOPIC/site/vehicleSoC:.* {}
$DEVICETOPIC/site/vehicleTargetSoC:.* {}
$DEVICETOPIC/loadpoints/1/batterySoC:.* {}
$DEVICETOPIC/loadpoints/1/bufferSoC:.* {}
$DEVICETOPIC/loadpoints/1/minSoC:.* {}
$DEVICETOPIC/loadpoints/1/prioritySoC:.* {}
$DEVICETOPIC/loadpoints/1/targetSoC:.* {}
$DEVICETOPIC/loadpoints/1/vehicleSoC:.* {}
$DEVICETOPIC/loadpoints/1/vehicleTargetSoC:.* {}
```

habe das alles gerade nochmal rausgenommen, evcc restart und siehe da:

```
evcc/site/stats:.* stats
evcc/site/vehicleSoC:.* vehicleSoC
evcc/site/batterySoC:.* batterySoC
evcc/site/bufferSoC:.* bufferSoC
evcc/site/minSoC:.* minSoC
evcc/site/prioritySoC:.* prioritySoC
evcc/site/targetSoC:.* targetSoC
evcc/site/vehicleTargetSoC:.* vehicleTargetSoC
evcc/loadpoints/1/vehicleSoC:.* loadpoints_1_vehicleSoC
evcc/loadpoints/1/batterySoC:.* loadpoints_1_batterySoC
evcc/loadpoints/1/bufferSoC:.* loadpoints_1_bufferSoC
evcc/loadpoints/1/minSoC:.* loadpoints_1_minSoC
evcc/loadpoints/1/prioritySoC:.* loadpoints_1_prioritySoC
evcc/loadpoints/1/targetSoC:.* loadpoints_1_targetSoC
evcc/loadpoints/1/vehicleTargetSoC:.* loadpoints_1_vehicleTargetSoC
evcc/loadpoints/1/savingsAmount:.* loadpoints_1_savingsAmount
evcc/loadpoints/1/savingsEffectivePrice:.* loadpoints_1_savingsEffectivePrice
evcc/loadpoints/1/savingsGridCharged:.* loadpoints_1_savingsGridCharged
evcc/loadpoints/1/savingsSelfConsumptionCharged:.* loadpoints_1_savingsSelfConsumptionCharged
evcc/loadpoints/1/savingsSelfConsumptionPercent:.* loadpoints_1_savingsSelfConsumptionPercent
evcc/loadpoints/1/savingsTotalCharged:.* loadpoints_1_savingsTotalCharged
```

im MQTT Explorer sieht man das jedoch alles nicht mehr bzw. fast: site/log ist noch da und site/stats:.* ebenso. Oder braucht ihr dieses ominöse Format von stats mit map usw. für die InfluxDB?

emcepe, 2023-10-28:
ich vermisse bei den Stats via MQTT noch 2 Werte (so wie in der GUI zu sehen ist)

- eine Aufsplittung von chargedKWh in Sonne vs. Netz
- wieviel €uro gespart wurden

die Stats spucken bei avgCo2 einen Wert aus, in der GUI sehe ich aber bei 30 Tagen, 365 Tagen und Gesamt jeweils __
g/kWh

emcepe, 2023-10-28:
> ok, das ist schon die richtige Version bzw. die, welche bei mir läuft.
>
> Evtl. hast Du noch einen Bug gefunden :)

bei mir tauchen die Fehler auch noch auf und evcc schmiert ab, sporadisch:

```
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: panic: reflect.Value.Interface: cannot return value obtained from unexported field or method
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: goroutine 172 [running]:
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: reflect.valueInterface({0x1ee64b0, 0x58c73c8, 0xab}, 0x1)
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: #011reflect/value.go:1501 +0x11c
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: reflect.Value.Interface(...)
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: #011reflect/value.go:1490
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x52f7020, {0x58fba20, 0x1c}, 0x1, {0x222ab30, 0x58c73c8})
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: #011github.com/evcc-io/evcc/server/mqtt.go:97 +0x440
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: github.com/evcc-io/evcc/server.(*MQTT).publish(0x52f7020, {0x58fba20, 0x1c}, 0x1, {0x222ab30, 0x58c73c8})
Oct 29 01:03:51 raspberry-pi-4b evcc[27266]: #011github.com/evcc-io/evcc/server/mqtt.go:123 +0x8c
Oct 29 01:03:52 raspberry-pi-4b evcc[27266]: github.com/evcc-io/evcc/server.(*MQTT).Run(0x52f7020, {0x2c14c70, 0x4d64dc0}, 0x5625040)
Oct 29 01:03:52 raspberry-pi-4b evcc[27266]: #011github.com/evcc-io/evcc/server/mqtt.go:306 +0xdcc
Oct 29 01:03:52 raspberry-pi-4b evcc[27266]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
```

docolli, 2023-10-29:
Danke für die schnelle Veröffentlichung von 0121.4, aber bei mir stürzt 0121.4 nach den Start sofort wieder ab:

```
Okt 29 09:12:47 fhem-heizung systemd[1]: Started evcc.
Okt 29 09:12:48 fhem-heizung evcc[11404]: [main  ] INFO 2023/10/29 09:12:48 evcc 0.121.4
Okt 29 09:12:48 fhem-heizung evcc[11404]: [main  ] INFO 2023/10/29 09:12:48 using config file: /etc/evcc.yaml
Okt 29 09:12:48 fhem-heizung evcc[11404]: [main  ] INFO 2023/10/29 09:12:48 starting ui and api at :7070
Okt 29 09:12:49 fhem-heizung evcc[11404]: [db    ] INFO 2023/10/29 09:12:49 using sqlite database: /var/lib/evcc/evcc.db
Okt 29 09:12:49 fhem-heizung evcc[11404]: [mqtt  ] INFO 2023/10/29 09:12:49 connecting 653555047 at tcp://localhost:1883
Okt 29 09:12:57 fhem-heizung evcc[11404]: [lp-1  ] WARN 2023/10/29 09:12:57 poll mode '{always 12h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] INFO 2023/10/29 09:12:58 site config:
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] INFO 2023/10/29 09:12:58   meters:      grid ✓ pv ✓ battery ✓
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] INFO 2023/10/29 09:12:58     grid:      power ✓ energy ✗ currents ✗
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] INFO 2023/10/29 09:12:58     pv 1:      power ✓ energy ✗ currents ✗
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] INFO 2023/10/29 09:12:58     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] INFO 2023/10/29 09:12:58   vehicles:
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] INFO 2023/10/29 09:12:58     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✗
Okt 29 09:12:58 fhem-heizung evcc[11404]: [lp-1  ] INFO 2023/10/29 09:12:58 loadpoint 1:
Okt 29 09:12:58 fhem-heizung evcc[11404]: [lp-1  ] INFO 2023/10/29 09:12:58   mode:        minpv
Okt 29 09:12:58 fhem-heizung evcc[11404]: [lp-1  ] INFO 2023/10/29 09:12:58   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Okt 29 09:12:58 fhem-heizung evcc[11404]: [lp-1  ] INFO 2023/10/29 09:12:58   meters:      charge ✓
Okt 29 09:12:58 fhem-heizung evcc[11404]: [lp-1  ] INFO 2023/10/29 09:12:58     charge:    power ✓ energy ✓ currents ✓
Okt 29 09:12:58 fhem-heizung evcc[11404]: [lp-1  ] INFO 2023/10/29 09:12:58 vehicle updated: unknown -> Nissan Leaf
Okt 29 09:12:58 fhem-heizung evcc[11404]: [site  ] WARN 2023/10/29 09:12:58 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Okt 29 09:12:58 fhem-heizung evcc[11404]: [lp-1  ] INFO 2023/10/29 09:12:58 car disconnected
Okt 29 09:12:58 fhem-heizung evcc[11404]: panic: reflect.Value.Interface: cannot return value obtained from unexported field or method
Okt 29 09:12:58 fhem-heizung evcc[11404]: goroutine 90 [running]:
Okt 29 09:12:58 fhem-heizung evcc[11404]: reflect.valueInterface({0x1ee64b0, 0x5563c08, 0xab}, 0x1)
Okt 29 09:12:58 fhem-heizung evcc[11404]:       reflect/value.go:1501 +0x11c
Okt 29 09:12:58 fhem-heizung evcc[11404]: reflect.Value.Interface(...)
Okt 29 09:12:58 fhem-heizung evcc[11404]:       reflect/value.go:1490
Okt 29 09:12:58 fhem-heizung evcc[11404]: github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x5c29cf0, {0x5e559c0, 0x1c}, 0x1, {0x222ab30, 0x5563c08})
Okt 29 09:12:58 fhem-heizung evcc[11404]:       github.com/evcc-io/evcc/server/mqtt.go:97 +0x440
Okt 29 09:12:58 fhem-heizung evcc[11404]: github.com/evcc-io/evcc/server.(*MQTT).publish(0x5c29cf0, {0x5e559c0, 0x1c}, 0x1, {0x222ab30, 0x5563c08})
Okt 29 09:12:58 fhem-heizung evcc[11404]:       github.com/evcc-io/evcc/server/mqtt.go:123 +0x8c
Okt 29 09:12:58 fhem-heizung evcc[11404]: github.com/evcc-io/evcc/server.(*MQTT).Run(0x5c29cf0, {0x2c14c70, 0x5eb8500}, 0x5c2c240)
Okt 29 09:12:58 fhem-heizung evcc[11404]:       github.com/evcc-io/evcc/server/mqtt.go:306 +0xdcc
Okt 29 09:12:58 fhem-heizung evcc[11404]: created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
Okt 29 09:12:58 fhem-heizung evcc[11404]:       github.com/evcc-io/evcc/cmd/root.go:202 +0xf44
Okt 29 09:12:58 fhem-heizung systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Okt 29 09:12:58 fhem-heizung systemd[1]: evcc.service: Failed with result 'exit-code'.
```

docolli, 2023-10-29:
@emcepe:
Wenn ich mir den Code in server\mqtt.go anschaue, dann gibt es zwar diese Variable, aber die Funktion in der sie verwendet wird steht noch auf ToDo...
[image]

Ich kann zwar kein go, aber ich sehe in der Funktion keinen Hinweis, dass hier was entfernt wird. Alle Werte werden ganz normal über publish verarbeitet.

docolli, 2023-10-29:
Ich denke, der Fehler hat mit dieser Code Änderung zu tun: https://github.com/evcc-io/evcc/pull/10529
Aber das weiß andig sicher schon 😉

andig, 2023-10-29:
> im MQTT Explorer sieht man das jedoch alles nicht mehr bzw. fast: site/log ist noch da und site/stats:.* ebenso

Ich verstehe nur Bahnhof. Sind da jetzt überzählige Werte oder nicht?

docolli, 2023-10-29:
Version 0121.5 läuft bei mir jetzt wieder stabil ohne Fehlermeldungen.

Bei mir im FHEM MQTT2_Device für evcc haben sich inzwischen einen Haufen MQTT Topics in der ReadingList angesammelt:
```

evcc/status:.* status
evcc/loadpoints:.* loadpoints
evcc/site/activePhases:.* activePhases
evcc/site/range:.* range
evcc/site/socCharge:.* socCharge
evcc/site/vehicleSoC:.* vehicleSoC
evcc/site/batterySoC:.* batterySoC
evcc/site/bufferSoC:.* bufferSoC
evcc/site/minSoC:.* minSoC
evcc/site/prioritySoC:.* prioritySoC
evcc/site/targetSoC:.* targetSoC
evcc/site/vehicleTargetSoC:.* vehicleTargetSoC
evcc/loadpoints/1/activePhases:.* loadpoints_1_activePhases
evcc/loadpoints/1/range:.* loadpoints_1_range
evcc/loadpoints/1/socCharge:.* loadpoints_1_socCharge
evcc/loadpoints/1/vehicleSoC:.* loadpoints_1_vehicleSoC
evcc/loadpoints/1/batterySoC:.* loadpoints_1_batterySoC
evcc/loadpoints/1/bufferSoC:.* loadpoints_1_bufferSoC
evcc/loadpoints/1/minSoC:.* loadpoints_1_minSoC
evcc/loadpoints/1/prioritySoC:.* loadpoints_1_prioritySoC
evcc/loadpoints/1/targetSoC:.* loadpoints_1_targetSoC
evcc/loadpoints/1/vehicleTargetSoC:.* loadpoints_1_vehicleTargetSoC
evcc/site/pv/0:.* pv_0
evcc/site/battery/0:.* battery_0
evcc/site/vehicles/0:.* vehicles_0
evcc/site/pv/1:.* pv_1
evcc/site/battery/1:.* battery_1
evcc/site/vehicles/1:.* vehicles_1
evcc/site/pv/2:.* pv_2
evcc/site/battery/2:.* battery_2
evcc/site/vehicles/2:.* vehicles_2
evcc/site/pv/3:.* pv_3
evcc/site/battery/3:.* battery_3
evcc/site/vehicles/3:.* vehicles_3
evcc/site/pv/4:.* pv_4
evcc/site/battery/4:.* battery_4
evcc/site/vehicles/4:.* vehicles_4
evcc/site/pv/5:.* pv_5
evcc/site/battery/5:.* battery_5
evcc/site/vehicles/5:.* vehicles_5
evcc/site/pv/6:.* pv_6
evcc/site/battery/6:.* battery_6
evcc/site/vehicles/6:.* vehicles_6
evcc/site/pv/7:.* pv_7
evcc/site/battery/7:.* battery_7
evcc/site/vehicles/7:.* vehicles_7
evcc/site/pv/8:.* pv_8
evcc/site/battery/8:.* battery_8
evcc/site/vehicles/8:.* vehicles_8
evcc/site/pv/9:.* pv_9
evcc/site/battery/9:.* battery_9
evcc/site/vehicles/9:.* vehicles_9
evcc/updated:.* updated
evcc/site/siteTitle:.* siteTitle
evcc/site/gridConfigured:.* gridConfigured
evcc/site/pvConfigured:.* pvConfigured
evcc/site/batteryConfigured:.* batteryConfigured
evcc/site/bufferSoc:.* bufferSoc
evcc/site/prioritySoc:.* prioritySoc
evcc/site/residualPower:.* residualPower
evcc/site/smartCostLimit:.* smartCostLimit
evcc/site/smartCostUnit:.* smartCostUnit
evcc/site/smartCostAvailable:.* smartCostAvailable
evcc/site/currency:.* currency
evcc/site/savingsSince:.* savingsSince
[... 108 lines cut ...]
evcc/loadpoints/1/savingsGridCharged:.* loadpoints_1_savingsGridCharged
evcc/loadpoints/1/savingsSelfConsumptionCharged:.* loadpoints_1_savingsSelfConsumptionCharged
evcc/loadpoints/1/savingsSelfConsumptionPercent:.* loadpoints_1_savingsSelfConsumptionPercent
evcc/loadpoints/1/savingsTotalCharged:.* loadpoints_1_savingsTotalCharged
evcc/loadpoints/1/chargeVoltages/1:.* chargeVoltages_1
evcc/loadpoints/1/chargeVoltages/2:.* chargeVoltages_2
evcc/loadpoints/1/chargeVoltages/3:.* chargeVoltages_3
evcc/loadpoints/1/chargeCurrents/1:.* chargeCurrents_1
evcc/loadpoints/1/chargeCurrents/2:.* chargeCurrents_2
evcc/loadpoints/1/chargeCurrents/3:.* chargeCurrents_3
evcc/site/stats/30d/solarPercentage:.* solarPercentage
evcc/site/stats/30d/chargedKWh:.* chargedKWh
evcc/site/stats/30d/avgPrice:.* avgPrice
evcc/site/stats/30d/avgCo2:.* avgCo2
evcc/site/stats/365d/solarPercentage:.* solarPercentage
evcc/site/stats/365d/chargedKWh:.* chargedKWh
evcc/site/stats/365d/avgPrice:.* avgPrice
evcc/site/stats/365d/avgCo2:.* avgCo2
evcc/site/stats/total/chargedKWh:.* chargedKWh
evcc/site/stats/total/avgPrice:.* avgPrice
evcc/site/stats/total/avgCo2:.* avgCo2
evcc/site/stats/total/solarPercentage:.* solarPercentage
```

Ob die allerdings daher kommen, dass FHEM die im device behält, obwohl der MQTT Topic bereits gelöscht wurde, oder ob die immer noch per MQTT von evcc kommen, kann ich leider nicht sagen. @emcepe : Wie kann ich FHEM hier "bereinigen", sodaß ich nur noch die Topics automatisch angelegt bekomme, die evcc aktuell noch anbietet?

emcepe, 2023-10-29:
@docolli 
soweit ich weiß gar nicht.

Ich lösche die alle ab und zu mal, wenn da Topics umbenannt wurden und ich diverse Versionen verpasst habe und warte dann, das entsprechend die noch da sind erstellt werden.

Wie ich ja geschrieben habe, werden leider nach wie vor noch Topics erstellt die es eigentlich nicht mehr geben sollte (z.B. die ganzen SoC Sachen, die wurden vor langer Zeit in Soc umbenannt - C vs c)

Solche Sachen ändere ich dann in der Readinglist auf {} so dass die nicht mehr erstellt werden.

emcepe, 2023-10-29:
> > im MQTT Explorer sieht man das jedoch alles nicht mehr bzw. fast: site/log ist noch da und site/stats:.* ebenso
> 
> 
> 
> Ich verstehe nur Bahnhof. Sind da jetzt überzählige Werte oder nicht?

Ja.

docolli, 2023-10-29:
Im MQTT Explorer sieht es jetzt gut aus, die Topics machen Sinn. Das Problem ist jetzt bei mir (und vermutlich @emcepe ) noch, was FHEM aus den Topics für Readings macht.

Ist zwar etwas OT, aber falls sich ein evcc Nutzer mit Anbindung an FHEM per MQTT hierher findet:

FHEM mit autocreate hat meiner Meinung nach ein komisches Verhalten beim Anlegen der ReadingsList und den Namen, der den MQTT Topics zugewiesen wird. So hatte ich z.B. folgende Einträge:

```
evcc/loadpoints/1/savingsSelfConsumptionPercent:.* loadpoints_1_savingsSelfConsumptionPercent\
evcc/loadpoints/1/savingsTotalCharged:.* loadpoints_1_savingsTotalCharged\
evcc/loadpoints/1/chargeVoltages/1:.* chargeVoltages_1\
evcc/loadpoints/1/chargeVoltages/2:.* chargeVoltages_2\
evcc/loadpoints/1/chargeVoltages/3:.* chargeVoltages_3\
```
Ich verstehe nicht, weshalb der Topic `evcc/loadpoints/1/savingsSelfConsumptionPercent` den Namen `loadpoints_1_savingsSelfConsumptionPercent` bekommen hat, aber der Topic `evcc/loadpoints/1/chargeVoltages/1` den Namen `chargeVoltages_1`. Hier hätte ich jetzt `loadpoints_1_chargeVoltages_1` erwartet.

Problematisch ist es mit den neuen stats Topics:
```
evcc/site/stats/30d/solarPercentage:.* solarPercentage\
evcc/site/stats/30d/chargedKWh:.* chargedKWh\
evcc/site/stats/30d/avgPrice:.* avgPrice\
evcc/site/stats/30d/avgCo2:.* avgCo2\
evcc/site/stats/365d/solarPercentage:.* solarPercentage\
evcc/site/stats/365d/chargedKWh:.* chargedKWh\
evcc/site/stats/365d/avgPrice:.* avgPrice\
evcc/site/stats/365d/avgCo2:.* avgCo2\
evcc/site/stats/total/chargedKWh:.* chargedKWh\
evcc/site/stats/total/avgPrice:.* avgPrice\
evcc/site/stats/total/avgCo2:.* avgCo2\
evcc/site/stats/total/solarPercentage:.* solarPercentage
```

FHEM hat für 3 unterschiedliche Topics jeweils den gleichen Namen vergeben! Das geht natürlich vollkommen schief bei der Auswertung der MQTT Werte. 
Ich habe nun die ReadingsList folgendermaßen manuell angepasst, sodaß ich die MQTT Werte in FHEM eindeutig unterscheiden kann.

```
evcc/updated:.* updated
evcc/site/smartCostActive:.* site_smartCostActive
evcc/loadpoints/1/chargePower:.* loadpoints_1_chargePower
evcc/site/pvPower:.* site_pvPower
evcc/site/pvEnergy:.* site_pvEnergy
evcc/site/pv:.* site_pv
evcc/site/pv/1/power:.* site_pv_1_power
evcc/site/pv/1/energy:.* site_pv_1_energy
evcc/site/batteryCapacity:.* site_batteryCapacity
evcc/site/batterySoc:.* site_batterySoc
evcc/site/batteryPower:.* site_batteryPower
evcc/site/batteryEnergy:.* site_batteryEnergy
evcc/site/battery:.* site_battery
evcc/site/battery/1/power:.* site_battery_1_power
evcc/site/battery/1/energy:.* site_battery_1_energy
evcc/site/battery/1/soc:.* site_battery_1_soc
evcc/site/battery/1/capacity:.* site_battery_1_capacity
evcc/site/gridPower:.* site_gridPower
evcc/site/homePower:.* site_homePower
evcc/loadpoints/1/chargeVoltages/l1:.* loadpoints_1_chargeVoltages_l1
evcc/loadpoints/1/chargeVoltages/l2:.* loadpoints_1_chargeVoltages_l2
evcc/loadpoints/1/chargeVoltages/l3:.* loadpoints_1_chargeVoltages_l3
evcc/loadpoints/1/chargeVoltages:.* loadpoints_1_chargeVoltages
evcc/loadpoints/1/chargeVoltages/1:.* loadpoints_1_chargeVoltages_1
evcc/loadpoints/1/chargeVoltages/2:.* loadpoints_1_chargeVoltages_2
evcc/loadpoints/1/chargeVoltages/3:.* loadpoints_1_chargeVoltages_3
evcc/loadpoints/1/chargeCurrents/l1:.* loadpoints_1_chargeCurrents_l1
evcc/loadpoints/1/chargeCurrents/l2:.* loadpoints_1_chargeCurrents_l2
evcc/loadpoints/1/chargeCurrents/l3:.* loadpoints_1_chargeCurrents_l3
evcc/loadpoints/1/chargeCurrents:.* loadpoints_1_chargeCurrents
evcc/loadpoints/1/chargeCurrents/1:.* loadpoints_1_chargeCurrents_1
evcc/loadpoints/1/chargeCurrents/2:.* loadpoints_1_chargeCurrents_2
evcc/loadpoints/1/chargeCurrents/3:.* loadpoints_1_chargeCurrents_3
evcc/loadpoints/1/phasesActive:.* loadpoints_1_phasesActive
evcc/loadpoints/1/chargeCurrent:.* loadpoints_1_chargeCurrent
[... 122 lines cut ...]
evcc/loadpoints/1/phaseAction:.* loadpoints_1_phaseAction
evcc/loadpoints/1/phaseRemaining:.* loadpoints_1_phaseRemaining
evcc/loadpoints/1/pvAction:.* loadpoints_1_pvAction
evcc/loadpoints/1/pvRemaining:.* loadpoints_1_pvRemaining
evcc/loadpoints/1/chargerFeatureIntegratedDevice:.* loadpoints_1_chargerFeatureIntegratedDevice
evcc/loadpoints/1/chargerFeatureHeating:.* loadpoints_1_chargerFeatureHeating
evcc/loadpoints/1/chargerIcon:.* loadpoints_1_chargerIcon
evcc/loadpoints/1/minSoc:.* loadpoints_1_minSoc
evcc/loadpoints/1/targetEnergy:.* loadpoints_1_targetEnergy
evcc/loadpoints/1/vehiclePresent:.* loadpoints_1_vehiclePresent
evcc/loadpoints/1/vehicleTitle:.* loadpoints_1_vehicleTitle
evcc/loadpoints/1/vehicleIcon:.* loadpoints_1_vehicleIcon
evcc/loadpoints/1/vehicleCapacity:.* loadpoints_1_vehicleCapacity
evcc/loadpoints/1/vehicleTargetSoc:.* loadpoints_1_vehicleTargetSoc
evcc/loadpoints/1/vehicleFeatureOffline:.* loadpoints_1_vehicleFeatureOffline
evcc/loadpoints/1/priority:.* loadpoints_1_priority
evcc/loadpoints/1/vehicleDetectionActive:.* loadpoints_1_vehicleDetectionActive
evcc/site/version:.* site_version
evcc/site/sponsor:.* site_sponsor
evcc/site/sponsorTokenExpires:.* site_sponsorTokenExpires
evcc/loadpoints/1/connectedDuration:.* loadpoints_1_connectedDuration
evcc/site/tariffGrid:.* site_tariffGrid
evcc/site/tariffFeedIn:.* site_tariffFeedIn
evcc/site/tariffCo2:.* site_tariffCo2
```

emcepe, 2023-10-29:
@docolli 

das ist wirklich ziemlich offtopic hier und gerade die mega Listen von FHEM MQTT Readinglisten machen es nicht übersichtlicher ;-(

Ich hab mich da auch schon drüber geärgert und versucht die Logik dahinter zu verstehen, hab's aber sein gelassen und ändere ggf. den Readingnamen.

Über PM können wir uns darüber natürlich gerne unterhalten 😃

docolli, 2023-10-31:
> > > im MQTT Explorer sieht man das jedoch alles nicht mehr bzw. fast: site/log ist noch da und site/stats:.* ebenso
> > 
> > 
> > Ich verstehe nur Bahnhof. Sind da jetzt überzählige Werte oder nicht?
> 
> Ja.

Mit 0.121.5 sehe ich mit dem MQTT Explorer keine überzähligen Werte mehr.
@emcepe Vermutlich hat dein FHEM noch irgendwo was "im Cache", aber das machen wir per PM.