evcc-io/evcc #10517: Inkonsitente Werte für savings zwischen GUI und MQTT bzw. Werte in Datenbank

docolli, 2023-10-27:
### Describe the bug

Bei mir sind die Werte zur Sonnenenergie nicht konsistent zu den Werten, die per MQTT als savings... ausgegeben werden.
[image]
[image]

Hier die Werte aus der evcc.db:
[image]

Mir ist klar, dass ihr die Berechnungsgrundlage geändert habt. Aber dann sollten die Werte, die EVCC über die verschiedenen Ausgabekanäle ausgibt, konsistent sein.

### Steps to reproduce

Werte im GUI "Auswertung Ladeenenergie" vergleichen mit den MQTT Werten für savings...

### Configuration details

```yaml
<kommt bei Bedarf>
```

### Log details

```text
<kommt bei Bedarf>
```

### What type of operating system are you running?

Linux

### Version

0121.2

StefanSchoof, 2023-10-27:
Schau mal unter https://github.com/evcc-io/evcc/issues/10482

closed 2023-10-28 by andig

docolli, 2023-10-28:
@StefanSchoof 
Danke für die Hintergrundinfo. Ich verstehe, dass sich die Berechnungsgrundlage geändert hat und somit seit 0121.2 nur noch Ladevorgänge in diese Statistik eingehen, die auch den Wert "Solar %" haben.

**ABER:** Die per MQTT ausgegebenen Statistikwerte für die "Ladung Solar" zu "Ladung Netz", also alle MQTT Werte die mit "savings." beginnen, sind damit nicht mehr mit den Werten im GUI konsistent! Die MQTT Werte werden seit der Installation von evcc aufsummiert, während die GUI Werte jetzt seit Einführung von "Solar %" aufsummiert werden.
Dies "savings." Werte sind in der DB zu finden und werden vermutlich nach jeder Ladung aufkummultiert. Da schon seit längerem die Statistikberechnung im GUI auf den einzelnen Datenbankeinträgen beruht, sollte der Code zur Ermittelung der MQTT Statistikwerte an diese Berechungsweise angepasst werden.

Der Issue sollte daher wieder geöffnet werden. 

Noch was: Ich habe in der Vergangenheit einzelne (Test-)Ladungen per GUI aus der Datenbank gelöscht. Aus den kummulierten Werten für "savings..." wurde diese Ladevorgänge aber offensichtlich nicht gelöscht. Sieht man ganz gut oben in den Werten, die ich gepostet habe. Die Summe aller Ladungen aus der DB sind 783,089kWh, addiere ich jedoch savings.gridCharged (70,62 kWh) und savings.selfConsumptionCharged (722,13 kWh), so sind dies 792,75 kWh. Dies spricht ebenfalls für eine dynamische Berechung der "savings..." Werte aus der DB bei einer MQTT Kommunikation.

andig, 2023-10-28:
> Noch was: Ich habe in der Vergangenheit einzelne (Test-)Ladungen per GUI aus der Datenbank gelöscht. Aus den kummulierten Werten für "savings..." wurde diese Ladevorgänge aber offensichtlich nicht gelöscht.

Da es die savings jetzt nicht mehr gibt ist das Problem gelöst :)

docolli, 2023-10-28:
Was? Moment.... 😲
Achso! Jetzt verstehe ich das. Stimmt seit der Installation von 0121.2 wurden die savings nicht mehr aktualisiert. Sorry, das ist mir nicht aufgefallen. 🙈

Gibt es diese Werte jetzt an anderer Stelle per MQTT? Oder sind die ersatzlos entfallen?

andig, 2023-10-28:
Schaust Du: https://github.com/evcc-io/evcc/pull/10341

docolli, 2023-10-28:
Sorry, GENAU Lesen bildet... Ja, dort steht, dass und wie sich die API geändert hat. Bin schon still 🤐 🤣
Schönen Abend und großes Lob an Euch für evcc !!!