evcc-io/evcc #10522: geänderte Schriftart

StevieC121176, 2023-10-28:
### Describe the bug

Gibt es jetzt verschiedene Schriftarten in EVCC? Besonders sieht man es an der "1" . die sehen selbst innerhalb einer Ansicht verschieden aus (sh. Bilder unten). Ich habe mal ein älteren Screenshot angehängt, da ist noch die normale "1" zu sehen. Welche mir btw. viel besser gefallen hat :-)
Hat der Wechsel einen Grund? 

EVCC alt:
[image]

EVCC aktuell:
[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

naltatis, 2023-10-28:
Hm, dein Screenshot überrascht mich. Was ich an einigen Stellen in der Anwendung verändert habe ist das Spacing bei Zahlen, die tabellarisch dargestellt werden bzw. wo wir Animationen haben. Bislang war es so, dass die Digits nicht immer übereinander standen bzw. bei Wertänderung der Eintrag "gewackelt" hat.

Konkret ist das jetzt per CSS mit `font-variant-numeric: tabular-nums;` umgesetzt.
https://sebastiandedeyne.com/tabular-numbers

Das sollte aber eigentlich keine Auswirkung auf die Schriftart selbst haben. Ich schau mir das mal genauer an.

StevieC121176, 2023-10-28:
Ich dachte evtl. das der Browser in meiner IOS 17.2 Beta nicht damit klar kommt, aber daran liegt es auch nicht. Ich habe es jetzt auf meinem Fernseher und einem alten Android Telefon probiert, mit Chrome, Firefox und Safari Browser. Überall die gleiche Darstellung.

PR #11039 (2023-12-07): UI: revert tabular nums
fixes #10522

Monserrat unterstützt Formatierung für tabellarische. Ändert dabei aber einige Zeichen so, dass es ein inkonsistentes Erscheinungsbild gibt. Durchgängiges Verwenden von tabular-nums (auch in nicht tabellarischen Anwendungen) wäre eine Option, sperrt aber die Schrift unnötig.

Daher reverte ich hier die Änderung wieder zugunsten einheitlicher Zeichen.

`1` im normalen Text
[image]

`1` in tabular
[image]
closed 2023-12-07 by naltatis