evcc-io/evcc #10527: Vestel WB Fehler "charger out of sync: expected disabled, got enabled" nach Update auf EVCC 0.121.2

ogheinz77, 2023-10-28:
### Describe the bug

Hallo,

ich bekomme seit dem Update auf EVCC 0.121.2 folgenden Fehler:

Oct 28 14:13:46 evcc evcc[20132]: [lp-1  ] WARN 2023/10/28 14:13:46 charger out of sync: expected disabled, got enabled
Oct 28 14:15:34 evcc evcc[20132]: [lp-1  ] WARN 2023/10/28 14:15:34 charger out of sync: expected disabled, got enabled
Oct 28 14:17:22 evcc evcc[20132]: [lp-1  ] WARN 2023/10/28 14:17:22 charger out of sync: expected disabled, got enabled
Oct 28 14:19:10 evcc evcc[20132]: [lp-1  ] WARN 2023/10/28 14:19:10 charger out of sync: expected disabled, got enabled
Oct 28 14:20:58 evcc evcc[20132]: [lp-1  ] WARN 2023/10/28 14:20:58 charger out of sync: expected disabled, got enabled

Die Meldungen kommen vorwiegend wenn kein Auto dran ist.

Hatte den Fehler schon mal, war dann allerdings weg als ich EVCC Intervall auf 18s gestellt habe.
Heute habe ich auf EVCC 0.121.2 aktualisiert und der Fehler kommt ca. alle 2 min.
Mit dem EVCC Intervall weiter runter zugehen bringt kein Erfolg.
Bei der EVCC 0.121.0 & 0.121.1 ist der Fehler auch vorhanden.

Wenn ich ein Downgrade auf EVCC 0.120.03 mache bekomme ich den Fehler nicht!

Info:
Vestel Firmware: v3.118.0 (neuste)
EVCC: 0.121.2

Hatte das auch schon hier rein geschrieben:
https://github.com/evcc-io/evcc/discussions/5920 & https://github.com/evcc-io/evcc/discussions/10409

### Steps to reproduce

1. Start Evcc
2. Wait 2-3 min
3. Check Log --> journalctl -u evcc --since xx:xx 

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 18s

sponsortoken:xxxxxxxxxxxxxx

meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: xxx.xxx.xxx.xxx  
  port: 502  
  name: Fronius-SmartMeter
- type: template
  template: fronius-gen24 
  usage: pv  
  host: xxx.xxx.xxx.xxx  
  port: 502  
  name: FroniusGen24
- type: template
  template: fronius-gen24 
  usage: battery  
  host: xxx.xxx.xxx.xxx  
  port: 502  
  name: BYD HVS 10.2  
[... 275 lines cut ...]
[lp-1  ] DEBUG 2023/10/28 14:35:26 detected connected phases: 3p
[lp-1  ] DEBUG 2023/10/28 14:35:26 charge currents: [0.049 0.018 0.018]A
[lp-1  ] DEBUG 2023/10/28 14:35:26 charge total import: 369.700kWh
[lp-1  ] DEBUG 2023/10/28 14:35:26 charger status: A
[lp-1  ] DEBUG 2023/10/28 14:35:26 charger: guard elapse
[lp-1  ] DEBUG 2023/10/28 14:35:26 guard timer inactive
[lp-1  ] DEBUG 2023/10/28 14:35:26 charger disable
[lp-1  ] DEBUG 2023/10/28 14:35:26 wake-up timer: stop
[site  ] DEBUG 2023/10/28 14:35:43 ----
[lp-1  ] DEBUG 2023/10/28 14:35:43 charge power: 4W
[site  ] DEBUG 2023/10/28 14:35:44 pv power: 326W
[site  ] DEBUG 2023/10/28 14:35:44 battery soc: 41%
[site  ] DEBUG 2023/10/28 14:35:44 battery power: 67W
[site  ] DEBUG 2023/10/28 14:35:44 grid power: 2W
[site  ] DEBUG 2023/10/28 14:35:44 grid currents: [1.08 -0.457 -0.15]A
[site  ] DEBUG 2023/10/28 14:35:44 site power: 168W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.2

andig, 2023-10-28:
Ich mache das jetzt zum 3. mal zu. 

> [site  ] WARN 2023/10/28 14:30:01 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

Zuerst: Fehler beseitigen! Dann- wie schonmal geschrieben- vollständiges Logfile mit Charger auf trace das den Fehler zeigt. Alles andere ist nur Beschäftigungstherapie.

ogheinz77, 2023-10-28:
Hallo,

sorry ist mein erstes Issue was ich hier eröffne.

Configuration details:

log: info
levels:
  cache: error
  vestel: trace

Log details:

evcc --log debug
[main  ] INFO 2023/10/28 18:44:41 evcc 0.121.2
[main  ] INFO 2023/10/28 18:44:41 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/28 18:44:41 starting ui and api at :7070
[db    ] INFO 2023/10/28 18:44:41 using sqlite database: /root/.evcc/evcc.db
[vestel] TRACE 2023/10/28 18:44:42 modbus: send 00 01 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/10/28 18:44:42 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 00 00
[site  ] INFO 2023/10/28 18:44:42 site config:
[site  ] INFO 2023/10/28 18:44:42   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/28 18:44:42     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/28 18:44:42     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/28 18:44:42     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/10/28 18:44:42   vehicles:
[site  ] INFO 2023/10/28 18:44:42     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/28 18:44:42 loadpoint 1:
[lp-1  ] INFO 2023/10/28 18:44:42   mode:        minpv
[lp-1  ] INFO 2023/10/28 18:44:42   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/28 18:44:42   meters:      charge ✓
[lp-1  ] INFO 2023/10/28 18:44:42     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/28 18:44:42 phase timer inactive
[lp-1  ] DEBUG 2023/10/28 18:44:42 pv timer inactive
[lp-1  ] DEBUG 2023/10/28 18:44:42 guard timer inactive
[lp-1  ] INFO 2023/10/28 18:44:42 vehicle updated: unknown -> MINI SE
[lp-1  ] DEBUG 2023/10/28 18:44:42 set min current: 8
[lp-1  ] DEBUG 2023/10/28 18:44:42 set max current: 16
[vestel] TRACE 2023/10/28 18:44:42 modbus: send 00 02 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/10/28 18:44:42 modbus: recv 00 02 00 00 00 05 ff 03 02 00 06
[vestel] TRACE 2023/10/28 18:44:42 modbus: send 00 03 00 00 00 06 ff 06 13 8c 00 08
[vestel] TRACE 2023/10/28 18:44:42 modbus: recv 00 03 00 00 00 06 ff 06 13 8c 00 08
[lp-1  ] DEBUG 2023/10/28 18:44:42 max charge current: 8A
[vestel] TRACE 2023/10/28 18:44:44 modbus: send 00 04 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:44:44 modbus: recv 00 04 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:44:47 modbus: send 00 05 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:44:47 modbus: recv 00 05 00 00 00 06 ff 06 17 70 00 01
[site  ] DEBUG 2023/10/28 18:44:50 ----
[vestel] TRACE 2023/10/28 18:44:50 modbus: send 00 06 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/10/28 18:44:50 modbus: recv 00 06 00 00 00 07 ff 04 04 00 00 00 04
[lp-1  ] DEBUG 2023/10/28 18:44:50 charge power: 4W
[main  ] INFO 2023/10/28 18:44:50 new version available: 0.121.3
[vestel] TRACE 2023/10/28 18:44:50 modbus: send 00 07 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:44:50 modbus: recv 00 07 00 00 00 06 ff 06 17 70 00 01
[lp-1  ] DEBUG 2023/10/28 18:44:51 vehicle odometer: 1261km
[vestel] TRACE 2023/10/28 18:44:51 modbus: send 00 08 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/10/28 18:44:51 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e3
[vestel] TRACE 2023/10/28 18:44:51 modbus: send 00 09 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/10/28 18:44:51 modbus: recv 00 09 00 00 00 05 ff 04 02 00 e4
[vestel] TRACE 2023/10/28 18:44:51 modbus: send 00 0a 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/10/28 18:44:51 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 e9
[lp-1  ] DEBUG 2023/10/28 18:44:51 charge voltages: [227 228 233]V
[lp-1  ] DEBUG 2023/10/28 18:44:51 detected connected phases: 3p
[vestel] TRACE 2023/10/28 18:44:51 modbus: send 00 0b 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/10/28 18:44:51 modbus: recv 00 0b 00 00 00 05 ff 04 02 00 31
[vestel] TRACE 2023/10/28 18:44:51 modbus: send 00 0c 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/10/28 18:44:51 modbus: recv 00 0c 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/10/28 18:44:51 modbus: send 00 0d 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/10/28 18:44:51 modbus: recv 00 0d 00 00 00 05 ff 04 02 00 12
[lp-1  ] DEBUG 2023/10/28 18:44:51 charge currents: [0.049 0.018 0.018]A
[... 328 lines cut ...]
[vestel] TRACE 2023/10/28 18:57:20 modbus: recv 01 82 00 00 00 06 ff 06 17 70 00 01
[site  ] DEBUG 2023/10/28 18:57:21 pv power: 0W
[site  ] DEBUG 2023/10/28 18:57:21 battery soc: 15%
[site  ] DEBUG 2023/10/28 18:57:21 battery power: 0W
[site  ] DEBUG 2023/10/28 18:57:21 grid power: 2962W
[site  ] DEBUG 2023/10/28 18:57:21 grid currents: [11.8 1.13 0.328]A
[site  ] DEBUG 2023/10/28 18:57:21 site power: 3062W
[vestel] TRACE 2023/10/28 18:57:23 modbus: send 01 83 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:23 modbus: recv 01 83 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:26 modbus: send 01 84 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:26 modbus: recv 01 84 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:29 modbus: send 01 85 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:29 modbus: recv 01 85 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:32 modbus: send 01 86 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:32 modbus: recv 01 86 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:35 modbus: send 01 87 00 00 00 06 ff 06 17 70 00 01
[... cut ...]
[vestel] TRACE 2023/10/28 18:57:41 modbus: send 01 89 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:41 modbus: recv 01 89 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:44 modbus: send 01 8a 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:44 modbus: recv 01 8a 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:47 modbus: send 01 8b 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 18:57:47 modbus: recv 01 8b 00 00 00 06 ff 06 17 70 00 01

andig, 2023-10-28:
Danke. Was bedeuten die ~ zwischendrin? Hättest du das gleiche Log auch nochmal mit alter Version wo es keine Probleme gibt?

ogheinz77, 2023-10-28:
Wollte das Log nicht so lange machen, hab ein paar Zeilen raus und mit ~ gekennzeichnet. Hoffe das passt so.
Log mit der 0.120.03 kommt gleich

ogheinz77, 2023-10-28:
Noch was zur Info, das "charger out of sync" kommt bei der Version 0.120.03 nur nicht wenn interval: 18s. !!

--> https://github.com/evcc-io/evcc/discussions/10409

Logs kommen gleich

ogheinz77, 2023-10-28:
Log mit evcc 0.120.3 und interval: 30s

evcc --log debug
[main  ] INFO 2023/10/28 20:55:17 evcc 0.120.3
[main  ] INFO 2023/10/28 20:55:17 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/28 20:55:17 starting ui and api at :7070
[db    ] INFO 2023/10/28 20:55:17 using sqlite database: /root/.evcc/evcc.db
[vestel] TRACE 2023/10/28 20:55:18 modbus: send 00 01 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/10/28 20:55:18 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 00 00
[site  ] INFO 2023/10/28 20:55:18 site config:
[site  ] INFO 2023/10/28 20:55:18   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/28 20:55:18     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/28 20:55:18     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/28 20:55:18     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/10/28 20:55:18   vehicles:
[site  ] INFO 2023/10/28 20:55:18     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/28 20:55:18 loadpoint 1:
[lp-1  ] INFO 2023/10/28 20:55:18   mode:        minpv
[lp-1  ] INFO 2023/10/28 20:55:18   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/28 20:55:18   meters:      charge ✓
[lp-1  ] INFO 2023/10/28 20:55:18     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/28 20:55:18 phase timer inactive
[lp-1  ] DEBUG 2023/10/28 20:55:18 pv timer inactive
[lp-1  ] DEBUG 2023/10/28 20:55:18 guard timer inactive
[lp-1  ] INFO 2023/10/28 20:55:18 vehicle updated: unknown -> MINI SE
[lp-1  ] DEBUG 2023/10/28 20:55:18 set min current: 8
[lp-1  ] DEBUG 2023/10/28 20:55:18 set max current: 16
[vestel] TRACE 2023/10/28 20:55:18 modbus: send 00 02 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/10/28 20:55:18 modbus: recv 00 02 00 00 00 05 ff 03 02 00 06
[vestel] TRACE 2023/10/28 20:55:18 modbus: send 00 03 00 00 00 06 ff 06 13 8c 00 08
[vestel] TRACE 2023/10/28 20:55:18 modbus: recv 00 03 00 00 00 06 ff 06 13 8c 00 08
[lp-1  ] DEBUG 2023/10/28 20:55:18 max charge current: 8A
[vestel] TRACE 2023/10/28 20:55:20 modbus: send 00 04 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 20:55:20 modbus: recv 00 04 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 20:55:23 modbus: send 00 05 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 20:55:23 modbus: recv 00 05 00 00 00 06 ff 06 17 70 00 01
[site  ] DEBUG 2023/10/28 20:55:26 ----
[vestel] TRACE 2023/10/28 20:55:26 modbus: send 00 06 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/10/28 20:55:26 modbus: recv 00 06 00 00 00 07 ff 04 04 00 00 00 04
[lp-1  ] DEBUG 2023/10/28 20:55:26 charge power: 4W
[main  ] INFO 2023/10/28 20:55:26 new version available: 0.121.3
[vestel] TRACE 2023/10/28 20:55:26 modbus: send 00 07 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 20:55:26 modbus: recv 00 07 00 00 00 06 ff 06 17 70 00 01
[site  ] DEBUG 2023/10/28 20:55:27 pv power: 0W
[site  ] DEBUG 2023/10/28 20:55:27 battery soc: 14%
[site  ] DEBUG 2023/10/28 20:55:27 battery power: 0W
[site  ] DEBUG 2023/10/28 20:55:27 grid power: 494W
[site  ] DEBUG 2023/10/28 20:55:27 grid currents: [1.8 0.844 0.375]A
[site  ] DEBUG 2023/10/28 20:55:27 site power: 594W
[lp-1  ] DEBUG 2023/10/28 20:55:28 vehicle odometer: 1261km
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 08 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e6
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 09 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 09 00 00 00 05 ff 04 02 00 e8
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 0a 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 ea
[lp-1  ] DEBUG 2023/10/28 20:55:28 charge voltages: [230 232 234]V
[lp-1  ] DEBUG 2023/10/28 20:55:28 detected connected phases: 3p
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 0b 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 0b 00 00 00 05 ff 04 02 00 31
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 0c 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 0c 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 0d 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 0d 00 00 00 05 ff 04 02 00 12
[lp-1  ] DEBUG 2023/10/28 20:55:28 charge currents: [0.049 0.018 0.018]A
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 0e 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 0e 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 0f 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/10/28 20:55:28 modbus: recv 00 0f 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/10/28 20:55:28 modbus: send 00 10 00 00 00 06 ff 04 04 0c 00 02
[... 621 lines cut ...]
[vestel] TRACE 2023/10/28 21:03:05 modbus: recv 00 f3 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:03:08 modbus: send 00 f4 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:03:08 modbus: recv 00 f4 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:03:11 modbus: send 00 f5 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:03:11 modbus: recv 00 f5 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:03:14 modbus: send 00 f6 00 00 00 06 ff 06 17 70 00 01
[... cut ...]
[lp-1  ] DEBUG 2023/10/28 21:03:26 charge power: 4W
[vestel] TRACE 2023/10/28 21:03:26 modbus: send 00 fb 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:03:26 modbus: recv 00 fb 00 00 00 06 ff 06 17 70 00 01
[site  ] DEBUG 2023/10/28 21:03:27 pv power: 0W
[site  ] DEBUG 2023/10/28 21:03:27 battery soc: 14%
[site  ] DEBUG 2023/10/28 21:03:27 battery power: 0W
[site  ] DEBUG 2023/10/28 21:03:27 grid power: 488W
[site  ] DEBUG 2023/10/28 21:03:27 grid currents: [1.84 0.793 0.374]A
[site  ] DEBUG 2023/10/28 21:03:27 site power: 588W

ogheinz77, 2023-10-28:
**Log mit evcc 0.120.3 und interval: 18s**
Hier kommt der out of sync nicht:

evcc --log debug
[main  ] INFO 2023/10/28 21:09:20 evcc 0.120.3
[main  ] INFO 2023/10/28 21:09:20 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/28 21:09:20 starting ui and api at :7070
[db    ] INFO 2023/10/28 21:09:20 using sqlite database: /root/.evcc/evcc.db
[vestel] TRACE 2023/10/28 21:09:21 modbus: send 00 01 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/10/28 21:09:21 modbus: recv 00 01 00 00 00 07 ff 04 04 00 00 00 00
[site  ] INFO 2023/10/28 21:09:21 site config:
[site  ] INFO 2023/10/28 21:09:21   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/28 21:09:21     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/10/28 21:09:21     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/10/28 21:09:21     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/10/28 21:09:21   vehicles:
[site  ] INFO 2023/10/28 21:09:21     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/10/28 21:09:21     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/10/28 21:09:21     vehicle 3: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2023/10/28 21:09:21     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/28 21:09:21 loadpoint 1:
[lp-1  ] INFO 2023/10/28 21:09:21   mode:        minpv
[lp-1  ] INFO 2023/10/28 21:09:21   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/28 21:09:21   meters:      charge ✓
[lp-1  ] INFO 2023/10/28 21:09:21     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/28 21:09:21 phase timer inactive
[lp-1  ] DEBUG 2023/10/28 21:09:21 pv timer inactive
[lp-1  ] DEBUG 2023/10/28 21:09:21 guard timer inactive
[lp-1  ] INFO 2023/10/28 21:09:21 vehicle updated: unknown -> MINI SE
[lp-1  ] DEBUG 2023/10/28 21:09:21 set min current: 8
[lp-1  ] DEBUG 2023/10/28 21:09:21 set max current: 16
[vestel] TRACE 2023/10/28 21:09:21 modbus: send 00 02 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/10/28 21:09:21 modbus: recv 00 02 00 00 00 05 ff 03 02 00 06
[vestel] TRACE 2023/10/28 21:09:21 modbus: send 00 03 00 00 00 06 ff 06 13 8c 00 08
[vestel] TRACE 2023/10/28 21:09:21 modbus: recv 00 03 00 00 00 06 ff 06 13 8c 00 08
[lp-1  ] DEBUG 2023/10/28 21:09:21 max charge current: 8A
[vestel] TRACE 2023/10/28 21:09:24 modbus: send 00 04 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:09:24 modbus: recv 00 04 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:09:27 modbus: send 00 05 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:09:27 modbus: recv 00 05 00 00 00 06 ff 06 17 70 00 01
[site  ] WARN 2023/10/28 21:09:29 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/28 21:09:29 ----
[vestel] TRACE 2023/10/28 21:09:29 modbus: send 00 06 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/10/28 21:09:29 modbus: recv 00 06 00 00 00 07 ff 04 04 00 00 00 04
[lp-1  ] DEBUG 2023/10/28 21:09:29 charge power: 4W
[main  ] INFO 2023/10/28 21:09:29 new version available: 0.121.4
[vestel] TRACE 2023/10/28 21:09:30 modbus: send 00 07 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:09:30 modbus: recv 00 07 00 00 00 06 ff 06 17 70 00 01
[site  ] DEBUG 2023/10/28 21:09:30 pv power: 0W
[site  ] DEBUG 2023/10/28 21:09:30 battery soc: 14%
[site  ] DEBUG 2023/10/28 21:09:30 battery power: 0W
[site  ] DEBUG 2023/10/28 21:09:30 grid power: 469W
[site  ] DEBUG 2023/10/28 21:09:30 grid currents: [1.82 0.791 0.311]A
[site  ] DEBUG 2023/10/28 21:09:30 site power: 569W
[lp-1  ] DEBUG 2023/10/28 21:09:31 vehicle odometer: xxxx km
[vestel] TRACE 2023/10/28 21:09:31 modbus: send 00 08 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/10/28 21:09:31 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2023/10/28 21:09:31 modbus: send 00 09 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/10/28 21:09:31 modbus: recv 00 09 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2023/10/28 21:09:31 modbus: send 00 0a 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/10/28 21:09:31 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 eb
[lp-1  ] DEBUG 2023/10/28 21:09:31 charge voltages: [231 231 235]V
[lp-1  ] DEBUG 2023/10/28 21:09:31 detected connected phases: 3p
[vestel] TRACE 2023/10/28 21:09:31 modbus: send 00 0b 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/10/28 21:09:31 modbus: recv 00 0b 00 00 00 05 ff 04 02 00 31
[vestel] TRACE 2023/10/28 21:09:31 modbus: send 00 0c 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/10/28 21:09:31 modbus: recv 00 0c 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/10/28 21:09:31 modbus: send 00 0d 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/10/28 21:09:31 modbus: recv 00 0d 00 00 00 05 ff 04 02 00 12
[lp-1  ] DEBUG 2023/10/28 21:09:31 charge currents: [0.049 0.018 0.018]A
[... 831 lines cut ...]
[site  ] DEBUG 2023/10/28 21:17:19 grid power: 422W
[site  ] DEBUG 2023/10/28 21:17:19 grid currents: [1.81 0.607 0.312]A
[site  ] DEBUG 2023/10/28 21:17:19 site power: 522W
[vestel] TRACE 2023/10/28 21:17:21 modbus: send 01 24 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:17:21 modbus: recv 01 24 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:17:24 modbus: send 01 25 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:17:24 modbus: recv 01 25 00 00 00 06 ff 06 17 70 00 01
[... cut ...]
[lp-1  ] DEBUG 2023/10/28 21:17:35 charge power: 4W
[vestel] TRACE 2023/10/28 21:17:36 modbus: send 01 2a 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/28 21:17:36 modbus: recv 01 2a 00 00 00 06 ff 06 17 70 00 01
[site  ] DEBUG 2023/10/28 21:17:36 pv power: 0W
[site  ] DEBUG 2023/10/28 21:17:36 battery soc: 14%
[site  ] DEBUG 2023/10/28 21:17:36 battery power: 0W
[site  ] DEBUG 2023/10/28 21:17:36 grid power: 418W
[site  ] DEBUG 2023/10/28 21:17:36 grid currents: [1.81 0.6 0.312]A
[site  ] DEBUG 2023/10/28 21:17:36 site power: 518W

andig, 2023-10-29:
Ich sehe da kein Problem. Die Vestel ist relativ langsam und wir warnen extra von intervvsl <30s. Nicht ohne Grund…

ogheinz77, 2023-10-29:
Also, neuste EVCC Version installieren, interval auf 30s und die out of sync Meldungen ignorieren?

andig, 2023-10-29:
Ich verstehe nur Bahnhof. Mir fehlt weiter eine klare Fehlerbeschreibung wann es auftrii und wann nicht.

ogheinz77, 2023-10-29:
Ok, ich versuche es noch mal:

**Fehlerbeschreibung:**

**Meldung charger out of sync: expected disabled, got enabled im log vorhanden wenn:**
1. EVCC Version 0.120.03 und interval größer als 18s in yaml
2. EVCC Version größer als 0.120.03 (interval in yaml egal, Fehler kommt immer)

**Kein charger out of sync: expected disabled, got enabled im log wenn:**
EVCC Version 0.120.03 und interval 18s in yaml

andig, 2023-10-29:
> modbus: send 00 c4 00 00 00 06 ff 06 17 70 00 01

@premultiply braucht die Box wirklich alle 3s ein Heartbeat? Oder können wir das Intervall irgendwo auslesen?

Und dann:

```
[lp-1  ] DEBUG 2023/10/28 21:01:27 guard timer inactive
[vestel] TRACE 2023/10/28 21:01:27 modbus: send 00 c3 00 00 00 06 ff 06 13 8c 00 00
[vestel] TRACE 2023/10/28 21:01:27 modbus: recv 00 c3 00 00 00 06 ff 06 13 8c 00 00
[lp-1  ] DEBUG 2023/10/28 21:01:27 charger disable

[lp-1  ] DEBUG 2023/10/28 21:02:57 charger status: A
[vestel] TRACE 2023/10/28 21:02:57 modbus: send 00 ef 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/10/28 21:02:57 modbus: recv 00 ef 00 00 00 05 ff 03 02 00 06
**[lp-1  ] WARN 2023/10/28 21:02:57 charger out of sync: expected disabled, got enabled**
```

Wir schalten mittels Stromvorgabe 0 aus, später stehen da aber wieder 6A drin. Kannst Du Dir da einen reim drauf machen? Ich sehe auch nicht, was sich da geändert haben soll?

ogheinz77, 2023-10-29:
Gebt mir Bescheid wenn ich was für euch testen soll :)

andig, 2023-10-29:
Du könntest mal probieren:

    evcc charger --log trace
    evcc charger --log trace -d
    evcc charger --log trace
    evcc charger --log trace -e
    evcc charger --log trace

Also status, aus, status, ein, status.

ogheinz77, 2023-10-29:
Mit evcc 0.120.03 und interval 30s?
0.120.03 ist gerade installiert.

ogheinz77, 2023-10-29:
Evcc 0.120.3 mit  Interval: 30s

**evcc charger --log trace**

Vestel-EVC04
------------
[vestel] ERROR 2023/10/29 16:03:46 heartbeat: read tcp 192.178.66.112:51874->192.178.66.110:502: i/o timeout
[vestel] TRACE 2023/10/29 16:03:46 modbus: send 00 02 00 00 00 06 ff 06 17 70 00 01
[vestel] ERROR 2023/10/29 16:03:56 heartbeat: read tcp 192.178.66.112:40408->192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:04:26 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:04:36 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] TRACE 2023/10/29 16:04:56 modbus: closing connection due to idle timeout: 1m10.005151483s
[vestel] ERROR 2023/10/29 16:05:06 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:05:16 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:05:46 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:05:56 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
Power:          dial tcp 192.178.66.110:502: i/o timeout
Energy:         dial tcp 192.178.66.110:502: i/o timeout
Current L1..L3: dial tcp 192.178.66.110:502: i/o timeout
Voltage L1..L3: dial tcp 192.178.66.110:502: i/o timeout
Charge status:  dial tcp 192.178.66.110:502: i/o timeout
Enabled:        dial tcp 192.178.66.110:502: i/o timeout
Charged:        dial tcp 192.178.66.110:502: i/o timeout
Duration:       dial tcp 192.178.66.110:502: i/o timeout

**evcc charger --log trace -d**

[main  ] ERROR 2023/10/29 16:07:11 disable: dial tcp 192.178.66.110:502: i/o timeout

**evcc charger --log trace**

Vestel-EVC04
------------
[vestel] ERROR 2023/10/29 16:08:43 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:08:53 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:09:23 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:09:33 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:10:03 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:10:13 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:10:43 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:10:53 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
Power:          dial tcp 192.178.66.110:502: i/o timeout
Energy:         dial tcp 192.178.66.110:502: i/o timeout
Current L1..L3: dial tcp 192.178.66.110:502: i/o timeout
Voltage L1..L3: dial tcp 192.178.66.110:502: i/o timeout
Charge status:  dial tcp 192.178.66.110:502: i/o timeout
Enabled:        dial tcp 192.178.66.110:502: i/o timeout
Charged:        dial tcp 192.178.66.110:502: i/o timeout
Duration:       dial tcp 192.178.66.110:502: i/o timeout

**evcc charger --log trace -e**

[vestel] ERROR 2023/10/29 16:14:15 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:14:25 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[main  ] ERROR 2023/10/29 16:14:35 enable: dial tcp 192.178.66.110:502: i/o timeout

**evcc charger --log trace**

Vestel-EVC04
------------
[vestel] ERROR 2023/10/29 16:15:47 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:15:57 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:16:27 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:16:37 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:17:07 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:17:17 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:17:47 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
[vestel] ERROR 2023/10/29 16:17:57 heartbeat: dial tcp 192.178.66.110:502: i/o timeout
Power:          dial tcp 192.178.66.110:502: i/o timeout
Energy:         dial tcp 192.178.66.110:502: i/o timeout
Current L1..L3: dial tcp 192.178.66.110:502: i/o timeout
Voltage L1..L3: dial tcp 192.178.66.110:502: i/o timeout
Charge status:  dial tcp 192.178.66.110:502: i/o timeout
Enabled:        dial tcp 192.178.66.110:502: i/o timeout
Charged:        dial tcp 192.178.66.110:502: i/o timeout
Duration:       dial tcp 192.178.66.110:502: i/o timeout

andig, 2023-10-29:
Da es nur timeouts gibt nicht zu verwerten. evcc hast Du sicher vorher gestoppt?

ogheinz77, 2023-10-29:
Shit ne

ogheinz77, 2023-10-29:
Sorry, kommt gleich noch mal neu

ogheinz77, 2023-10-29:
So jetzt mit evcc stop und dann evcc charger --log trace

**evcc charger --log trace**

[main  ] INFO 2023/10/29 16:52:29 evcc 0.120.3
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 02 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 02 00 00 00 06 ff 06 17 70 00 01
[fritzdect] TRACE 2023/10/29 16:52:35 0
Power:         0W
Energy:        1415.1kWh
Charge status: B
Enabled:       false
Features:      [IntegratedDevice]

Vestel-EVC04
------------
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 03 00 00 00 06 ff 04 03 fc 00 02
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 03 00 00 00 07 ff 04 04 00 00 00 04
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 04 00 00 00 06 ff 04 04 0c 00 02
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 04 00 00 00 07 ff 04 04 00 00 0e 71
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 05 00 00 00 06 ff 04 03 f0 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 05 00 00 00 05 ff 04 02 00 31
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 06 00 00 00 06 ff 04 03 f2 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 06 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 07 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 07 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 08 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 09 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 09 00 00 00 05 ff 04 02 00 e6
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 0a 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 e8
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 0b 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 0b 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: send 00 0c 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/10/29 16:52:35 modbus: recv 00 0c 00 00 00 05 ff 03 02 00 06
[vestel] TRACE 2023/10/29 16:52:36 modbus: send 00 0d 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/10/29 16:52:36 modbus: recv 00 0d 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/10/29 16:52:36 modbus: send 00 0e 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/10/29 16:52:36 modbus: recv 00 0e 00 00 00 07 ff 04 04 00 00 00 00
Power:          4W
Energy:         369.7kWh
Current L1..L3: 0.049A 0.018A 0.018A
Voltage L1..L3: 231V 230V 232V
Charge status:  A
Enabled:        true
Charged:        0.0kWh
Duration:       0s

**evcc charger --log trace -d**

[main  ] INFO 2023/10/29 16:52:52 evcc 0.120.3
[main  ] INFO 2023/10/29 16:52:52 using config file: /etc/evcc.yaml
[vestel] TRACE 2023/10/29 16:52:55 modbus: send 00 01 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/29 16:52:55 modbus: recv 00 01 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/29 16:52:55 modbus: send 00 02 00 00 00 06 ff 06 13 8c 00 00
[vestel] TRACE 2023/10/29 16:52:55 modbus: recv 00 02 00 00 00 06 ff 06 13 8c 00 00

**evcc charger --log trace**

[main  ] INFO 2023/10/29 16:53:05 evcc 0.120.3
[main  ] INFO 2023/10/29 16:53:05 using config file: /etc/evcc.yaml
[vestel] TRACE 2023/10/29 16:53:08 modbus: send 00 01 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/29 16:53:08 modbus: recv 00 01 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/29 16:53:11 modbus: send 00 02 00 00 00 06 ff 06 17 70 00 01
[vestel] TRACE 2023/10/29 16:53:11 modbus: recv 00 02 00 00 00 06 ff 06 17 70 00 01

Vestel-EVC04
------------
[vestel] TRACE 2023/10/29 16:53:12 modbus: send 00 03 00 00 00 06 ff 04 03 fc 00 02
[... 59 lines cut ...]
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 06 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 07 00 00 00 06 ff 04 03 f4 00 01
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 07 00 00 00 05 ff 04 02 00 12
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 08 00 00 00 06 ff 04 03 f6 00 01
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 08 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 09 00 00 00 06 ff 04 03 f8 00 01
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 09 00 00 00 05 ff 04 02 00 e7
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 0a 00 00 00 06 ff 04 03 fa 00 01
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 0a 00 00 00 05 ff 04 02 00 e6
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 0b 00 00 00 06 ff 04 03 ec 00 01
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 0b 00 00 00 05 ff 04 02 00 01
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 0c 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 0c 00 00 00 05 ff 03 02 00 06
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 0d 00 00 00 06 ff 04 05 de 00 02
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 0d 00 00 00 07 ff 04 04 00 00 00 00
[vestel] TRACE 2023/10/29 16:53:46 modbus: send 00 0e 00 00 00 06 ff 04 05 e4 00 02
[vestel] TRACE 2023/10/29 16:53:46 modbus: recv 00 0e 00 00 00 07 ff 04 04 00 00 00 00
Power:          3W
Energy:         369.7kWh
Current L1..L3: 0.049A 0.018A 0.018A
Voltage L1..L3: 231V 231V 230V
Charge status:  A
Enabled:        true
Charged:        0.0kWh
Duration:       0s

ogheinz77, 2023-10-29:
Wenn nötig kann ich das auch noch mal mit evcc 0.121.5 machen.

andig, 2023-10-30:
Das Problem ist klar- die Box rückt den Status nicht wieder raus:

```
**evcc charger --log trace -d**

[vestel] TRACE 2023/10/29 16:52:55 modbus: send 00 02 00 00 00 06 ff 06 13 8c 00 00
[vestel] TRACE 2023/10/29 16:52:55 modbus: recv 00 02 00 00 00 06 ff 06 13 8c 00 00

```
aber

```
**evcc charger --log trace**

[vestel] TRACE 2023/10/29 16:53:12 modbus: send 00 0c 00 00 00 06 ff 03 13 8c 00 01
[vestel] TRACE 2023/10/29 16:53:12 modbus: recv 00 0c 00 00 00 05 ff 03 02 00 06
```

Vorschlag: frag mal den Vestel Support wie sich die Ladung unterbinden lässt und wie der Status der Ladefreigabe gelesen werden kann. So wie wir das grad machen geht es nicht.

Hört die Box denn überhaupt mit laden auf wenn evcc das versucht zu beenden???

ogheinz77, 2023-10-30:
_Hört die Box denn überhaupt mit laden auf wenn evcc das versucht zu beenden???_

--> Ja die Steuerung mit EVCC funktioniert tadellos, top!!

andig, 2023-10-30:
Ok- dann liegt das Problem evtl. nur im Rücklesen. Dafür könnte ich einen (hässlichen) Fix bereit stellen, Update kommt.

ogheinz77, 2023-10-30:
Ok, super danke.
Ich frage mal beim Support nach.
Was wäre denn die richtige Frage für den Support, etwa so?

Guten Tag,

ich steuere das Lastmanagement meiner Vestel EVC04 Wallbox via TCP Modbus und habe folgende Fragen dazu:

1. Mit welchen Register/Parameter lässt sich die Ladung im Modbus unterbinden?
2. Mit welchen Register/Parameter lässt sich der Status der Ladefreigabe auslesen?

Passt das so?

ogheinz77, 2023-10-30:
Hilft das hier?
[EVC04_Modbus_TCP_Specification.pdf](https://github.com/evcc-io/evcc/files/13204948/EVC04_Modbus_TCP_Specification.pdf)

andig, 2023-10-30:
Leider nein- da ist das Register 5004 r/w. Nur liefert es beim lesen eine 6 (A) nachdem man vorher eine 0 (=disabled) rein geschrieben hat. Das ist... nicht gut :)

ogheinz77, 2023-10-30:
Oh je, denke da kommt nix gescheites vom Support.
Aber ich frage mal nach, passen die Fragen von oben so?

ogheinz77, 2023-10-31:
Anfrage an den Support is raus, mal sehen was da zurückkommt.

closed 2023-10-31 by andig

ogheinz77, 2023-11-02:
@andig 
Hallo,
hab Antwort von Support bekommen:

_Sehr geehrter Herr,
vielen Dank für Ihre Anfrage. Am Anhang befindet sich das Modbus-Dokument, dort sind alle Register für unseren EVC04 aufgeführt. Bei Rückfragen stehen wir Ihnen gerne unter der Rufnummer 089-21129750 von Mo-Sa in der Zeit von 8-18 Uhr zur Verfügung._

Das PDF hat die Version 1.8, das alte PDF von mir war Version 1.5 geändert hat sich das hier:

[image]

[EVC04_Modbus_TCP_Specification v1.8.pdf](https://github.com/evcc-io/evcc/files/13239564/EVC04_Modbus_TCP_Specification.v1.8.pdf)

andig, 2023-11-02:
Die typisch sinnlose Antwort eines Supports der das Problem nicht versteht. Die Frage ist:

> Reg. 5004 ist R/W. Nach schreiben von 0 (=Laden beenden) führt ein lesen zum Wert 6 (=Laden mit Minimalstrom erlaubt).

Wie ist das möglich? Ohne die Antwort keine Lösung :(

ogheinz77, 2023-11-02:
Ich frage noch mal nach aber mit wenig Hoffnung auf Erfolg.

ogheinz77, 2023-11-02:
Hier die Nachfrage an den Support:

_Guten Tag,

vielen Dank für die Übersendung des PDFs. Die Frage die ich noch dazu habe ist:
Wenn man in den Modbus Reg. 5004 eine 0 (=Laden beenden) schreibt und den Register danach direkt wieder ausliest bekommt man den Wert 6 (=Laden mit Minimalstrom erlaubt) zurück.
Wie ist das möglich, wenn man den Reg mit 0 beschreibt?_

premultiply, 2023-11-02:
Das könnte auch passieren wenn aus unbekanntem Grund der Watchdog (mit 0s?) sofort zuschlägt und somit immer mit 6A geladen wird.

ogheinz77, 2023-11-06:
@andig

Hallo,

heute ist das hier nochmal vom Support gekommen:

Vielen Dank für Ihre Anfrage. Ladevorgang soll unterbunden werden.
Es ist generell so, dass das Fahrzeug nicht mit dem Laden beginnt, solange das Register 5004 auf einen Wert unter 6 Ampere eingestellt ist. Erst wenn das Register 5004 auf einen Wert über 6 Ampere gesetzt wird, beginnt das Fahrzeug zu laden
Unser EVC04 arbeitet mit mindestens 6 Ampere.
 
Bei Rückfragen stehen wir Ihnen gerne unter der Rufnummer 089-21129750 von Mo-Sa in der Zeit von 8-18 Uhr zur Verfügung.

premultiply, 2023-11-06:
Wenn der Watchdog zuschlägt fängt sie an wie konfiguriert zu laden.

andig, 2023-11-06:
> heute ist das hier nochmal vom Support gekommen:

Die Frage war ja "warum xy", nicht "bitte erzähl mir irgendwas zu einem anderen Thema". Schade, aber vollkommen irrelevant.