evcc-io/evcc #10528: Binary for ARMv5tel

christophhhh, 2023-10-28:
# Motivation
It makes sense to run this amazong software on hardware that is already available and running to avoid double energy consumpption

# My Problem
A NAS is clearly an obvious host for it. Currently you offer binarys for ARM64 and ARMv6. However I have a NAS with ARMv5tel that should have enough power to run this.

# The preferred solution
It would be great if you could provide a binary for ARMv5tel aswell.

# What I tried
I tried to compile it myself with my very limited knowledge about this, but only managed to compile for the architecture of the build system (AMD64 in that case).  I think this applies for a big part of the possible users. Compoiling directly on the NAS failed due to missing software there.

# Alternatives
If that is not an option, I would be grateful for any advice on how to cross compile evcc for ARMv5tel ;-)

# Additional Information
The NAS I'm trying to get this to run is a "QNAP TS-219P II" [https://www.qnap.com/de-de/product/ts-219p%20ii/specs/hardware](url)

andig, 2023-10-28:
@christophhhh probier mal bitte, das selbst zu bauen: https://github.com/evcc-io/evcc#contribute

Stattt `make` dann mal:

    GOOS=linux GOARCH=arm GOARM=5 make build

und mal das Binary ausprobieren. Evtl. brauchst Du auch noch softfloat?

christophhhh, 2023-10-28:
Hallo andig, herzlichen Dank damit konnte ich das für ARMv5 kompilieren. Bisher läuft es wunderbar auf dem alten NAS. Die Entscheidung die Architektur zu unterstützen oder nicht liegt natürlich bei dir. Jetzt muss ich mich halt selbst ums kompilieren kümmen wenn es ein Update geben soll und die Umgebung dazu bereit halten. Aber da ich mich nicht damit auskenne kann ich nicht enschätzen was der Aufwand wäre das von Seiten des Projekts zu unterstützen.

closed 2023-10-29 by andig