evcc-io/evcc #10540: Crash after update to 0.121.3

slewantoski, 2023-10-28:
### Describe the bug

After updating my evcc from 0.121.2 to 0.121.3 it instantly crashes on startup.
Same configuration works perfectly with the older versions.

### Steps to reproduce

1. Update evcc to 0.121.3
2. start evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.home.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

mqtt:
  broker: 192.168.0.70:1883
  topic: evcc
  clientid: evcc

meters:
- type: custom
  name: pv1
  power:
    source: mqtt
    topic: pvpower/pv
- type: custom
  name: grid1
  power:
    source: mqtt
    topic: pvpower/grid
- type: custom
  name: battery1
  power:
    source: mqtt
    topic: pvpower/battery
    scale: -1
  soc:
    source: mqtt
    topic: pvpower/batterysoc

chargers:
- type: template
  template: abl
  modbus: rs485tcpip
  id: 1
  host: 192.168.0.7
  port: 502
  timeout: 10s
  name: wallbox1

loadpoints:
- title: Garage
  charger: wallbox1
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Zuhause
  residualPower: 200
  prioritySoc: 70
  meters:
    grid: grid1
    pv: pv1
    battery: battery1

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.40 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1283 # [currency]/kWh

vehicles:
  - name: ev6
    type: template
    template: offline
    title: EV6
    capacity: 78
```

### Log details

```text
[main  ] INFO 2023/10/28 21:06:58 evcc 0.121.3
[main  ] INFO 2023/10/28 21:06:58 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/28 21:06:58 starting ui and api at :7070
[db    ] INFO 2023/10/28 21:06:58 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/10/28 21:06:58 connecting evcc at tcp://192.168.0.70:1883
[mqtt  ] DEBUG 2023/10/28 21:06:58 tcp://192.168.0.70:1883 connected
[site  ] INFO 2023/10/28 21:07:08 site config:
[site  ] INFO 2023/10/28 21:07:08   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/28 21:07:08     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/28 21:07:08     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/28 21:07:08     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/28 21:07:08   vehicles:
[site  ] INFO 2023/10/28 21:07:08     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/28 21:07:08 loadpoint 1:
[lp-1  ] INFO 2023/10/28 21:07:08   mode:        off
[lp-1  ] INFO 2023/10/28 21:07:08   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/10/28 21:07:08   meters:      charge ✗
[lp-1  ] DEBUG 2023/10/28 21:07:08 phase timer inactive
[lp-1  ] DEBUG 2023/10/28 21:07:08 pv timer inactive
[lp-1  ] DEBUG 2023/10/28 21:07:08 guard timer inactive
[site  ] WARN 2023/10/28 21:07:08 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/28 21:07:08 ----
[lp-1  ] DEBUG 2023/10/28 21:07:08 charge power: 0W
[site  ] DEBUG 2023/10/28 21:07:08 pv power: 0W
[site  ] DEBUG 2023/10/28 21:07:08 battery soc: 0%
[site  ] DEBUG 2023/10/28 21:07:08 battery power: 0W
[site  ] DEBUG 2023/10/28 21:07:08 grid power: 607W
[site  ] DEBUG 2023/10/28 21:07:08 site power: 807W
[lp-1  ] DEBUG 2023/10/28 21:07:08 charger status: A
[lp-1  ] INFO 2023/10/28 21:07:08 car disconnected
[lp-1  ] DEBUG 2023/10/28 21:07:08 set charge mode: off
[lp-1  ] DEBUG 2023/10/28 21:07:08 set min current: 6
[lp-1  ] DEBUG 2023/10/28 21:07:08 set max current: 16
[lp-1  ] DEBUG 2023/10/28 21:07:08 set priority: 0
panic: reflect.Value.Interface: cannot return value obtained from unexported field or method

goroutine 43 [running]:
reflect.valueInterface({0x1c564a0?, 0x4000496288?, 0x4000e0bb10?}, 0x2?)
        reflect/value.go:1501 +0xc8
reflect.Value.Interface(...)
        reflect/value.go:1490
github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x1c?, {0x400074e180, 0x1c}, 0x2?, {0x2110580, 0x4000496288})
        github.com/evcc-io/evcc/server/mqtt.go:97 +0x390
github.com/evcc-io/evcc/server.(*MQTT).publish(0x2183edd?, {0x400074e180?, 0x48996e0?}, 0x20?, {0x2110580?, 0x4000496288?})
        github.com/evcc-io/evcc/server/mqtt.go:123 +0x84
github.com/evcc-io/evcc/server.(*MQTT).Run(0x4000cc2540, {0x2bba440?, 0x4000a34600}, 0x4000808000?)
        github.com/evcc-io/evcc/server/mqtt.go:306 +0xabc
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
        github.com/evcc-io/evcc/cmd/root.go:202 +0xcbc
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.3

normanmaurer, 2023-10-28:
Same problem here... same is also true with 0.121.4

TobiasSt78, 2023-10-28:
I have the same problem.. with 0.121.3 and 0.121.4.
The downgrade to 0.121.2 works fine again.

Failure:
```
...
[lp-1  ] DEBUG 2023/10/28 23:54:13 vehicle odometer: 84147km
[lp-1  ] DEBUG 2023/10/28 23:54:13 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/28 23:54:13 charger status: A
[lp-1  ] INFO 2023/10/28 23:54:13 car disconnected
[lp-1  ] DEBUG 2023/10/28 23:54:13 set charge mode: pv
[lp-1  ] DEBUG 2023/10/28 23:54:13 set target soc: 85
[lp-1  ] DEBUG 2023/10/28 23:54:13 set max current: 16
panic: reflect.Value.Interface: cannot return value obtained from unexported field or method

goroutine 293 [running]:
reflect.valueInterface({0x1c564a0?, 0x40016ba108?, 0x4001bb7b10?}, 0x2?)
        reflect/value.go:1501 +0xc8
reflect.Value.Interface(...)
        reflect/value.go:1490
github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x1c?, {0x4001f44160, 0x1c}, 0x2?, {0x2110580, 0x40016ba108})
        github.com/evcc-io/evcc/server/mqtt.go:97 +0x390
github.com/evcc-io/evcc/server.(*MQTT).publish(0x2183edd?, {0x4001f44160?, 0x48996e0?}, 0x20?, {0x2110580?, 0x40016ba108?})
        github.com/evcc-io/evcc/server/mqtt.go:123 +0x84
github.com/evcc-io/evcc/server.(*MQTT).Run(0x4000912020, {0x2bba440?, 0x40018ed800}, 0x0?)
        github.com/evcc-io/evcc/server/mqtt.go:306 +0xabc
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
        github.com/evcc-io/evcc/cmd/root.go:202 +
```

bekovan, 2023-10-28:
Disabling MQTT solved the problem for me

andig, 2023-10-29:
0.120.4 released

closed 2023-10-29 by andig

snh, 2023-10-29:
I'm still seeing this problem with 0.120.4.

```
[main  ] INFO 2023/10/29 11:36:45 evcc 0.121.4
...
panic: reflect.Value.Interface: cannot return value obtained from unexported field or method

goroutine 578 [running]:
reflect.valueInterface({0x1c564a0?, 0x40010e84c8?, 0x4000e67b10?}, 0x2?)
	reflect/value.go:1501 +0xc8
reflect.Value.Interface(...)
	reflect/value.go:1490
github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x1c?, {0x40014c5e00, 0x1c}, 0x2?, {0x2110580, 0x40010e84c8})
	github.com/evcc-io/evcc/server/mqtt.go:97 +0x390
github.com/evcc-io/evcc/server.(*MQTT).publish(0x2183eea?, {0x40014c5e00?, 0x4899740?}, 0xa0?, {0x2110580?, 0x40010e84c8?})
	github.com/evcc-io/evcc/server/mqtt.go:123 +0x84
github.com/evcc-io/evcc/server.(*MQTT).Run(0x4000bc11c0, {0x2bba4c0?, 0x40012bf000}, 0x0?)
	github.com/evcc-io/evcc/server/mqtt.go:306 +0xabc
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:202 +0xcbc
```

thorsten-gehrig, 2023-10-29:
Same problem here - with 0.121.3 and 0.121.4 (just tested). Must return to 0.121.2 ... I've using MQTT as well...

slewantoski, 2023-10-29:
Not fixed with 0.121.4.
Same Problem:
`[main  ] INFO 2023/10/29 08:42:11 evcc 0.121.4
[main  ] INFO 2023/10/29 08:42:11 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/29 08:42:11 starting ui and api at :7070
[db    ] INFO 2023/10/29 08:42:12 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/10/29 08:42:12 connecting evcc at tcp://192.168.0.70:1883
[mqtt  ] DEBUG 2023/10/29 08:42:12 tcp://192.168.0.70:1883 connected
[mqtt  ] DEBUG 2023/10/29 08:42:12 tcp://192.168.0.70:1883 subscribe pvpower/pv
[site  ] INFO 2023/10/29 08:42:22 site config:
[site  ] INFO 2023/10/29 08:42:22   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/29 08:42:22     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/29 08:42:22     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/29 08:42:22     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/29 08:42:22   vehicles:
[site  ] INFO 2023/10/29 08:42:22     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/29 08:42:22 loadpoint 1:
[lp-1  ] INFO 2023/10/29 08:42:22   mode:        off
[lp-1  ] INFO 2023/10/29 08:42:22   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/10/29 08:42:22   meters:      charge ✗
[lp-1  ] DEBUG 2023/10/29 08:42:22 phase timer inactive
[lp-1  ] DEBUG 2023/10/29 08:42:22 pv timer inactive
[lp-1  ] DEBUG 2023/10/29 08:42:22 guard timer inactive
[site  ] WARN 2023/10/29 08:42:22 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/29 08:42:22 ----
[lp-1  ] DEBUG 2023/10/29 08:42:22 charge power: 0W
[site  ] DEBUG 2023/10/29 08:42:22 pv power: 161W
[site  ] DEBUG 2023/10/29 08:42:22 battery soc: 0%
[site  ] DEBUG 2023/10/29 08:42:22 battery power: 1W
[site  ] DEBUG 2023/10/29 08:42:22 grid power: 505W
[site  ] DEBUG 2023/10/29 08:42:22 site power: 706W
[lp-1  ] DEBUG 2023/10/29 08:42:22 charger status: A
[lp-1  ] INFO 2023/10/29 08:42:22 car disconnected
[lp-1  ] DEBUG 2023/10/29 08:42:22 set charge mode: off
[lp-1  ] DEBUG 2023/10/29 08:42:22 set min current: 6
[lp-1  ] DEBUG 2023/10/29 08:42:22 set max current: 16
[lp-1  ] DEBUG 2023/10/29 08:42:22 set priority: 0
panic: reflect.Value.Interface: cannot return value obtained from unexported field or method

goroutine 47 [running]:
reflect.valueInterface({0x1c564a0?, 0x40009d5200?, 0x4000d57b10?}, 0x2?)
        reflect/value.go:1501 +0xc8
reflect.Value.Interface(...)
        reflect/value.go:1490
github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x1c?, {0x400157a100, 0x1c}, 0x2?, {0x2110580, 0x40009d5200})
        github.com/evcc-io/evcc/server/mqtt.go:97 +0x390
github.com/evcc-io/evcc/server.(*MQTT).publish(0x2183eea?, {0x400157a100?, 0x4899740?}, 0xa0?, {0x2110580?, 0x40009d5200?})
        github.com/evcc-io/evcc/server/mqtt.go:123 +0x84
github.com/evcc-io/evcc/server.(*MQTT).Run(0x40014e8980, {0x2bba4c0?, 0x40014be400}, 0x40011b2000?)
        github.com/evcc-io/evcc/server/mqtt.go:306 +0xabc
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
        github.com/evcc-io/evcc/cmd/root.go:202 +0xcbc
`

TobiasSt78, 2023-10-31:
Fixed with 0.121.5. 
THANK YOU!