evcc-io/evcc #10541: out of sync

wb3103, 2023-10-28:
Nach oder während des Verbindungsaufbaus (anstecken Ladekabel) erscheint die folgende Fehlermeldung:

charger out of sync: expected disabled, got enabled

Das sieht so aus, als ob die Box im Moment der Verbindungsaufnahme ein Verbindungssignal erhält, obwohl dieses noch nicht erforderlich, also zu früh ist/kommt.
Es gibt aber keine Auswirkungen in der weiteren Arbeitsweise.
Hardware: VW ID3, Pulsares simple box

andig, 2023-10-29:
Logfile fehlt

closed 2023-10-29 by andig

wb3103, 2023-10-29:
yournalctl

Oct 21 13:29:22 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:22 charge power: 0W
Oct 21 13:29:23 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:23 pv power: 1738W
Oct 21 13:29:23 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:23 battery soc: 0%
Oct 21 13:29:23 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:23 battery power: 0W
Oct 21 13:29:23 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:23 grid power: -1514W
Oct 21 13:29:23 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:23 grid powers: [-351 -845 -318]W
Oct 21 13:29:23 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:23 grid currents: [-1.57 -3.67 -1.39]A
Oct 21 13:29:23 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:23 site power: -1514W
Oct 21 13:29:23 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:23 charge voltages: [231 229 231]V
Oct 21 13:29:23 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:23 detected connected phases: 3p
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 charge currents: [0.03 0.03 0.03]A
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 charge total import: 31.417kWh
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 charger status: B
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 pv charge current: 6.58A = 0A + 6.58A (-1514W @ 1p)
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 site power -1514W <= 0W enable threshold
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 pv enable in 0s
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 pv enable timer elapsed
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 max charge current: 6A
Oct 21 13:29:24 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:24 guard enable in 2m47s
Oct 21 13:29:28 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:28 ----
Oct 21 13:29:28 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:28 charge power: -25W
Oct 21 13:29:28 evcc1 evcc[757]: [lp-1  ] WARN 2023/10/21 13:29:28 charge power must not be negative: -25
Oct 21 13:29:28 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:28 pv power: 1707W
Oct 21 13:29:29 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:29 battery soc: 0%
Oct 21 13:29:29 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:29 battery power: 0W
Oct 21 13:29:29 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:29 grid power: -77W
Oct 21 13:29:29 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:29 grid powers: [-339 579 -317]W
Oct 21 13:29:29 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:29 grid currents: [-1.52 2.86 -1.39]A
Oct 21 13:29:29 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:29 site power: -77W
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 charge voltages: [229 228 231]V
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 detected connected phases: 3p
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 charge currents: [0.03 12.5 0.03]A
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 charge total import: 31.417kWh
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 charger status: C
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 charge start energy: 31.4kWh

Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] INFO 2023/10/21 13:29:30 start charging ->
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 wake-up timer: stop
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 soc estimated: 49.12% (vehicle: 49.00%)
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 vehicle soc: 49%
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 vehicle soc limit: 80%
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 vehicle range: 270km

Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] WARN 2023/10/21 13:29:30 charger out of sync: expected disabled, got enabled

Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 charger: guard elapse
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 guard timer inactive
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 pv charge current: 6.34A = 6A + 0.336A (-77W @ 1p)
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 pv timer reset
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 pv timer inactive
Oct 21 13:29:30 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:30 max charge current: 6.34A
Oct 21 13:29:34 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:34 ----
Oct 21 13:29:34 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:34 charge power: 1082W
Oct 21 13:29:34 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:34 pv power: 1629W
Oct 21 13:29:35 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:35 battery soc: 0%
Oct 21 13:29:35 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:35 battery power: 0W
Oct 21 13:29:35 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:35 grid power: -21W
Oct 21 13:29:35 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:35 grid powers: [-325 605 -300]W
Oct 21 13:29:35 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:35 grid currents: [-1.45 2.9 -1.34]A
Oct 21 13:29:35 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:35 site power: -21W
Oct 21 13:29:35 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:35 charge voltages: [229 229 231]V
Oct 21 13:29:35 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:35 detected connected phases: 3p
[... cut ...]
Oct 21 13:29:35 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:35 vehicle soc: 49%
Oct 21 13:29:35 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:35 vehicle soc limit: 80%
Oct 21 13:29:35 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:35 vehicle range: 270km
Oct 21 13:29:35 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:35 pv charge current: 6.43A = 6.34A + 0.0915A (-21W @ 1p)
Oct 21 13:29:35 evcc1 evcc[757]: [lp-1  ] DEBUG 2023/10/21 13:29:35 max charge current: 6.43A
Oct 21 13:29:40 evcc1 evcc[757]: [site  ] DEBUG 2023/10/21 13:29:40 ----