evcc-io/evcc #10542: Amperfied Wallbox connect.business:  modbus: exception '2'

dnkh, 2023-10-28:
### Describe the bug

Beim hinzufügen meiner Amperfied Wallbox connect.business Wallbox zur evcc Konfiguration bekomme ich folgende Fehlermeldung:
Fehler: cannot create charger 'heidelberg': modbus: exception '2' (illegal data address), function '16'

### Steps to reproduce

1. In der Powershell unter Windows gebe ich den Befehl: _.\evcc.exe configure_ ein
2. Ich wähle den _Fortgeschrittenen Modus_ aus
3. Dann Wallbox und und anschließend die Wallbox _Amperfied Wallbox connect.business_
4. Anschließend die folgende Konfiguration
 
 Modbus ID 255
 IP-Adresse oder Hostname [? for help] 10.0.0.60
 IP-Adresse oder Hostname 10.0.0.60
 Port 502

6. Dann wird der Test von evcc durchgeführt:

Teste die Konfiguration von Amperfied Wallbox connect.business ...
Fehler: cannot create charger 'heidelberg': modbus: exception '2' (illegal data address), function '16'
Der Test von Amperfied Wallbox connect.business ist fehlgeschlagen.

Als Eintrag für die evcc.yaml wird folgende Konfiguration erstellt

Die Konfiguration lautet:

  type: template
  template: amperfied
  id: 255
  host: 10.0.0.60
  port: 502
  modbus: tcpip
  name: wallbox1

### Configuration details

```yaml
chargers:
- type: template
  template: amperfied
  id: 255
  host: 10.0.0.60
  port: 502
  modbus: tcpip
  name: wallbox1
```

### Log details

```text
.\evcc.exe --log trace                                                                                                        22:09:56
[main  ] INFO 2023/10/28 22:11:04 evcc 0.121.2
[main  ] INFO 2023/10/28 22:11:04 using config file: C:\evcc\evcc.yaml
[main  ] INFO 2023/10/28 22:11:04 starting ui and api at :7070
[db    ] INFO 2023/10/28 22:11:04 using sqlite database: C:\Users\chris\.evcc\evcc.db
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT * FROM `settings` 10 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%       device_id       %") -1 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2023/10/28 22:11:04 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[heidel] TRACE 2023/10/28 22:11:04 modbus: send 00 01 00 00 00 09 ff 10 01 02 00 01 02 00 04
[heidel] TRACE 2023/10/28 22:11:04 modbus: recv 00 01 00 00 00 03 ff 90 02
[main  ] FATAL 2023/10/28 22:11:04 cannot create charger 'wallbox1': cannot create charger 'template': cannot create charger 'heidelberg': modbus: exception '2' (illegal data address), function '16'
[main  ] FATAL 2023/10/28 22:11:04 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Windows

### Version

evcc version 0.121.2

premultiply, 2023-10-29:
Was sagt denn `evcc charger --log trace`?

dnkh, 2023-10-29:
Leider nicht mehr als das evcc --log trace von oben

**.\evcc.exe charger --log trace**                                                                                                09:36:38
[main  ] INFO 2023/10/29 09:37:45 evcc 0.121.2
[main  ] INFO 2023/10/29 09:37:45 using config file: C:\evcc\evcc.yaml
[db    ] INFO 2023/10/29 09:37:46 using sqlite database: C:\Users\chris\.evcc\evcc.db
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT * FROM `settings` 10 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%       device_id       %") -1 <nil>
[db    ] TRACE 2023/10/29 09:37:46 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[heidel] TRACE 2023/10/29 09:37:46 modbus: send 00 01 00 00 00 09 ff 10 01 02 00 01 02 00 04
[heidel] TRACE 2023/10/29 09:37:46 modbus: recv 00 01 00 00 00 03 ff 90 02
[main  ] FATAL 2023/10/29 09:37:46 cannot create charger 'wallbox1': cannot create charger 'template': cannot create charger 'heidelberg': modbus: exception '2' (illegal data address), function '16'

dnkh, 2023-10-29:
Wenn als Fehler "function 16" kommt, ist die 16 dann die Bus Adresse die abgefragt werden soll ?

premultiply, 2023-10-29:
Nein, das bezieht sich auf die Modbus-Operation, s.g. Function Code.
16 ist Write multiple holding registers.

premultiply, 2023-10-29:
Ich sehe aber schon das Problem.
Braucht eine neue Implementierung.
Kommt...

andig, 2023-10-29:
@dnkh alte Version funktioniert? Hättest Du nochmal Log im Vergleich?
@premultiply hier Register wieder einzeln lösen? Dann würde ich das zurück bauen.

dnkh, 2023-10-30:
HI @andig,
das ging mal, da bin ich mir sicher.
Hab dann eine Weile nichts mit evcc gemacht weil ich es in HomeAssistent nutzen möchte.
Jetzt war ich soweit und dann hab ich diese Fehlermeldung bekommen.
Die Fehlermeldung ist die gleiche ob ich evcc Standalone nutze oder in HomeAssistant.

Ich hab eine "ältere"  Version von evcc nochmal genommen, damit geht es leider auch nicht mehr.
Inzwischen hat sich allerdings auch die Firmware der Wallbox aktualisiert, das könnte auch eine Grund sein

premultiply, 2023-10-30:
In der (neuen?) Firmware fehlt jedenfalls die Standby-Funktion der seriellen Boxen.
Daher fällt die Implementierung gleich auf die Nase.

andig, 2023-10-30:
@premultiply hatten wir nicht Kontakt zu Amperified? Vielleicht da mal anfragen?

premultiply, 2023-10-30:
Das Problem ist schon längst gelöst?
Siehe PR.

andig, 2023-10-30:
> Der letzte Commit ist 17h alt, Dein Kommentar aber 44 Minuten. Das kann ich wirklich nicht hellsehen :)

Der letzte Commit ist 17h alt, Dein Kommentar aber 47 Minuten. Das kann ich wirklich nicht hellsehen das der schon veraltet war :)

dnkh, 2023-10-30:
@andig @premultiply 
Pingt mich gerne an wenn ich was testen soll

premultiply, 2023-10-30:
@dnkh Ping

dnkh, 2023-10-30:
@premultiply  Habt ihr da schon eine Prebuild Version oder baue ich mit die Version selbst oder gibt es einen anderen Weg das zu testen ?

andig, 2023-10-30:
PR auschecken und gem. README bauen. Ist kein Hexenwerk. Oder wir mergen es einfach ins Nightly...

dnkh, 2023-10-30:
Bin noch bei der Arbeit, mach ich dann heute Abend und geb euch Feedback

closed 2023-10-30 by andig

dnkh, 2023-10-31:
@andig @premultiply 
Ich hab den Nightly Build in docker laufen und bis jetzt sieht das gut aus.
Nur der Identifier sieht irgendwie komisch aus: g�zt�

Die Wallbox wurde über eine RFID Karte freigeschaltet.

premultiply, 2023-10-31:
Danke für das Feedback!

Kannst du mal die Ausgabe von `evcc charger --log trace` zeigen?

Der Identifier ist dann wohl direkt binär und wir müssen den als Hex-String darstellen.
Ging aus der Doku für mich nicht so klar hervor wie das genau ist.

dnkh, 2023-10-31:
@premultiply gerne, ist ja auch für euch schwierig das zu testen
Hier das Log:

/ # evcc charger --log trace
[main  ] INFO 2023/10/31 09:10:39 evcc 0.121.5 (f0b25967)
[main  ] INFO 2023/10/31 09:10:39 using config file: /etc/evcc.yaml
[db    ] INFO 2023/10/31 09:10:39 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[db    ] TRACE 2023/10/31 09:10:39 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 01 00 00 00 06 ff 03 01 01 00 01
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 01 00 00 00 05 ff 03 02 3a 98
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 02 00 00 00 06 ff 04 00 0e 00 01
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 02 00 00 00 05 ff 04 02 00 00
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 03 00 00 00 06 ff 04 00 11 00 02
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 03 00 00 00 07 ff 04 04 00 2f a3 ce
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 04 00 00 00 06 ff 04 00 06 00 03
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 04 00 00 00 09 ff 04 06 00 00 00 00 00 00
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 05 00 00 00 06 ff 04 00 0a 00 03
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 05 00 00 00 09 ff 04 06 00 e9 00 ea 00 eb
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 06 00 00 00 06 ff 04 00 05 00 01
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 06 00 00 00 05 ff 04 02 00 04
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 07 00 00 00 06 ff 03 01 05 00 01
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 07 00 00 00 05 ff 03 02 00 00
[amperfied] TRACE 2023/10/31 09:10:39 modbus: send 00 08 00 00 00 06 ff 04 07 d2 00 06
[amperfied] TRACE 2023/10/31 09:10:39 modbus: recv 00 08 00 00 00 0f ff 04 0c 04 0f 67 92 7a 74 80 00 00 00 00 00
Power:          0W
Energy:         3122.1kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 233V 234V 235V
Charge status:  B
Enabled:        false
Identifier:     g�zt�