evcc-io/evcc #10549: 0.121.4 crash at startup

djfanatix, 2023-10-29:
### Describe the bug

Since the update to 0.121.4, EVCC doesn't start anymore.
See log below

### Steps to reproduce

1. Start EVCC

### Configuration details

```yaml
EVCC Runs as Home Assistant Add-on
```

### Log details

```text
[lp-2  ] DEBUG 2023/10/29 08:16:06 set max current: 16
[lp-2  ] DEBUG 2023/10/29 08:16:06 set priority: 0
panic: reflect.Value.Interface: cannot return value obtained from unexported field or method

goroutine 154 [running]:
reflect.valueInterface({0x2253580?, 0xc000b56de0?, 0xc001553b88?}, 0x2?)
	reflect/value.go:1501 +0xc6
reflect.Value.Interface(...)
	reflect/value.go:1490
github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x1c?, {0xc001527120, 0x1c}, 0x2?, {0x270d8e0, 0xc000b56de0})
	github.com/evcc-io/evcc/server/mqtt.go:97 +0x4e9
github.com/evcc-io/evcc/server.(*MQTT).publish(0x27811af?, {0xc001527120?, 0x4ed6620?}, 0x20?, {0x270d8e0?, 0xc000b56de0?})
	github.com/evcc-io/evcc/server/mqtt.go:123 +0x96
github.com/evcc-io/evcc/server.(*MQTT).Run(0xc00138ff00, {0x31b7c40?, 0xc0008aa000}, 0xc0002a1b80?)
	github.com/evcc-io/evcc/server/mqtt.go:306 +0xd67
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:202 +0xe69
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.4

slewantoski, 2023-10-29:
Same Problem on my side:

```text
[main  ] INFO 2023/10/29 08:42:11 evcc 0.121.4
[main  ] INFO 2023/10/29 08:42:11 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/29 08:42:11 starting ui and api at :7070
[db    ] INFO 2023/10/29 08:42:12 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/10/29 08:42:12 connecting evcc at tcp://192.168.0.70:1883
[mqtt  ] DEBUG 2023/10/29 08:42:12 tcp://192.168.0.70:1883 connected
[mqtt  ] DEBUG 2023/10/29 08:42:12 tcp://192.168.0.70:1883 subscribe pvpower/pv
[site  ] INFO 2023/10/29 08:42:22 site config:
[site  ] INFO 2023/10/29 08:42:22   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/29 08:42:22     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/29 08:42:22     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/29 08:42:22     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/29 08:42:22   vehicles:
[site  ] INFO 2023/10/29 08:42:22     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/29 08:42:22 loadpoint 1:
[lp-1  ] INFO 2023/10/29 08:42:22   mode:        off
[lp-1  ] INFO 2023/10/29 08:42:22   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/10/29 08:42:22   meters:      charge ✗
[lp-1  ] DEBUG 2023/10/29 08:42:22 phase timer inactive
[lp-1  ] DEBUG 2023/10/29 08:42:22 pv timer inactive
[lp-1  ] DEBUG 2023/10/29 08:42:22 guard timer inactive
[site  ] WARN 2023/10/29 08:42:22 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/29 08:42:22 ----
[lp-1  ] DEBUG 2023/10/29 08:42:22 charge power: 0W
[site  ] DEBUG 2023/10/29 08:42:22 pv power: 161W
[site  ] DEBUG 2023/10/29 08:42:22 battery soc: 0%
[site  ] DEBUG 2023/10/29 08:42:22 battery power: 1W
[site  ] DEBUG 2023/10/29 08:42:22 grid power: 505W
[site  ] DEBUG 2023/10/29 08:42:22 site power: 706W
[lp-1  ] DEBUG 2023/10/29 08:42:22 charger status: A
[lp-1  ] INFO 2023/10/29 08:42:22 car disconnected
[lp-1  ] DEBUG 2023/10/29 08:42:22 set charge mode: off
[lp-1  ] DEBUG 2023/10/29 08:42:22 set min current: 6
[lp-1  ] DEBUG 2023/10/29 08:42:22 set max current: 16
[lp-1  ] DEBUG 2023/10/29 08:42:22 set priority: 0
panic: reflect.Value.Interface: cannot return value obtained from unexported field or method

goroutine 47 [running]:
reflect.valueInterface({0x1c564a0?, 0x40009d5200?, 0x4000d57b10?}, 0x2?)
        reflect/value.go:1501 +0xc8
reflect.Value.Interface(...)
        reflect/value.go:1490
github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x1c?, {0x400157a100, 0x1c}, 0x2?, {0x2110580, 0x40009d5200})
        github.com/evcc-io/evcc/server/mqtt.go:97 +0x390
github.com/evcc-io/evcc/server.(*MQTT).publish(0x2183eea?, {0x400157a100?, 0x4899740?}, 0xa0?, {0x2110580?, 0x40009d5200?})
        github.com/evcc-io/evcc/server/mqtt.go:123 +0x84
github.com/evcc-io/evcc/server.(*MQTT).Run(0x40014e8980, {0x2bba4c0?, 0x40014be400}, 0x40011b2000?)
        github.com/evcc-io/evcc/server/mqtt.go:306 +0xabc
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
        github.com/evcc-io/evcc/cmd/root.go:202 +0xcbc
```

See my bug report #10540

Macrisu, 2023-10-29:
0.121.4 and 0.121.3 is affected

0.121.2 working

Sebreiter, 2023-10-29:
Das gleiche bei mir

[site  ] WARN 2023/10/29 09:20:25 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/10/29 09:20:25 ----
[lp-1  ] DEBUG 2023/10/29 09:20:25 charge power: 0W
[lp-2  ] DEBUG 2023/10/29 09:20:25 charge power: 0W
[site  ] DEBUG 2023/10/29 09:20:26 pv power: 2291W
[site  ] DEBUG 2023/10/29 09:20:26 battery soc: 59%
[site  ] DEBUG 2023/10/29 09:20:26 battery power: -1088W
[site  ] DEBUG 2023/10/29 09:20:26 grid power: 40W
[site  ] DEBUG 2023/10/29 09:20:26 battery has priority at soc 59% (< 95%)
[site  ] DEBUG 2023/10/29 09:20:26 site power: 140W
[lp-1  ] DEBUG 2023/10/29 09:20:26 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/29 09:20:26 charge total import: 7433.272kWh
[lp-1  ] DEBUG 2023/10/29 09:20:26 charger status: A
[lp-1  ] INFO 2023/10/29 09:20:26 car disconnected
[lp-1  ] DEBUG 2023/10/29 09:20:26 set charge mode: pv
[lp-1  ] DEBUG 2023/10/29 09:20:26 set min current: 6
[lp-1  ] DEBUG 2023/10/29 09:20:26 set max current: 16
[lp-1  ] DEBUG 2023/10/29 09:20:26 set priority: 8
panic: reflect.Value.Interface: cannot return value obtained from unexported field or method

goroutine 245 [running]:
reflect.valueInterface({0x2253580?, 0xc000c0ef18?, 0xc000d65b88?}, 0x2?)
	reflect/value.go:1501 +0xc6
reflect.Value.Interface(...)
	reflect/value.go:1490
github.com/evcc-io/evcc/server.(*MQTT).publishComplex(0x1c?, {0xc0013ea520, 0x1c}, 0x2?, {0x270d8e0, 0xc000c0ef18})
	github.com/evcc-io/evcc/server/mqtt.go:97 +0x4e9
github.com/evcc-io/evcc/server.(*MQTT).publish(0x27811af?, {0xc0013ea520?, 0x4ed6620?}, 0x20?, {0x270d8e0?, 0xc000c0ef18?})
	github.com/evcc-io/evcc/server/mqtt.go:123 +0x96
github.com/evcc-io/evcc/server.(*MQTT).Run(0xc000b779e0, {0x31b7c40?, 0xc0009b5400}, 0x0?)
	github.com/evcc-io/evcc/server/mqtt.go:306 +0xd67
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:202 +0xe69

MichaelWitkenkamp, 2023-10-29:
Moin,
Ich habe das gleiche Problem. Sowohl die 0.121.3, wie auch die 0.121.4 schmeißen den Fehler raus.
Ein downgrade auf 0.121.2 gemacht und alles läuft wieder normal.

deluxe19841, 2023-10-29:
@MichaelWitkenkamp Bei mir genau die gleiche Sitation. Ebenfalls Downgrade auf 0.121.2 vorgenommen.

andig, 2023-10-29:
Fixed in .5

closed 2023-10-29 by andig