evcc-io/evcc #10554: Ständige timeout Fehler Openwb 2.0

atze09, 2023-10-29:
### Describe the bug

Nach Umstellung von openwb 1.9 auf 2.0 bekomme ich nun ständig timeout Fehler und die Anbindung funktioniert nicht zuverlässig. An der Topologie wurde nichts verändert. Die auch per WLAN angebundene goe in gleichem Abstand funktioniert einwandfrei.

Ein Ping ist unauffällig.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
type: template
template: openwb-2.0
modbus: tcpip
id: 1
host: 192.168.17.80 # Hostname
port: 1502 # Port
```

### Log details

```text
[lp-2  ] ERROR 2023/10/29 12:31:40 charger: dial tcp 192.168.17.80:1502: i/o timeout
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.121.5

andig, 2023-10-29:
Da antwortet nichts- die Frage geht sn den OpenWB Support :/

closed 2023-10-29 by andig

atze09, 2023-10-29:
> Da antwortet nichts- die Frage geht an den OpenWB Support :/

Ja da war ich das Problem. Hatte unter secondary den zweiten Haken zum aktivieren von Morbus nicht gesetzt.

andig, 2023-10-29:
Gut zu wissen, danke!