evcc-io/evcc #10562: Keine Aktualisierung „Auswertung Ladeenergie“ mehr

An-Wi, 2023-10-29:
### Describe the bug

„Meine Daten“ werden während des Ladevorgangs und beim Abschluss nicht mehr aktualisiert. Das ging bis vor Kurzem noch.
Erst nach Neustart von evcc ist die Statistik aktuell.

### Steps to reproduce

1. Laden mit „Min+PV“
2. Vorgang stoppen durch Abziehen am Auto
3. Ladevorgang in Anzeige „Ladevorgänge“ vorhanden (Summe = 600,4 kWh)
[image]

4. Anzeige „Auswertung Ladeenergie“ (48 + 528 = 576 kWh)
[image]

5. Neustart evcc
6. Anzeige „Auswertung Ladeenergie“ (50 + 550 = 600 kWh -> passt jetzt zur Summe Ladevorgänge)
[image]

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Windows

### Version

0.121.4

naltatis, 2023-10-29:
Da es in dem Dialog um eher Langzeitdaten geht haben wir die Aktualisierung auf 30min eingestellt. Berücksichtigt werden abgeschlossene Ladevorgänge.

An-Wi, 2023-10-29:
> Da es in dem Dialog um eher Langzeitdaten geht haben wir die Aktualisierung auf 30min eingestellt. Berücksichtigt werden abgeschlossene Ladevorgänge.

Ah, ok. Danke für die Info. Werde es dahingehend mal beobachten...
Macht mit deiner Erklärung auch ein gewissen Sinn. 👍

naltatis, 2023-10-30:
@andig könnten wir, dann sollten wir die offenen Vorgänge aber regelmäßig persistieren. Aktuell machen wir das mWn. nur beim Pausieren einer Ladung.

An-Wi, 2023-10-30:
> @naltatis könnten wir das um offene Vorgänge ergänzen? Ganz intuitiv ist es so vllt. nicht?

Mir würde es reichen, wenn nach Abschluss des "offenen Vorgangs" und Speicherung in der Liste der Ladevorgänge anschließend direkt die Langzeitdaten mit aufgefrischt werden würden. Das wäre für mich "intuitiv" genug...
Momentan scheint der "Refresh" aber erst nach etwa/max. 30min zu erfolgen (konnte ich noch nicht testen - Auto ist unterwegs).

naltatis, 2023-11-05:
@An-Wi ja, guter Punkt. Da sich die Daten sowieso erst beim Abschluss eines Ladevorgangs verändern, könnten wir die Aktualisierung der Statistikdaten auch immer dir danach machen.

An-Wi, 2023-11-08:
> Da es in dem Dialog um eher Langzeitdaten geht haben wir die Aktualisierung auf 30min eingestellt. Berücksichtigt werden abgeschlossene Ladevorgänge.

Das hat soweit funktioniert. Nach knapp 30min des letzten Ladevorgangs wurden die Daten aktualisiert.

---

Heute habe ich in dieser Auswertung ein neues, **sehr merkwürdiges Verhalten** entdeckt!!!

[image]

- Die Anzeige am "Handy" (als HomeScreen-Saved-App) stimmt. 
- Die Anzeige auf anderen Geräten zeigt "alte Werte" (immer die gleichen Werte wie im Screenshot zu sehen) an - auch nach gut 2h (Website offen gelassen) aktualisiert sich die Anzeige nicht. 
- weiteren "neuen" Browser getestet, auf dem die Webansicht noch nie lief: zeigt auch die alten Werte an
- Handy (mit der funktionierenden App): Wechsel in Safari-Browser: zeigt auch alte Werte an

3 Anmerkungen dazu:
_1)_ Die Ansicht "Ladevorgänge" ist in/auf allen Ansichten/Geräten identisch und ergibt eine Gesamtmenge von:
749 kWh -> was mit Rundungsdifferenzen "61 kWh Sonne + 687 kWh" (= funktionierende "Handy"-Anzeige) entspricht.
_2)_ Auf den Handy habe ich die letzten Ladevorgänge gesteuert und überwacht. Die Anzeige dort hat sich wie oben beschrieben nach ca. 30min bzw. nach Neustart von evcc immer aktualisiert.
_3)_ Die fehlenden Lademengen (18 + 68 = 86 kWh) lassen sich aber auch nicht mit den letzten Ladevorgängen erklären 🤷‍♂️:
[image]

getestete evcc-Versionen: **0.121.5** & **0.122.0**

andig, 2023-11-09:
@naltatis schaust Du da rein?

An-Wi, 2023-11-11:
Oh shit! Ich habe gerade "**meinen**" Fehler bemerkt!!!
(durch Zufall, weil, die "/api/states" durchgeschaut habe)

In den Screenshots auch klar zu erkennen: die _Zeiträume_ sind unterschiedlich eingestellt!
Ein dickes "SORRY" dafür!

Die kürzliche Anpassung der Statistik mit Einführung der Auswahlmöglichkeit des Zeitraums war mir zwar aufgefallen, aber ich hatte sie nur am Handy auf "gesamt" eingestellt.

Per default scheint aber "30 Tage" voreingestellt zu sein und dieses wird von Gerät zu Gerät separat gespeichert (ist das so glücklich/gewollt?)

An-Wi, 2023-11-15:
Mir ist leider noch ein **_richtiger_** Bug bei der Auswertung der Ladeenergie aufgefallen:

Entstanden bei mir durch Tarifwechsel beim Netzanbieter und damit verbundener Preissenkung (von 33ct auf 31,2ct je kWh).
[image]

Da ich erst seit Anfang Oktober mit evcc arbeite und damit nur sehr wenig Sonnenenergie tanken konnte, ist mein durch die Ladevorgänge errechneter Energiepreis nun größer als mein neuer Netzbezugspreis - das scheint soweit korrekt, da sich der errechnete Energiepreis vermutlich durch die bisherigen Ladevorgänge ermittelt, bei dem der höhere Netzbezugspreis noch galt. 
Aber ins Auge sticht der nun angezeigte negative Wert bei "Ersparnis", die **falsch** berechnet wird: 
`(aktuellerNetzbezugsPreis - EnergiePreis ) * GesamtladeMenge`
(dabei geht die Info des alten Netzbezugspreises komplett verloren)

**Richtig** wäre hier, die Ersparnis pro einzelnem Ladevorgang zu ermitteln und anschließend diese Einzelwerte zu summieren.
Eine Tarif-/Preisänderung kommt sicherlich bei jedem hin und wieder vor...

Da ich in absehbarer Zeit auf einen Zweitarif-Vertrag (nachts deutlich geringerer Netzbezugspreis) umstellen darf, wird die Problematik dann noch auffälliger, wenn ich überwiegend nachts das Auto vollladen lasse. _(Btw.: ich freue mich schon auf die geplante Möglichkeit, wochentagsbezogen das Zielladen zu hinterlegen - die Previews, die ich gesehen habe, sehen super aus - aber das ist ein anderes Thema)_

closed 2023-12-11 by github-actions[bot]

An-Wi, 2023-12-11:
Mh, schade - irgendwie nervt die "-1€ gespart" immer noch. Muss ich wohl mit leben...

naltatis, 2023-12-13:
> Mh, schade - irgendwie nervt die "-1€ gespart" immer noch. Muss ich wohl mit leben...

Das wäre aus meiner Sicht noch mal ein anderes Thema. Der Ersparnisswert wird aktuell ja relativ dumm berechnet. Wir nehmen einfach die Summe und halten den aktuellen Strompreis als Referenz dagegen. Intelligenter wäre die Ersparnis direkt im jeweiligen Ladevorang zu speichern. Dann ist der Referenzwert der zu dem Zeitpunkt gültige Strompreis oder bei dyn. Strompreisen evtl. das aktuelle Mittel. Mach dazu gerne ein neues Issue auf.

An-Wi, 2023-12-21:
ok - erledigt. #11160

Danke für deine Rückmeldung 👍