evcc-io/evcc #10563: OpenWB 2 Fehler charger vehicle id: modbus: exception '4'

atze09, 2023-10-29:
### Describe the bug

Hallo zusammen,
Nach erfolgreicher Einrichtung openWB2 und der Verbindung zu EVCC, bekomme ich nun folgende Fehlermeldung.

Garage: charger vehicle id: modbus: exception '4' (server device failure), function '4'

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: c35b3729cfe9020d0aaac81f785228a9ae1ad6170550d501403ac0e198a62728

interval: 30s # control cycle interval

sponsortoken: xxx

# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.17.3  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.17.3  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.17.3  
  port: 502  
  name: battery3

chargers:
- type: template
  template: go-e 
  host: 192.168.17.83  
  name: wallbox5
- name: openwb
  type: template
  template: openwb-2.0
  # Modbus TCP
  modbus: tcpip
  id: 1
  host: 192.168.17.80 # Hostname
  port: 1502 # Port 
  
vehicles:
- type: template
  template: opel 
  title: Corsa  
  icon: car  
  user: xxx 
  password: xxx
  vin: xxx
  capacity: 46  
  phases: 1  
  cache: 15m  
  mode: pv  
  minSoc: 0  
  targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4
  identifiers:
  - 2
  - 3
  - 4
- type: template
  template: offline
  title: Zoe
  icon: car
  capacity: 50
  phases: 3
  mode: pv
  minSoc: 0
  targetSoc: 100
  minCurrent: 6
  maxCurrent: 16
  identifiers:
  - 1
  name: ev5

loadpoints:
- title: Outdoor
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
- title: Garage
  charger: openwb
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 25
  resetOnDisconnect: true
  vehicle: ev4
site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/10/29 19:53:50 evcc 0.121.5
[main  ] INFO 2023/10/29 19:53:50 using config file: /config/evcc.yaml
[main  ] INFO 2023/10/29 19:53:50 starting ui and api at :7070
[db    ] INFO 2023/10/29 19:53:51 using sqlite database: /data/evcc.db
[lp-2  ] WARN 2023/10/29 19:53:54 locking phase config to 3p for switchable charger
[lp-1  ] DEBUG 2023/10/29 19:53:54 charge total import: 302.000kWh
[site  ] WARN 2023/10/29 19:53:54 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2023/10/29 19:53:54 vehicle 'Zoe' does not support automatic detection
[site  ] INFO 2023/10/29 19:53:54 site config:
[site  ] INFO 2023/10/29 19:53:54   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/29 19:53:54     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/29 19:53:54     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/29 19:53:54     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/29 19:53:54   vehicles:
[site  ] INFO 2023/10/29 19:53:54     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[site  ] INFO 2023/10/29 19:53:54     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/10/29 19:53:54 loadpoint 1:
[lp-1  ] INFO 2023/10/29 19:53:54   mode:        pv
[lp-1  ] INFO 2023/10/29 19:53:54   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/29 19:53:54   meters:      charge ✓
[lp-1  ] INFO 2023/10/29 19:53:54     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2023/10/29 19:53:54 loadpoint 2:
[lp-2  ] INFO 2023/10/29 19:53:54   mode:        pv
[lp-2  ] INFO 2023/10/29 19:53:54   charger:     power ✗ energy ✗ currents ✓ phases ✓ wakeup ✓
[lp-2  ] INFO 2023/10/29 19:53:54   meters:      charge ✗
[lp-1  ] DEBUG 2023/10/29 19:53:54 phase timer inactive
[lp-1  ] DEBUG 2023/10/29 19:53:54 pv timer inactive
[lp-1  ] DEBUG 2023/10/29 19:53:54 guard timer inactive
[lp-2  ] DEBUG 2023/10/29 19:53:54 phase timer inactive
[lp-2  ] DEBUG 2023/10/29 19:53:54 pv timer inactive
[lp-2  ] DEBUG 2023/10/29 19:53:54 guard timer inactive
[lp-2  ] INFO 2023/10/29 19:53:54 vehicle updated: unknown -> Corsa
[lp-2  ] DEBUG 2023/10/29 19:53:54 set charge mode: pv
[lp-2  ] DEBUG 2023/10/29 19:53:54 set min current: 6
[lp-2  ] DEBUG 2023/10/29 19:53:54 set target soc: 80
[lp-2  ] DEBUG 2023/10/29 19:53:54 set max current: 16
[site  ] DEBUG 2023/10/29 19:53:54 ----
[lp-1  ] DEBUG 2023/10/29 19:53:54 charge power: 0W
[lp-2  ] DEBUG 2023/10/29 19:53:54 charge power: 0W
[site  ] DEBUG 2023/10/29 19:53:54 pv power: 0W
[site  ] DEBUG 2023/10/29 19:53:54 battery soc: 86%
[site  ] DEBUG 2023/10/29 19:53:54 battery power: 2681W
[site  ] DEBUG 2023/10/29 19:53:54 grid power: -11W
[site  ] DEBUG 2023/10/29 19:53:54 site power: 2670W
[lp-1  ] DEBUG 2023/10/29 19:53:54 charge voltages: [224 224 222]V
[lp-1  ] DEBUG 2023/10/29 19:53:54 detected connected phases: 3p
[lp-1  ] DEBUG 2023/10/29 19:53:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/29 19:53:54 charge total import: 302.000kWh
[lp-1  ] DEBUG 2023/10/29 19:53:54 charger status: A
[lp-1  ] INFO 2023/10/29 19:53:54 car disconnected
[lp-1  ] DEBUG 2023/10/29 19:53:54 set charge mode: pv
[lp-1  ] DEBUG 2023/10/29 19:53:54 set min current: 6
[lp-1  ] DEBUG 2023/10/29 19:53:54 set max current: 16
[lp-1  ] DEBUG 2023/10/29 19:53:54 set priority: 0
[site  ] DEBUG 2023/10/29 19:54:24 ----
[lp-1  ] DEBUG 2023/10/29 19:54:25 charge power: 0W
[lp-2  ] DEBUG 2023/10/29 19:54:25 charge power: 0W
[site  ] DEBUG 2023/10/29 19:54:25 pv power: 0W
[site  ] DEBUG 2023/10/29 19:54:25 battery soc: 86%
[site  ] DEBUG 2023/10/29 19:54:25 battery power: 2655W
[site  ] DEBUG 2023/10/29 19:54:25 grid power: -7W
[site  ] DEBUG 2023/10/29 19:54:25 site power: 2648W
[lp-2  ] DEBUG 2023/10/29 19:54:26 vehicle odometer: 14220km
[lp-2  ] DEBUG 2023/10/29 19:54:26 charger status: B
[lp-2  ] INFO 2023/10/29 19:54:26 car connected
[lp-2  ] DEBUG 2023/10/29 19:54:26 pv timer elapse
[lp-2  ] DEBUG 2023/10/29 19:54:26 pv timer inactive
[lp-2  ] DEBUG 2023/10/29 19:54:26 charger: guard elapse
[lp-2  ] DEBUG 2023/10/29 19:54:26 guard timer inactive
[lp-2  ] ERROR 2023/10/29 19:54:26 charger vehicle id: modbus: exception '4' (server device failure), function '4'
[lp-2  ] DEBUG 2023/10/29 19:54:26 vehicle status: B
[lp-2  ] DEBUG 2023/10/29 19:54:26 vehicle soc: 63%
[lp-2  ] DEBUG 2023/10/29 19:54:26 vehicle range: 178km
[lp-2  ] DEBUG 2023/10/29 19:54:26 !! scalePhases: GetPhases 0p <> 3p wanted
[lp-2  ] DEBUG 2023/10/29 19:54:26 switched phases: 3p
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

 0.121.5

andig, 2023-10-30:
Laut https://openwb.de/main/wp-content/uploads/2023/10/ModbusTCP-openWB-series2-Pro-1.pdf machen wir alles richtig- 10 Register ab 10160.

atze09, 2023-10-30:
> Laut https://openwb.de/main/wp-content/uploads/2023/10/ModbusTCP-openWB-series2-Pro-1.pdf machen wir alles richtig- 10 Register ab 10160.

Kann ich so bestätigen. Alle anderen Register erreiche ich auch via Windows. Jedoch bei 10160 bekomme ich auch dort einen Fehler.

closed 2023-10-30 by andig