evcc-io/evcc #10571: Alphatec Wallbox: Status 8 an Register 0

deadrabbit87, 2023-10-30:
### Describe the bug

Die Alphatec Wallbox gibt, trotz aktueller Firmware, sporadisch den Status 8 am Register  0 zurück. 

Laut Alphatec bedeutet dies "warten auf Ladefreigabe". 

Hier der Tracelog: 

<details>
<summary>sudo evcc -l trace</summary>

```
[alphatec] TRACE 2023/10/30 09:41:51 modbus: send 01 10 00 05 00 01 02 00 06 26                                                                                 07
[alphatec] TRACE 2023/10/30 09:42:06 modbus: recv
[lp-1  ] ERROR 2023/10/30 09:42:06 max charge current 6A: serial: timeout
[site  ] DEBUG 2023/10/30 09:42:06 ----
[lp-1  ] DEBUG 2023/10/30 09:42:06 charge power: 0W
[modbus] TRACE 2023/10/30 09:42:06 modbus: send 00 13 00 00 00 06 03 03 78 35 00 02
[modbus] TRACE 2023/10/30 09:42:06 modbus: recv 00 13 00 00 00 07 03 03 04 00 00 05 a1
[modbus] TRACE 2023/10/30 09:42:06 modbus: send 00 14 00 00 00 06 03 03 78 f1 00 02
[modbus] TRACE 2023/10/30 09:42:06 modbus: recv 00 14 00 00 00 07 03 03 04 00 00 05 a9
[site  ] DEBUG 2023/10/30 09:42:06 pv power: 2890W
[tasmota] TRACE 2023/10/30 09:42:06 GET http://192.168.178.36/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/10/30 09:42:06 {"StatusSNS":{"Time":"2023-10-30T09:42:05","SML":{"total_in":2114.0900,"total_out":7737.2998,"power_curr":-2570,"Voltage":229.1,"Current":3.93,"phase_angle_L1":184.0,"Freq":49.9,"ID":"0a014546522203e0c1e7"}}}
[site  ] DEBUG 2023/10/30 09:42:06 grid power: -2570W
[site  ] DEBUG 2023/10/30 09:42:06 site power: -2570W
[alphatec] TRACE 2023/10/30 09:42:06 modbus: send 01 03 00 00 00 01 84 0a
[alphatec] TRACE 2023/10/30 09:42:06 modbus: recv 01 03 02 00 08 b9 82
[lp-1  ] ERROR 2023/10/30 09:42:06 charger: invalid status: 8
[site  ] DEBUG 2023/10/30 09:42:16 ----
[lp-1  ] DEBUG 2023/10/30 09:42:16 charge power: 0W
[modbus] TRACE 2023/10/30 09:42:16 modbus: send 00 15 00 00 00 06 03 03 78 35 00 02
[modbus] TRACE 2023/10/30 09:42:16 modbus: recv 00 15 00 00 00 07 03 03 04 00 00 05 a3
[modbus] TRACE 2023/10/30 09:42:16 modbus: send 00 16 00 00 00 06 03 03 78 f1 00 02
[modbus] TRACE 2023/10/30 09:42:16 modbus: recv 00 16 00 00 00 07 03 03 04 00 00 05 a8
[site  ] DEBUG 2023/10/30 09:42:16 pv power: 2891W
[tasmota] TRACE 2023/10/30 09:42:16 GET http://192.168.178.36/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/10/30 09:42:16 {"StatusSNS":{"Time":"2023-10-30T09:42:15","SML":{"total_in":2114.0900,"total_out":7737.3070,"power_curr":-2593,"Voltage":228.8,"Current":3.96,"phase_angle_L1":183.0,"Freq":49.9,"ID":"0a014546522203e0c1e7"}}}
[site  ] DEBUG 2023/10/30 09:42:16 grid power: -2593W
[site  ] DEBUG 2023/10/30 09:42:16 site power: -2593W
[alphatec] TRACE 2023/10/30 09:42:16 modbus: send 01 03 00 00 00 01 84 0a
[alphatec] TRACE 2023/10/30 09:42:16 modbus: recv 01 03 02 00 08 b9 82
[lp-1  ] ERROR 2023/10/30 09:42:16 charger: invalid status: 8
[site  ] DEBUG 2023/10/30 09:42:26 ----
[lp-1  ] DEBUG 2023/10/30 09:42:26 charge power: 0W
[modbus] TRACE 2023/10/30 09:42:26 modbus: send 00 17 00 00 00 06 03 03 78 35 00 02
[modbus] TRACE 2023/10/30 09:42:26 modbus: recv 00 17 00 00 00 07 03 03 04 00 00 05 93
[modbus] TRACE 2023/10/30 09:42:26 modbus: send 00 18 00 00 00 06 03 03 78 f1 00 02
[modbus] TRACE 2023/10/30 09:42:26 modbus: recv 00 18 00 00 00 07 03 03 04 00 00 05 9c
[site  ] DEBUG 2023/10/30 09:42:26 pv power: 2863W
[tasmota] TRACE 2023/10/30 09:42:26 GET http://192.168.178.36/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/10/30 09:42:26 {"StatusSNS":{"Time":"2023-10-30T09:42:25","SML":{"total_in":2114.0900,"total_out":7737.3141,"power_curr":-2552,"Voltage":229.1,"Current":3.90,"phase_angle_L1":184.0,"Freq":49.9,"ID":"0a014546522203e0c1e7"}}}
[site  ] DEBUG 2023/10/30 09:42:26 grid power: -2552W
[site  ] DEBUG 2023/10/30 09:42:26 site power: -2552W
[alphatec] TRACE 2023/10/30 09:42:26 modbus: send 01 03 00 00 00 01 84 0a
[alphatec] TRACE 2023/10/30 09:42:26 modbus: recv 01 03 02 00 08 b9 82
[lp-1  ] ERROR 2023/10/30 09:42:26 charger: invalid status: 8
[site  ] DEBUG 2023/10/30 09:42:36 ----
[lp-1  ] DEBUG 2023/10/30 09:42:36 charge power: 0W
[modbus] TRACE 2023/10/30 09:42:36 modbus: send 00 19 00 00 00 06 03 03 78 35 00 02
[modbus] TRACE 2023/10/30 09:42:36 modbus: recv 00 19 00 00 00 07 03 03 04 00 00 05 8c
[modbus] TRACE 2023/10/30 09:42:36 modbus: send 00 1a 00 00 00 06 03 03 78 f1 00 02
[modbus] TRACE 2023/10/30 09:42:36 modbus: recv 00 1a 00 00 00 07 03 03 04 00 00 05 95
[site  ] DEBUG 2023/10/30 09:42:36 pv power: 2849W
[tasmota] TRACE 2023/10/30 09:42:36 GET http://192.168.178.36/cm?cmnd=Status+8&password=&user=
[tasmota] TRACE 2023/10/30 09:42:36 {"StatusSNS":{"Time":"2023-10-30T09:42:35","SML":{"total_in":2114.0900,"total_out":7737.3212,"power_curr":-2537,"Voltage":229.0,"Current":3.88,"phase_angle_L1":184.0,"Freq":49.9,"ID":"0a014546522203e0c1e7"}}}
[site  ] DEBUG 2023/10/30 09:42:36 grid power: -2537W
[site  ] DEBUG 2023/10/30 09:42:36 site power: -2537W
[alphatec] TRACE 2023/10/30 09:42:36 modbus: send 01 03 00 00 00 01 84 0a
[alphatec] TRACE 2023/10/30 09:42:36 modbus: recv 01 03 02 00 08 b9 82
[... 310 lines cut ...]

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.5 (da68bc03)

premultiply, 2023-10-30:
In welchem Ladezustand passiert das?
Nur bei "verbunden, lädt nicht" (also Status B) aber nie bei A oder C?

deadrabbit87, 2023-10-30:
Ja, passiert nur wenn angesteckt aber nicht geladen wird.

premultiply, 2023-10-30:
Ok, das ist einfach.
Baue ich ein.

deadrabbit87, 2023-10-30:
Danke. 

Noch ne Idee wegen dem Timeouts?

closed 2023-10-30 by premultiply