evcc-io/evcc #10580: Porsche Fahrzeug über API aufwecken (endlich!)

MarkusGH, 2023-10-30:
Bekanntlich schlafen Porsche Fahrzeuge nach 5 Unterbrechungen des Ladevorgangs - innerhalb eines gewissen Zeitraums - ein ("Spielschutz") und lassen sich dann auch nicht über eine CP Unterbrechung wecken.

Die Porsche API Implementation hat bereits eine "Stub" Implementation von  api.Resurrector, diese ruft aktuell den Status ab, was leider (oder glücklicherweise) **nicht** zum Aufwachen des Fahrzeugs führt.

Dank [pyporscheconnectapi ](https://github.com/CJNE/pyporscheconnectapi) konnte ich jetzt endlich testen, wie sich das Fahrzeug über API aufwecken lässt.

porschecli overview -v "VIN" führte bei meinen Tests zuverlässig zum Aufwachen des Fahrzeugs, auch wenn der Spielschutz aktiv war.

Die Implementation in Python sieht so aus:

```
    async def getCurrentOverview(self, vin):
        progressResult = await self._connection.post(
            f"https://api.porsche.com/service-vehicle/{self.locale_str}/vehicle-data/{vin}/current/request"
        )
        spinnerResult = await self._spinner(
            f"https://api.porsche.com/service-vehicle/{self.locale_str}/vehicle-data/{vin}/current/request/{progressResult['requestId']}/status"
        )
        data = await self._connection.get(
            f"https://api.porsche.com/service-vehicle/{self.locale_str}/vehicle-data/{vin}/current/request/{progressResult['requestId']}"
        )
        return data
```

Leider verlassen mich hier meine Fähigkeiten, das nach GO zu übersetzen. Ich bin aber zuversichtlich, dass es jemand fähigeres als ich es relativ einfach hinbekommen sollte diese Aufrufe aus der WakeUp() Funktion der Porsche Vehicle Anbindung heraus aufzurufen.

andig, 2023-10-31:
Wie sehen denn die Requests aus?

MarkusGH, 2023-10-31:
Nach meinem Verständnis (so wie ich den Python Code interpretiere):
1) Post https://api.porsche.com/service-vehicle/{self.locale_str}/vehicle-data/{vin}/current/request
2) requestId aus Ergebnis extrahieren
3) Get https://api.porsche.com/service-vehicle/{self.locale_str}/vehicle-data/{vin}/current/request/{progressResult['requestId']}

Was das mit dem "_spinner" da auf sich hat verstehe ich nicht.
Ob man die requestId wirklich braucht ebenfalls nicht...

MarkusGH, 2023-11-26:
@andig: Könntest Du bitte https://github.com/evcc-io/evcc/pull/10928 prüfen und dieses Issue wieder aufmachen? Danke!