evcc-io/evcc #10582: EVCC stop when OpenEvse charger is diconected

ioniks, 2023-10-31:
### Describe the bug

When the OpenEVSE charger is disconnected, EVCC stop working and i have to restart.

before With 0.117.0 i haven't this problem

OpenEVSE v4.2.2 (last version), Gui V1

### Steps to reproduce

1. Configure openevse charger
2. start EVCC
3. Disconnect wifi (or openEvse charger)

### Configuration details

```yaml
chargers:
- type: template
  template: openevse
  host: 192.168.0.27
  name: wallbox
```

### Log details

```text
[lp-1  ] ERROR 2023/10/30 23:45:37 charge meter: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[lp-1  ] ERROR 2023/10/30 23:45:43 charge rater: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[lp-1  ] ERROR 2023/10/30 23:45:43 charge timer: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[lp-1  ] ERROR 2023/10/30 23:45:43 charge total import: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[main  ] INFO 2023/10/31 09:40:42 evcc 0.121.5
[main  ] INFO 2023/10/31 09:40:42 using config file: /etc/evcc.yaml
[lp-1  ] ERROR 2023/10/31 12:14:49 charge meter: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[lp-1  ] ERROR 2023/10/31 12:14:57 charge rater: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[lp-1  ] ERROR 2023/10/31 12:14:57 charge timer: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[lp-1  ] ERROR 2023/10/31 12:14:57 charge total import: Get "http://192.168.0.27/status": dial tcp 192.168.0.27:80: i/o timeout
[main  ] INFO 2023/10/31 12:31:58 evcc 0.121.5
[main  ] INFO 2023/10/31 12:31:58 using config file: /etc/evcc.yaml
```

### What type of operating system are you running?

Docker container

### Version

0.121.5

andig, 2023-10-31:
It's unclear what you're asking. What do you mean by "evcc stop"?

ioniks, 2023-10-31:
The docker container is stopped.

andig, 2023-10-31:
We're not providing Docker support. Please reopen if you can repro this outside of Docker.

closed 2023-10-31 by andig

ioniks, 2023-10-31:
i don't understand why, i use your docker image .
I can't install EVCC in other way

andig, 2023-11-01:
As the issue template says:

> NOTE if you're using HomeAssistant or Docker we ask you to reproduce the problem on plain Linux or Windows first.