evcc-io/evcc #10584: Modbus Proxy leitet Funktionstyp "0x02 Read Discrete Inputs" nicht weiter

docolli, 2023-10-31:
### Describe the bug

Über den Modbus Proxy lassen sich keine Anfragen vom Modbus Funktionstyp "02 (0x02) Read Discrete Inputs" durch schicken. Als Antwort kommt nur eine Modbus Fehlermeldung zurück. Der Modbus Proxy schreibt bei einer solchen Anfrage auch nichts ins Log.

### Steps to reproduce

1. Modbus Proxy (ohne readonly) konfigurieren, Ziel bei mir: eine E3DC/Wallbe/Phoenix Wallbox
2. z.B. mit Modbus Master eine Anfrage vom Typ "Read Discrete" auf die Register 200 bis 208 an den evcc Modbus Proxy schicken (00 3a 00 00 00 06 ff 02 00 c8 00 09)
3. Modbus Proxy loggt nichts (!)
4. Als Antwort an Modbus Master kommt Fehler zurück (00 3a 00 00 00 03 ff 82 01)
5. Test ob Modbus Proxy und Log funktioniert: mit Modbus Master eine Anfrage "Read Coils" auf Register 480 schicken
6. Anfrage und Antwort steht im evcc Log

### Configuration details

```yaml
modbusproxy:
- port: 502
  uri: 192.168.1.125:502
#  readonly: true
```

### Log details

```text
Log TRACE:

sudo journalctl -fau evcc | grep proxy
Okt 31 14:48:03 fhem-heizung evcc[22593]: [proxy-502] DEBUG 2023/10/31 14:48:03 modbus proxy for 192.168.1.125:502 listening at :502
Okt 31 14:49:27 fhem-heizung evcc[22593]: [proxy-502] TRACE 2023/10/31 14:49:27 read coil: id 255 addr 480 qty 1
Okt 31 14:49:27 fhem-heizung evcc[22593]: [proxy-502] TRACE 2023/10/31 14:49:27 read coil: 00
```

### What type of operating system are you running?

Linux

### Version

0.121.5

andig, 2023-10-31:
Vmtl nicht implementiert weil ich das nicht verproben konnte. Kannst du selbst compilieren? Dann könnten wir mal zusammen basteln.

docolli, 2023-10-31:
Wie könnte ich denn selber compilieren? Den Sourcecode auf mein Raspi-System auschecken und dort irgendwie compilieren? Da habe ich halt leider gar keine Erfahrung damit. Da müsstest du mir kräftig unter die Arme greifen.
Oder kann ich den Code in mein Github Konto forken und dann dort bearbeiten und von github bauen lassen, so wie ihr eure Pakete released?

docolli, 2023-10-31:
Ok, schau ich mir an. Kann ich dort auch für den Raspi (armhf) kompilieren? 
Achso .... Ich kann ja auf dem PC eine eigene evcc exe-Instanz laufen lassen und gegen diese testen! 🙈

docolli, 2023-10-31:
Im Readme finde ich nur Infos, dass ich go und node.js  brauche. Habe ich auch instaliert und mir den Quellcode heruntergeladen. Wie lautet der go build Befehl, damit er mir eine exe aus meinem lokalen Repository baut?

Rufe ich das hier aus dem Quellcodeverzeichnis heraus auf
`go build -o d:\Sourcecode\evcc\evcc_proxytest.exe`
so lädt er einen Haufen Dateien herunter, endet dann aber mit
`main.go:16:13: pattern dist: no matching files found`

andig, 2023-10-31:
Readme befolgen: https://github.com/evcc-io/evcc#contribute

docolli, 2023-10-31:
Ja, habe ich schon gelesen, aber kann es sein, dass dies nur für Linux OS PCs gilt und man Windows basierte PCs nicht verwenden kann? `make` und `make install` kenne ich nur von Linux. Wenn ja, wäre es gut, wenn das als Voraussetzung genannt wird.
Ich versuche das jetzt mal in einer Alpine Linux VM zu bauen.

andig, 2023-10-31:
Die relevanten Schritte weg zu lassen funktioniert jedenfalls nicht. Auch make gibts für Windows, z.B. bei cygwin.

docolli, 2023-10-31:
Unter Alpine bin ich jetzt schon durch! Musste es nur von ´3.18´auf ´edge´ upgraden, damit ich go 1.21 bekomme. An die cygwin Umgebung unter Windows habe ich gar nicht gedacht.
Jetzt teste ich noch, ob die selbst kompilierte Version auch dieses Problem hat, dann ist für heute Schluß.

closed 2023-11-01 by andig