evcc-io/evcc #10585: Electricity transport tariff

odlg, 2023-10-31:
In Denmark Energinet provides grid tariffs, but in Denmark electricity prices consists of several components. These can not be modeled with current evcc configuration. Energinet grid tariff is the basic price.
Denmark has more than 20 electricity network providers that each has fees for transporting electricity. Many of them have like 3-4 different prices during the day. Most expensive in the evening hours, from eg. 17-21. 

Would it be possible to add ability to add time-based extra charges to tariffs? 

Here is part of a screenshot from an app, where red, blue and green are the varying extra charges. Yellow is the basic price from energinet.
[image]

andig, 2023-10-31:
Is this a request for time- dependent charges? Or for something related to Energinet?

odlg, 2023-10-31:
Tim-dependent charges

andig, 2024-01-13:
Looking at priorities it seems we don't have capacity to do this at this time. You can however use a custom tariff to inject the actual costs using the usual plugins for mqtt etc.

closed 2024-01-13 by andig

odlg, 2024-01-13:
@andig If I was able to inject prices for the next 24 hours or so, I could use it. But it makes little sense to just be able to inject current price via mqtt. My prices changes every hour.

andig, 2024-01-14:
OT, please open new issue if you have a suggestion how that should look api-wise.