evcc-io/evcc #10586: Alphatec Mini Wallbox lädt nicht (mehr)

Helios1974, 2023-10-31:
### Describe the bug

Ich schliesse den Stecker der Alphatec Wallbox - wie üblich - an mein E-Auto an, aber es erfolgt kein Ladevorgang. Das lief vor dem Update auf v0.121.5 problemlos.
EVCC läuft einem Container unter einer Synolgy NAS, bisher immer problemlos.
Fehler kann auch unter Mac via Terminalbefehle "reproduziert" werden, siehe dazu das Logging unten.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

interval: 45s # control cycle interval

meters:
- type: template
  template: senec-home
  usage: pv
  host: 192.168.1.34
  schema: https
  name: my_pv
- type: template
  template: senec-home
  usage: grid
  host: 192.168.1.34
  schema: https
  name: my_grid
- type: template
  template: senec-home
  usage: battery
  host: 192.168.1.34
  schema: https
  name: my_battery
- type: template
  template: eastron
  usage: grid
  name: zaehler
  # RS485 via TCP/IP (Modbus RTU)
  modbus: rs485tcpip
  id: 1
  host: 192.168.1.129
  #host: 192.0.2.2 # Hostname
  port: 502 # Port 

chargers:
  - name: wallbox # reference name
    type: template
    template: alphatec
    modbus: rs485tcpip
    # modbus: tcpip
    id: 1
    host: 192.168.1.76
    port: 502

loadpoints:
- title: Garage Oach # display name for UI
  charger: wallbox # Wallbox Referenz
  meter: zaehler
  vehicles:
    - eup
  mode: off

vehicles:
  - name: eup
    type: template
    template: vw
    title: Mein VW E-Up
    user: meinemailadresse
    password: MeinPasswort
    vin: MeineVIN
    capacity: 50

site:
  title: Daheim
  meters:
    grid: my_grid
    pvs:
    - my_pv
    batteries:
    - my_battery

sponsortoken: MeinSponsortoken
```

### Log details

```text
[main  ] INFO 2023/10/31 20:06:02 evcc 0.121.5
[main  ] INFO 2023/10/31 20:06:02 using config file: /Users/xxx/Downloads/evcc_0.121.5_macOS-all/evcc.yaml
[main  ] INFO 2023/10/31 20:06:02 starting ui and api at :7070
[db    ] INFO 2023/10/31 20:06:02 using sqlite database: /Users/xxx/.evcc/evcc.db
[lp-1  ] WARN 2023/10/31 20:06:04 vehicles option is deprecated
[lp-1  ] WARN 2023/10/31 20:06:04 phases not configured, assuming 3p
[lp-1  ] DEBUG 2023/10/31 20:06:04 charge total import: 761.645kWh
[site  ] WARN 2023/10/31 20:06:04 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/10/31 20:06:04 deprecated: use 'battery' instead of 'batteries'
[site  ] WARN 2023/10/31 20:06:04 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2023/10/31 20:06:04 site config:
[site  ] INFO 2023/10/31 20:06:04   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/31 20:06:04     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/31 20:06:04     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/31 20:06:04     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/10/31 20:06:04   vehicles:
[site  ] INFO 2023/10/31 20:06:04     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/10/31 20:06:04 loadpoint 1:
[lp-1  ] INFO 2023/10/31 20:06:04   mode:        off
[lp-1  ] INFO 2023/10/31 20:06:04   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/10/31 20:06:04   meters:      charge ✓
[lp-1  ] INFO 2023/10/31 20:06:04     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/10/31 20:06:04 phase timer inactive
[lp-1  ] DEBUG 2023/10/31 20:06:04 pv timer inactive
[lp-1  ] DEBUG 2023/10/31 20:06:04 guard timer inactive
[lp-1  ] DEBUG 2023/10/31 20:06:04 max charge current: 6A
[site  ] DEBUG 2023/10/31 20:06:04 ----
[lp-1  ] DEBUG 2023/10/31 20:06:04 charge power: 0W
[site  ] DEBUG 2023/10/31 20:06:05 pv power: 0W
[site  ] DEBUG 2023/10/31 20:06:05 battery soc: 0%
[site  ] DEBUG 2023/10/31 20:06:05 battery power: 0W
[site  ] DEBUG 2023/10/31 20:06:06 grid power: 1361W
[site  ] DEBUG 2023/10/31 20:06:06 site power: 1361W
[lp-1  ] DEBUG 2023/10/31 20:06:06 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/10/31 20:06:06 charge total import: 761.645kWh
[lp-1  ] DEBUG 2023/10/31 20:06:06 charger status: B
[lp-1  ] DEBUG 2023/10/31 20:06:06 charge start energy: 762kWh
[lp-1  ] INFO 2023/10/31 20:06:06 car connected
[lp-1  ] DEBUG 2023/10/31 20:06:06 vehicle api refresh
[lp-1  ] DEBUG 2023/10/31 20:06:06 pv timer elapse
[lp-1  ] DEBUG 2023/10/31 20:06:06 pv timer inactive
[lp-1  ] DEBUG 2023/10/31 20:06:06 charger: guard elapse
[lp-1  ] DEBUG 2023/10/31 20:06:06 guard timer inactive
[lp-1  ] DEBUG 2023/10/31 20:06:06 charge total import: 761.645kWh
[main  ] DEBUG 2023/10/31 20:06:06 vehicle status: B (Mein VW E-Up)
[lp-1  ] INFO 2023/10/31 20:06:06 vehicle updated: unknown -> Mein VW E-Up
[lp-1  ] DEBUG 2023/10/31 20:06:06 vehicle status: B
[lp-1  ] DEBUG 2023/10/31 20:06:06 vehicle soc: 45%
[lp-1  ] DEBUG 2023/10/31 20:06:06 vehicle range: 79km
[lp-1  ] DEBUG 2023/10/31 20:06:06 charger disable
[lp-1  ] DEBUG 2023/10/31 20:06:06 wake-up timer: stop
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.121.5

Helios1974, 2023-10-31:
Die Kommunikation via Telegram habe ich in der Konfigurationsdatei rausgelassen.

andig, 2023-10-31:
> site power: 1361W

Du hast Bezug und laut Log ist der Modus "off". Klar, dass da nichts lädt.