evcc-io/evcc #10589: geladene kWh passen nicht zum Zählerstand

edvandreas, 2023-11-01:
### Describe the bug

In der Ansicht der Ladevorgänge wird der Anfangs- und Endzählerstand der Wallbox ausgegeben. Zusaätzlich wird ausgegeben, wie viel kWh in Summe geladen wurde. Diese werte passen leider nicht zusammen.

evcc Version: 0.121.5
Wallbox: go-e Gemini

[image]
[image]

### Steps to reproduce

1. Fahrzeug laden
2. Werte vergleichen nach Ladevorgänge
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:80
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: info
levels:
  cache: error

# unique installation id
plant: 123

interval: 15s # control cycle interval

sponsortoken: vorhanden

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

mqtt:
  broker: 192.168.1.230:1883

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.1.199  
  name: Garage

vehicles:
- type: template
  template: tesla 
  title: Stromfix  
  accessToken: XXX
  capacity: 82  
  vin: XXX
  name: ev1

loadpoints:
- title: Garage
  charger: Garage
  vehicle: ev1
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

meters:
- name: grid
  type: custom
  power:
    source: calc
    add:
    - source: mqtt
      topic: evcc/bezug
    - source: mqtt
      topic: evcc/einspeisung
- name: pv
  type: custom
  power: 
    source: calc
    add:
    - source: mqtt
      topic: deye/dc/pv1/power
    - source: mqtt
      topic: deye/dc/pv2/power
  currents: 
    - source: mqtt
      topic: deye/ac/l1/current
    - source: mqtt
      topic: deye/ac/l2/current
    - source: mqtt
      topic: deye/ac/l3/current
- name: battery
  type: custom
  power: 
    source: mqtt
    topic: deye/battery/power
  soc: 
    source: mqtt
    topic: deye/battery/soc
  capacity: 12.2

site:
  title: Wallbox
  meters:
    grid: grid
    pv: pv
    battery: battery
  residualPower: 100 # im PV modus wird 150W für Laden des Hausakkus übrig gelassen, damit Regelkreise Tesla/EVCC nicht konkurrieren
  prioritySoc: 80 # Hausbatterie bekommt bis zum SoC Priorität beim laden
  #bufferSoc: 95 # Hausbatterie wird oberhalb Soc 95% als Puffer genutzt

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.28 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh
```

### Log details

```text
[main  ] INFO 2023/10/31 15:38:07 evcc 0.121.5
[main  ] INFO 2023/10/31 15:38:07 using config file: /etc/evcc.yaml
[main  ] INFO 2023/10/31 15:38:07 starting ui and api at :80
[db    ] INFO 2023/10/31 15:38:07 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/10/31 15:38:07 connecting evcc-950610288 at tcp://192.168.1.230:1883
[site  ] WARN 2023/10/31 15:38:08 bufferStartSoc must be larger than bufferSoc
[site  ] INFO 2023/10/31 15:38:08 site config:
[site  ] INFO 2023/10/31 15:38:08   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/10/31 15:38:08     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/10/31 15:38:08     pv 1:      power ✓ energy ✗ currents ✓
[site  ] INFO 2023/10/31 15:38:08     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/10/31 15:38:08   vehicles:
[site  ] INFO 2023/10/31 15:38:08     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/10/31 15:38:08 loadpoint 1:
[lp-1  ] INFO 2023/10/31 15:38:08   mode:        pv
[lp-1  ] INFO 2023/10/31 15:38:08   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/10/31 15:38:08   meters:      charge ✓
[lp-1  ] INFO 2023/10/31 15:38:08     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2023/10/31 15:38:08 vehicle updated: unknown -> Stromfix
[site  ] WARN 2023/10/31 15:38:08 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] ERROR 2023/10/31 15:38:11 vehicle odometer: asleep
[lp-1  ] INFO 2023/10/31 15:38:11 car connected
[lp-1  ] ERROR 2023/10/31 15:38:11 vehicle soc: asleep
[lp-1  ] ERROR 2023/10/31 15:38:23 vehicle odometer: asleep
[lp-1  ] INFO 2023/11/01 02:13:53 start charging ->
[lp-1  ] INFO 2023/11/01 03:03:23 stop charging <-
[lp-1  ] INFO 2023/11/01 03:03:38 start charging ->
[lp-1  ] INFO 2023/11/01 03:06:53 stop charging <-
[lp-1  ] INFO 2023/11/01 05:25:23 start charging ->
[lp-1  ] INFO 2023/11/01 08:01:23 stop charging <-
```

### What type of operating system are you running?

Docker container

### Version

0.121.5

andig, 2023-11-01:
Klingt nach https://github.com/evcc-io/evcc/issues/10452?

closed 2023-11-01 by andig

edvandreas, 2023-11-01:
Korrekt, das scheint das gleiche Problem zu sein. Habe nach ähnlichen Issues gesucht aber den nicht gefunden.

Aussagekräftigeres Log kommt zum nächsten Ladevorgang. Kann leider ein paar Tage dauern.