evcc-io/evcc #10593: Fahrzeug wird nicht mehr erkannt

pitware2005, 2023-11-01:
### Describe the bug

Ich habe seit heute das Problem, das mein Tesla nicht mehr erkannt wird.

Es gab zwar bereites ähnliche Probleme hier allerdings hat nichts davon geholfen.

Was kann ich noch tun?

Zuletzt wurde er am 22.10. geladen und da hat noch alles funktioniert. Habe heute sogar das aktuellste Update installiert und es hat nichts geholfen.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: tesla
  title: Model Y
  accessToken: XXXXXXXXXXXXXXXXXXXX
  refreshToken: XXXXXXXXXXXXXXXXXXXXXX
  vin: XXXXXXXXXXXXXXXXXXXXXXX
  capacity: 60
  name: ev3
```

### Log details

```text
[main  ] ERROR 2023/11/01 09:19:16 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'tesla': cannot get vehicles: Get "https://owner-api.teslamotors.com/api/1/vehicles": oauth2: "login_required" "Login required" "https://auth.tesla.com/error/reference/9059XXXXXXXXXXXXX"
```

### What type of operating system are you running?

Linux

### Version

0.121.5

andig, 2023-11-01:
Das könnte an https://github.com/evcc-io/evcc/issues/10460 liegen.

/cc @naltatis

pitware2005, 2023-11-01:
@andig sorry da war mein tippen wohl schneller als mein Kopf heute :-)

Das was Du verlinkt hast soll doch aber erst 2024 eintreffen?! Oder machen die vorher schon ernst und es geht der alte Weg nicht mehr.

naltatis, 2023-11-01:
Mit meinem Login/Fahrzeug funktioniert die aktuelle Implementierung aktuell. Also sieht nicht aus wie ein allgemeines Problem. @pitware2005 Tokens erneuern hast du vmtl. schon versucht?

phrozen77, 2023-11-01:
Nur fürs Protokoll: unser Model 3 (evcc 0.121.5, 2023.32.6 Firmware) hat heute einwandfrei geladen.