evcc-io/evcc #10594: (Hyundai) IONIQ und Tucson offline

lordvalium, 2023-11-01:
### Describe the bug

evcc cant connect to my hyundai cars anymore, hence not loading them

### Steps to reproduce

1. open evcc in browser
2. select car
3. car is offline
...

### Configuration details

```yaml
confiugration did not change, nor login data for my cars. It was working 2 days before. 
Standard template hyundai
```

### Log details

```text
sudo journalctl -u evcc --since "yesterday" | grep ioniq

[main  ] INFO 2023/11/01 10:48:03 evcc 0.121.1
[main  ] INFO 2023/11/01 10:48:03 using config file: /etc/evcc.yaml
[main  ] INFO 2023/11/01 10:48:03 starting ui and api at :7070
[db    ] INFO 2023/11/01 10:48:04 using sqlite database: /home/fox/.evcc/evcc.db
[site  ] WARN 2023/11/01 10:48:09 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2023/11/01 10:48:09 site config:
[site  ] INFO 2023/11/01 10:48:09   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/11/01 10:48:09     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/11/01 10:48:09     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/11/01 10:48:09     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/11/01 10:48:09   vehicles:
[site  ] INFO 2023/11/01 10:48:09     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2023/11/01 10:48:09     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/11/01 10:48:09 loadpoint 1:
[lp-1  ] INFO 2023/11/01 10:48:09   mode:        pv
[lp-1  ] INFO 2023/11/01 10:48:09   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/11/01 10:48:09   meters:      charge ✗
[lp-1  ] DEBUG 2023/11/01 10:48:09 phase timer inactive
[lp-1  ] DEBUG 2023/11/01 10:48:09 pv timer inactive
[lp-1  ] DEBUG 2023/11/01 10:48:09 guard timer inactive
[site  ] DEBUG 2023/11/01 10:48:09 ----
[lp-1  ] DEBUG 2023/11/01 10:48:09 charge power: 0W
[main  ] INFO 2023/11/01 10:48:10 new version available: 0.121.5
[site  ] DEBUG 2023/11/01 10:48:10 pv power: 4410W
[site  ] DEBUG 2023/11/01 10:48:10 battery soc: 100%
[site  ] DEBUG 2023/11/01 10:48:10 battery power: 8W
[site  ] DEBUG 2023/11/01 10:48:10 grid power: -618W
[site  ] DEBUG 2023/11/01 10:48:10 grid currents: [1.1 1.12 1.96]A
[site  ] DEBUG 2023/11/01 10:48:10 site power: -610W
[lp-1  ] DEBUG 2023/11/01 10:48:10 charger status: A
[lp-1  ] INFO 2023/11/01 10:48:10 car disconnected
[lp-1  ] DEBUG 2023/11/01 10:48:10 set charge mode: pv
[lp-1  ] DEBUG 2023/11/01 10:48:10 set min current: 6
[lp-1  ] DEBUG 2023/11/01 10:48:10 set max current: 16
[lp-1  ] DEBUG 2023/11/01 10:48:10 set priority: 0
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.1

closed 2023-11-01 by andig

lordvalium, 2023-11-01:
What was the fix @andig ? thx

andig, 2023-11-01:
No fix. Issue closed due to missing information, i.e. not log file that shows the error.