evcc-io/evcc #10596: Solarlog pv energy in Watt?

kai-morich, 2023-11-01:
### Describe the bug

Aufgrund des PR #10516 habe in meinem solarlog-kompatiblen [Eigenbau](https://www.photonenhuette.de) um das neue Register erweitert, da evcc permanent modbus parsing Fehler anzeigte.
Für das eigentliche Laden wird der Wert ja nicht wirklich benötigt, sondern nur an sonstige Verwender weitergereicht.

Mein Datenloger zeigt gerundet 124906.7 kWh
Per Modbus liefere ich 124906853
In InfluxDB sehe ich den Wert 124.9k

Ist das `scale: 0.001` in PR #10516 richtig? 
Laut https://docs.evcc.io/docs/reference/configuration/messaging sollte der Wert in Watt bleiben.

### Steps to reproduce

solarlog kompatibler Eigenbau

### Configuration details

```yaml
meters:
- name: pv
  type: template
  template: solarlog 
  usage: pv  
  host: 192.168.xxx
  port: 2081
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Docker container

### Version

0.121.5

VolkerK62, 2023-11-01:
>@VolkerK62 Ist das scale: 0.001 in PR https://github.com/evcc-io/evcc/pull/10516 richtig?

Wenn ich bei mir `evcc meter` abfrage, bekommen ich als Antwort 38000 kWh, was auch der Realität entspricht.
Der Wert kommt auch über Influx so in Grafana an.

premultiply, 2023-11-01:
> In InfluxDB sehe ich den Wert 124.9k

Stimmt doch. 124.9k kWh sind ~124900 kWh.

closed 2023-11-01 by premultiply

kai-morich, 2023-11-01:
https://docs.evcc.io/docs/reference/configuration/messaging/ ist verwirrend:

2 * kW

        ${pvPower:%.1fk} | Aktuell gemessene PV Leistung in kW

        msg: ${vehicleTitle} connected on wallbox ${title} at ${pvPower:%.1fk}kW PV

2 * Watt

        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW

        pvPower - Aktuelle Solaranlagen-Leistung in Watt (float)