evcc-io/evcc #10605: Wrong casing of vehicle name

berndkrannich, 2023-11-01:
### Describe the bug

Minor nitpick:

My car is defined like this:
```yaml
vehicles:
- name: Enyaq iV
...
```

However, in the UI it shows up as:
[image]
(note the uppercase 'i')

It would be great to get a fix for this. Thanks in advance.

### Steps to reproduce

1. Define the car as shown in the bug description.
2. Start evcc and observe the display of the vehicle.

### Configuration details

```yaml
See bug description
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Docker container

### Version

0.121.5

RTTTC, 2023-11-03:
Simple fix - Rename to just `Enyaq`.

naltatis, 2023-11-04:
Please use the `title` field for this. There you can set any charater you like (including Emoji :D). The `name` field should only be used for internal referencing inside the yaml file.

https://docs.evcc.io/en/docs/reference/configuration/vehicles/#title

closed 2023-11-04 by naltatis

andig, 2023-11-04:
Ach, und ich dachte das UI hätte irgendwo title-case eingebaut. Danke @naltatis

berndkrannich, 2023-11-05:
Perfect, thank you very much, @naltatis!