evcc-io/evcc #10608: Ladvorgänge addieren sich an vergangenem Tag (31.10.)

MackKaufmann, 2023-11-02:
### Describe the bug

- Ausgangslage: Hybridfahrzeug wird jede Nacht geladen. 
- Die Ladevorgänge addieren sich seit 0.121.5 alle auf den 31.10.
- Für November sagt die UI: "Noch keine Ladevorgänge in diesem Monat.", obwohl in der Nacht vom 1.11. sowie 2.11. geladen wurde.

### Steps to reproduce

Ladevorgang Hybrid jede Nacht mit variablen Stromtarif

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

interval: 10s # control cycle interval

tariffs:
  grid:
    type: tibber
    token: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX" # access token
    homeid: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX" # optional if multiple homes associated to account

meters:
- type: template
  template: shelly-1pm
  usage: grid
  host: 192.168.XXX.111
  user: admin #
  password: XXXXXXXXXXXXXXXXXXX #
  name: grid1

chargers:
- type: template
  template: shelly
  host: 192.168.XXX.111
  user: admin #
  password: XXXXXXXXXXXXXXXXXXX #
  channel: 0
  standbypower: 15
  name: wallbox3

vehicles:
- type: template
  template: audi
  title: Audi Q3
  user: XXXXXXXXXXXXXXXXXXXXXX
  password: XXXXXXXXXXXXXXXXXXX
  vin: XXXXXXXXXXXXXXXXXXXX
  capacity: 11.5
  name: ev2

loadpoints:
- title: Carport
  charger: wallbox3
  vehicle: ev2
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 10
  resetOnDisconnect: true
  soc:
    poll:
      mode: always
      interval: 30m
    estimate: true

site:
  title: AWW
  meters:
    grid: grid1
  smartCostLimit: 0.30

# push messages
messaging:
  events:
    start:
      title: Ladevorgang gestartet
      msg: Ladevorgang wurde mit Modus "${mode}" gestartet
    stop:
      title: Ladevorgang beendet
      msg: ${chargedEnergy:%.1fk}kWh in ${chargeDuration} geladen
    connect:
      title: Fahrzeug verbunden
      msg: Ladepunkt ${loadpoint}
    disconnect:
      title: Fahrzeug getrennt
      msg: Ladepunkt ${loadpoint} nach ${connectedDuration} getrennt
  services:
    - type: pushover
      app: XXXXXXXXXXXXXXXXXXXXX# app id
      recipients:
        - XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX # list of recipient ids
```

### Log details

```text
...
```

### What type of operating system are you running?

Docker container

### Version

0.121.5

andig, 2023-11-02:
> Es scheint, als ober der am 31.10. begonnene Ladevorgang in der Datenbank nicht "beendet" wird.

Der wird nicht beendet solange das Fahrzeug noch dran hängt.

closed 2023-11-02 by andig

MackKaufmann, 2023-11-02:
Sowohl am 1.11. als auch heute 2.11. hing das Fahrzeug nicht mehr dran. Auch der heute Nacht (2.11.) gestartete Ladevorgang (Fahrzeug wurde gestern Abend angeschlossen) erscheint unter dem Datum 31.10.

andig, 2023-11-02:
Dann braucht es ein Logfile in dem wir das nachvollziehen können, genau wir im Issue angefordert. Danke.

MackKaufmann, 2023-11-02:
Gern! Gibst Du mir kurz einen Hinweis, wie ich im Docker-Container da ran komme? Kann den Service ja nicht stoppen wie auf Linux.

premultiply, 2023-11-02:
Schaltbare Steckdosen sind keine Wallboxen.
Sie können nicht erkennen ob ein Fahrzeug angeschlossen ist oder nicht. Daher werden die Ladevorgänge nie beendet.
Lösung: Richtige Wallbox benutzen.

MackKaufmann, 2023-11-02:
>Daher werden die Ladevorgänge nie beendet. 

Stimmt nicht. Funktionierte den gesamten Oktober. Siehe verlinktes Lade-Logfile.

naltatis, 2023-11-02:
Hat evcc Schreibzugriff auf der Datenbank? Kannst du mal eine Zielzeit setzen und schauen ob die nach einem Neustart des Containers noch da ist?

MackKaufmann, 2023-11-02:
> Kannst du mal eine Zielzeit setzen und schauen ob die nach einem Neustart des Containers noch da ist?

Done. Zielzeit gesetzt, Container neu gestartet. Zielzeit noch da.

MackKaufmann, 2023-11-03:
Kleines Update: Der Ladevorgang heute Nacht wurde jetzt erstmals im Monat November unter Datum 3.11. in der DB gespeichert. Systemseitig gab es keinerlei Änderungen.

Warum sich die Ladevorgänge mal unter einem "alten" Datum jeweils auf addieren und mal unter neuem Datum anlegen, erschließt sich nicht. Das Fahrzeug ist jeden Abend leer, wird dann über Nacht geladen und morgens wieder genutzt.

naltatis, 2023-11-04:
Wird denn dein Getrennt/Verbunden Status korrekt in der evcc UI am Ladepunkt angezeigt?