evcc-io/evcc #10622: Enphase Envoy Battery Daten anzeigen

fano0001, 2023-11-03:
Hallo ich haben ein Enphase System mit einem Envoy metered als Gateway für die Daten. Meine Batterien sind vom Typ Enphase IQ Battery 3t

In EVCC sieht das UI wie folgt aus:
[image]

Zu sehen ist, was meine PV aktuell produziert, meinen Verbrauch aus dem Haus und den Ladestand der Batterien. Jedoch sehe ich nicht was an LEISTUNG in oder aus der Batterie geladen/gezogen wird.

Lokal konnte ich über die API von Enphase die Daten im Json Format abrufen.
Die API die ich hierfür verwendet habe ist http://envoy.ip/ivp/livedata/status

Daraus resultiert das folgende JSON
`{
    "connection": {
        "mqtt_state": "connected",
        "prov_state": "configured",
        "auth_state": "ok",
        "sc_stream": "enabled",
        "sc_debug": "disabled"
    },
    "meters": {
        "last_update": 1698997567,
        "soc": 44,
        "main_relay_state": 1,
        "gen_relay_state": 5,
        "backup_bat_mode": 1,
        "backup_soc": 0,
        "is_split_phase": 0,
        "phase_count": 3,
        "enc_agg_soc": 44,
        "enc_agg_energy": 3080,
        "acb_agg_soc": 0,
        "acb_agg_energy": 0,
        "pv": {
            "agg_p_mw": 851539,
            "agg_s_mva": 871074,
            "agg_p_ph_a_mw": 851539,
            "agg_p_ph_b_mw": 0,
            "agg_p_ph_c_mw": 0,
            "agg_s_ph_a_mva": 856998,
            "agg_s_ph_b_mva": 0,
            "agg_s_ph_c_mva": 14076
        },
        "storage": {
            "agg_p_mw": 1706000,
            "agg_s_mva": 1706000,
            "agg_p_ph_a_mw": 1706000,
            "agg_p_ph_b_mw": 0,
            "agg_p_ph_c_mw": 0,
            "agg_s_ph_a_mva": 1706000,
            "agg_s_ph_b_mva": 0,
            "agg_s_ph_c_mva": 0
        },
        "grid": {
            "agg_p_mw": 85903,
            "agg_s_mva": -374233,
            "agg_p_ph_a_mw": -168239,
            "agg_p_ph_b_mw": 248599,
            "agg_p_ph_c_mw": 5541,
            "agg_s_ph_a_mva": -661350,
            "agg_s_ph_b_mva": 259408,
            "agg_s_ph_c_mva": 27708
        },
        "load": {
            "agg_p_mw": 2643442,
            "agg_s_mva": 2202841,
            "agg_p_ph_a_mw": 2389300,
            "agg_p_ph_b_mw": 248599,
            "agg_p_ph_c_mw": 5541,
            "agg_s_ph_a_mva": 1901648,
            "agg_s_ph_b_mva": 259408,
            "agg_s_ph_c_mva": 41784
        },
        "generator": {
            "agg_p_mw": 0,
            "agg_s_mva": 0,
            "agg_p_ph_a_mw": 0,
            "agg_p_ph_b_mw": 0,
            "agg_p_ph_c_mw": 0,
            "agg_s_ph_a_mva": 0,
            "agg_s_ph_b_mva": 0,
            "agg_s_ph_c_mva": 0
        }
    },
    "tasks": {
        "task_id": -1392663803,
        "timestamp": 1698963183
    },
    "counters": {
        "main_CfgLoad": 8,
        "main_CfgChanged": 8,
        "main_CfgNotFound": 11013,
        "main_taskUpdate": 71,
        "MqttClient_publish": 537409,
        "MqttClient_live_debug": 180,
        "MqttClient_respond": 6050,
        "MqttClient_msgarrvd": 3027,
        "MqttClient_reconnect": 7,
        "MqttClient_create": 499,
        "MqttClient_setCallbacks": 499,
        "MqttClient_connect": 499,
        "MqttClient_connect_err": 238,
        "MqttClient_connect_Err": 238,
        "MqttClient_subscribe": 261,
        "SSL_Keys_Create": 499,
        "sc_hdlDataPub": 546278,
        "sc_SendStreamCtrl": 3112,
        "sc_SendDemandRspCtrl": 1,
        "rest_Status": 19221
    },
    "dry_contacts": {
        " \u0011": {
            "dry_contact_id": " \u0011",
            "dry_contact_type": " \u0011",
            "dry_contact_load_name": "\u0006",
            "dry_contact_status": 3050996
        },
        "": {
            "dry_contact_id": "",
            "dry_contact_type": "",
            "dry_contact_load_name": "",
            "dry_contact_status": 3050996
        },
        "": {
            "dry_contact_id": "",
            "dry_contact_type": "",
            "dry_contact_load_name": "",
            "dry_contact_status": 3050996
        },
        "": {
            "dry_contact_id": "",
            "dry_contact_type": "",
            "dry_contact_load_name": "",
            "dry_contact_status": 3050996
        }
    }
}`

Dies passt zu den Werten die ich von der Enphase Seite auslesen kann:
[image]

- unter meters.pv.agg_p_mw ist der aktuelle Verbrauch angegeben
- unter meters.storage.agg_p_mw ist die aktuelle Entladeleistung der Akkus angegeben
- unter meters.enc_agg_soc ist der aktuelle Akkustand in % angegeben

Wäre es möglich ein Template für EVCC zu erstellen, indem die Daten aus dem Speicher auch verwendet werden könne?

Falls weitere Daten oder Tests nötig sind, kann ich gerne weitere Daten bereitstellen.

Vielen Dank 
LG
Fabian

andig, 2023-11-03:
> Jedoch sehe ich nicht was an LEISTUNG in oder aus der Batterie geladen/gezogen wird.

Im Template ist das drin:
  
    {{- if eq .usage "battery" }}
    power:
      source: http
      uri: http://{{ .host }}/production.json
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .storage[] | .wNow
    soc:
      source: http
      uri: http://{{ .host }}/ivp/ensemble/inventory
      {{- if .token }}
      auth:
        type: bearer
        password: {{ .token }}
      insecure: true
      {{- end }}
      jq: .[].devices | map(.percentFull) | add / length
    {{- end }}

Warum funktioniert das bei Dir nicht?

andig, 2023-11-03:
> Daraus resultiert das folgende JSON

Das scheint übrigens kein gültiges JSON zu sein:

    Error: Duplicate key ''

premultiply, 2023-11-03:
Eine längliche Diskussion dazu gibt es z.B. hier: https://github.com/evcc-io/evcc/issues/9478

Kurz: Die oben genannte API taugt nicht, da die Daten nur begrenzt bzw. manuell aktualisiert werden.
Die von evcc genutzte API unterstützt diesen Batterietyp aber irgendwie nicht richtig - genau wie im Template vermerkt.

Ansprechpartner: Enphase Support.

closed 2023-11-03 by premultiply