evcc-io/evcc #10627: Frontend - OCPP  mit TerraAC - Leistung und Eigenverbrauch obwohl nicht geladen wird

sreichholf, 2023-11-03:
### Describe the bug

Ich hab eine TerraAC (tats. Firmware 1.6.9 aber das war auch vorher schon so).
Wenn das Laden z.B. aufgrund von zu wenig Überschuss angehalten wird bleiben im Frontend die Leistungswerte stehen und werden auch schön weiterhin mit in den Eigenverbrauch einberechnet.

Sieht dann in etwa so aus:
[image]

Ich denke in dem Fall müsste man vmtl. eine Ladeleistung von 0 forcieren wenn der Ladevorgang beendet wird (oder immer dann wenn er nicht läuft?).

Funktional hat es grundsätzlich keinen wirklichen Einfluß.
Das Laden mit "Überschuss" in meinem Fall mit Threshold von 1200/1400w funktioniert.
Sieht aber halt nicht schön aus.

### Steps to reproduce

1. Ladevorgang starten
2. Ladevorgang unterbrechen (zu wenig sonne oder Modus-Change von der zum Stop führt)
3. Leistungswerte bleiben im Frontend bestehen.

### Configuration details

```yaml
network:
  schema: https
  host: ***
  port: 7070

interval: 10s # control cycle interval

sponsortoken: *****

log: info
levels:
  core: info
  lp-1: info

meters:
  - name: pv
    type: custom
    power:
      source: calc
      add:
        - source: mqtt
          broker: ***
          topic: sbfspot/***
          timeout: 600s
          jq: .PACTot
        - source: mqtt
          broker: ***
          topic: sbfspot/***
          timeout: 600s
          jq: .PACTot

chargers:
  - name: TerraAC
    type: ocpp
    connecttimeout: 30m
    timeout: 20m
    getconfiguration: true
    bootnotification: true

site:
  title: Ed bei Kirn 1 # display name for UIr
  residualPower: 0
  meters:
    pvs:
      - pv

loadpoints:
  - title: Garage
    charger: TerraAC
    mode: pv
    resetOnDisconnect: true
    soc:
      poll:
        mode: charging
        interval: 15m
      estimate: true
    phases: 3
    enable:
      delay: 1m 
      threshold: 1200
    disable:
      delay: 3m
      threshold: 1400
    guardDuration: 5m
    minCurrent: 6
    maxCurrent: 16

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.**

  feedin:
    type: fixed
    price: 0.**

mqtt:
  broker: *****
  topic: evcc

messaging:
  events:
    start:
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop:
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect:
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect:
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc:
      title: SoC updated
      msg: Battery charged to ${vehicleSoC:%.0f}%
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

### Log details

```text
--- Wird gerne nahchgeliefert wenn nötig ---
```

### What type of operating system are you running?

Linux

### Version

https://github.com/evcc-io/evcc/commit/d39a20a1

andig, 2023-11-03:
> Wenn das Laden z.B. aufgrund von zu wenig Überschuss angehalten wird bleiben im Frontend die Leistungswerte stehen und werden auch schön weiterhin mit in den Eigenverbrauch einberechnet.

Bitte Logfile dazu.

sreichholf, 2023-11-03:
Nov 03 16:35:23 penny evcc[2843831]: {"status":"ok"}
Nov 03 16:35:29 penny evcc[2843831]: [ocpp  ] TRACE 2023/11/03 16:35:29 received JSON message from TACW************: [2, "5232698", "MeterValues", {"connectorId": 1, "transactionId": 1, "meterValue": [{"timestamp": "2023-11-03T15:35:29.000Z", "sampledValue": [{"value": "223.80", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L1-N", "unit": "V"}, {"value": "6.06", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L1", "unit": "A"}, {"value": "1340", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L1", "unit": "W"}, {"value": "223.50", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L2-N", "unit": "V"}, {"value": "6.07", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L2", "unit": "A"}, {"value": "1342", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L2", "unit": "W"}, {"value": "223.80", "context": "Sample.Periodic", "format": "Raw", "measurand": "Voltage", "phase": "L3-N", "unit": "V"}, {"value": "6.00", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "phase": "L3", "unit": "A"}, {"value": "1328", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "phase": "L3", "unit": "W"}, {"value": "326", "context": "Sample.Periodic", "format": "Raw", "measurand": "Energy.Active.Import.Register", "unit": "Wh"}, {"value": "6", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Offered", "unit": "A"}, {"value": "18.13", "context": "Sample.Periodic", "format": "Raw", "measurand": "Current.Import", "unit": "A"}, {"value": "4010", "context": "Sample.Periodic", "format": "Raw", "measurand": "Power.Active.Import", "unit": "W"}]}]}]
Nov 03 16:35:29 penny evcc[2843831]: [ocpp  ] TRACE 2023/11/03 16:35:29 sent JSON message to TACW************: [3,"5232698",{}]
Nov 03 16:35:33 penny evcc[2843831]: [site  ] DEBUG 2023/11/03 16:35:33 ----
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 charge power: 4010W
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/chargePower: 4010
Nov 03 16:35:33 penny evcc[2843831]: [site  ] DEBUG 2023/11/03 16:35:33 pv power: 0W
Nov 03 16:35:33 penny evcc[2843831]: [site  ] DEBUG 2023/11/03 16:35:33 site power: 4010W
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 smartCostActive: false
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 pvPower: 0
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 pvEnergy: 0
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 pv: [{0 0}]
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 homePower: 0
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/chargeCurrent: 6
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/sessionEnergy: 326
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/sessionSolarPercentage: 0
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/sessionPricePerKWh: 0.25
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/sessionPrice: 0.0815
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/sessionCo2PerKWh: <nil>
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/chargedEnergy: 326
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/chargeDuration: 4m50s
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 charge total import: 0.326kWh
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 charger status: C
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/chargeTotalImport: 0.326
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/connected: true
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/charging: true
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 soc estimated: 36.37% (vehicle: 36.00%)
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 vehicle soc: 36%
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 vehicle soc limit: 80%
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/vehicleSoc: 36.36675
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/vehicleTargetSoc: 80
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/chargeRemainingDuration: 9h40m20s
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 vehicle range: 137km
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/chargeRemainingEnergy: 38785.11111111111
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/vehicleRange: 137
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/enabled: true
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 pv charge current: 0.188A = 6A + -5.81A (4010W @ 3p)
Nov 03 16:35:33 penny evcc[2843831]: [lp-1  ] DEBUG 2023/11/03 16:35:33 site power 4010W >= 1400W disable threshold
Nov 03 16:35:33 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:33 lp-1/mode: pv
[... cut ...]
Nov 03 16:35:54 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:54 tariffPriceHome: 0.25
Nov 03 16:35:54 penny evcc[2843831]: [cache ] TRACE 2023/11/03 16:35:54 tariffPriceLoadpoints: 0.25
Nov 03 16:35:54 penny evcc[2843831]: [site  ] TRACE 2023/11/03 16:35:54 {"instanceId":"17410d9763a657b7cb6e19fb83eb19ce0599ba15ade4314f848de6c4be19de41","chargePower":4010,"greenPower":0,"chargeEnergy":0.05399999999999999,"greenEnergy":0}
Nov 03 16:35:54 penny evcc[2843831]: --
Nov 03 16:35:54 penny evcc[2843831]: {"status":"ok"}

sreichholf, 2023-11-03:
Ist nur ein Ausschnitt aber sollte hoffentlich alles Wichtige drin sein.

andig, 2023-11-03:
Habs mal lesbar formatiert.

andig, 2023-11-04:
Ist nicht zu verwerten. Es fehlt der Teil nach Ladeende.

andig, 2023-11-04:
> Wenn das Laden z.B. aufgrund von zu wenig Überschuss angehalten wird bleiben im Frontend die Leistungswerte stehen und werden auch schön weiterhin mit in den Eigenverbrauch einberechnet.

Bitte auch prüfen, ob das nach 2min immer noch der Fall ist. Leider weiss man bei OCPP nicht so genau, ob da noch eine Nachricht kommen wird...

sreichholf, 2023-11-04:
Ich konnte ocpp seitog mach dem StopTransaction außer Heartbeats nichts mehr finden. Ich werde es aber trotzdem im Laufe des heutigen oder morgigen Tages ein weiteres Mal nachstellen und mehr log zur Verfügung stellen

andig, 2023-11-04:
Wir könnten ja bei Stop einfach mal optimistisch anngehem, dass nichts mehr kommt- PR folgt

closed 2023-11-04 by andig