evcc-io/evcc #10628: Mindestladung mit Anzeige von Restzeit und Reichweite

docolli, 2023-11-03:
Aktuell nutzen wir häufiger die Mindestladung, da die PV Leistung nicht mehr immer ausreicht.
Dabei fällt mir auf, dass evcc leider nicht die Zeit anzeigt, bis die Mindestladung erreicht ist.

[image]

Die angezeigte Restzeit (hier 3:28h) bezieht sich auf das Ladeziel 80%, das ist hier aber nicht realistisch, da davon ausgegangen wird, dass nach der Mindestladung mit gleicher (maximaler) Ladeleistung weiter geladen wird. 

Informativer wäre hier eine Zeitangabe (evtl. rechts neben der Statuszeile "Mindestladung bis 50% - Restzeit 0:23h" ) zum Erreichen der Mindestladung.
Und unten bei der Angabe "Min. Ladung" 50%, könnte man die Reichweite bei Mindestladung mit anzeigen, analog, wie die Reichweite bei Ladestand und Ladeziel mit dabei steht.

andig, 2023-11-03:
Wir könnten- solang minSoc nicht erreicht ist- beide minRange minDuration Werte zusätzlich erreichen und publishen, wie auch im UI anzeigen?

Bei der Restzeit müsste allerdings deutlich werden, auf welches Ladeziel sie sich bezieht.

/cc @naltatis

naltatis, 2023-11-03:
Da wäre ich gegen. Glaube nicht dass wir unbedingt noch mehr Werte und Timer brauchen. Der Restwert kommt ja teilweise auch vom Fahrzeug selbst. 

Mit der Einführung wiederholender Pläne wird sich sowieso noch einiges ändern. Vieles was heute mit MinSoC passiert kann dann ein Plan werden. Evtl kann man in dem Zuge die statusmeldungen (Ladeplan) noch mal überarbeiten.

docolli, 2023-11-03:
> Mit der Einführung wiederholender Pläne wird sich sowieso noch einiges ändern. Vieles was heute mit MinSoC passiert kann dann ein Plan werden. Evtl kann man in dem Zuge die statusmeldungen (Ladeplan) noch mal überarbeiten.

Auch ich möchte das GUI nicht mit Infos überfrachten, schon gar nicht an der Stelle, wo jetzt die Restzeit steht, die Restzeit bis zum Erreichen der MinSoC anzeigen und dann noch eine Info auf welches Ladeziel sie sich bezieht. Das führt zu Verwirrungen.

Dann würde ich meine Anfrage ins Backlog legen und schauen, was davon sinnvoll realisierbar ist, wenn wiederholende Pläne kommen.

Evtentuell lässt sich aber die Anzeige der Reichweite bei Min.Ladung noch kurzfristig in einer der kommenden evcc Versionen einbauen. Jetzt, wenn es kälter wird, geht die Reichweite runter und eigentlich möchte ich eine gewisse Mindestreichweite haben, wenn ich einfach nur das Auto einstecke, ohne dass ich noch evcc bedienen muss. 
Bei der Auswahl des Ladeziels kennt evcc ja schon den Wert für mein MinSoC von 50%. Der Wert (hier 113km) ist intern also schon da.
[image]

andig, 2023-11-04:
Closing as wontfix for time being.

closed 2023-11-04 by andig

naltatis, 2023-11-04:
Den Wunsch mit der Reichweite hab’ ich in einem neuen Issue gepackt. Das ist sehr einfach umsetzbar und stellt symmetrie zu den beiden Auswahlfeldern her.