evcc-io/evcc #10631: Ladeleistung in UI für 1 Phasige Ladepunkte

Abyssion99, 2023-11-03:
### Describe the bug

Ich benutze einen Shelly 3em für meine Wärmepumpe und habe diesen als Loadpoint mit eingebunden. Wenn ich in der GUI auf die Einstellungen des Loadpointes klicke, kann ich die minimale und maximale Stromstärke einstellen. Rechts daneben wird die zugehörige Leistung ausgerechnet. In meinem Fall habe ich den Loadpoint 1-Phasig in der YAML angelegt. Die dort berechnete Leistung benutzt aber 3-Phasen.
[image]

### Steps to reproduce

1. open GUI
2. Click on setting for loadpoint
3. read wrong values

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info

# unique installation id
plant: 4bfac231f125a7beac18c850262576040c2ac93f08e46890bd625da98cedc2d4

interval: 10s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    # dynamic grid price
    type: tibber
    token: "XUtPnxo1hvay6sgHCMjETgkdXsW8AzCB0bzW9W7x5_Q" # [currency]/kWh
    # type: fixed
    # price: 0.39

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

site:
  title: Horn Castle
  meters:
    grid: grid1
    pv: pv2
    battery: battery3
  residualPower: 300

meters:
- name: grid1
  type: template
  template: solax 
  id: 1  
  host: 192.168.10.33
  port: 502
  usage: grid  
  modbus: tcpip
- name: pv2
  type: template
  template: solax 
  id: 1  
  host: 192.168.10.33
  port: 502  
  usage: pv  
  modbus: tcpip 
- name: battery3
  type: template
  template: solax 
  id: 1  
  host: 192.168.10.33
  port: 502  
  usage: battery  
  modbus: tcpip
- name: waermepumpe_leistung
  type: template
  template: shelly-3em
  usage: charge
  host: 192.168.10.64

chargers:
  - name: PulsarPlus
    type: template
    template: pulsarplus
  # - name: aussensteckdose
  #   type: template
  #   template: fritzdect
  #   uri: https://fritz.box # HTTP(S) Adresse (Optional)
  #   user: fritz3235
  #   password: abysbox666
  #   ain: 116570538458 # Die AIN ist auf dem Typenschild auf der Geräterückseite aufgedruckt. Bei führenden Nullen bitte in doppelte Hochkommata setzen.
  #   standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (Optional)
  - name: waermepumpe_schaltkontakt
    type: template
    template: shelly
    host: 192.168.10.64 # IP-Adresse oder Hostname
    # user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) (Optional)
    # password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (Optional)
    channel: 0 # Optional
    standbypower: 100 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (Optional)
    integrateddevice: true
    icon: heater

loadpoints:
  - title: Garage
    charger: PulsarPlus
    resetOnDisconnect: false
    mode: pv
    enable:
      threshold: 0 #residual power beachten, residualpower - threshold
      delay: 10s
    disable:
      threshold: 700 # maximum import power (W)
      delay: 10s
    guardduration: 5m
    phases: 3
    vehicle: zoe
    minCurrent: 8
    maxCurrent: 16
    priority: 5
  # - title: Pumpe
  #   charger: aussensteckdose
  #   phases: 1
  #   minCurrent: 1
  #   mode: off
  - title: Wärmepumpe und Lüftungsanlage
    charger: waermepumpe_schaltkontakt
    meter: waermepumpe_leistung
    phases: 1
    minCurrent: 10
    maxCurrent: 11
    mode: pv
    priority: 1
    enable:
      threshold: 0 #residual power beachten, residualpower - threshold
      delay: 2m
    disable:
      threshold: 700 # maximum import power (W)
      delay: 10s
    guardduration: 15m

vehicles:
  - name: zoe
    type: template
    template: renault
    title: Renault Zoe
    user: abyssion99@posteo.org
    password: fsqS913muHJjqXyg
    capacity: 52 # Akkukapazität in kWh (Optional)
    icon: car
    mode: pv
    mincurrent: 8
    maxcurrent: 16
    minSoc: 5
    targetSoc: 60
    priority: 5

influx:
  url: http://192.168.10.27:8086
  database: evcc_data
  token: HhDPMM9vKcD8w2k0w-npC4aA_IBiQclW1R-i9L7e2Cs8ZzEgwV5KRPFEEKff-inP_NfnYLGbs-tJwvVK-JnleA==
  org: Einhorn_Castle
```

### Log details

```text
Nicht nötig. Da kein spezieller Fehler hier.
```

### What type of operating system are you running?

Windows

### Version

0.121.5

VolkerK62, 2023-11-03:
Was sagt denn `evcc meter`?
Gibt der Shelly auf allen 3 Phasen eine Spannung aus?

Abyssion99, 2023-11-03:
I see the problem. Ja ist 3 Phasig angebunden. Hmmm macht es hier Sinn die YAML anzupassen? Ich möchte über den min Wert für die Stromstärke nur festlegen, wann genügend Überschuss zum schalten zur Verfügung steht. Jetzt Frage ich mich natürlich, welchen Wert der Algorithmus verwendet.

VolkerK62, 2023-11-03:
Wenn die Phasenspannungen gemeldet werden, schaltet evcc intern für die Berechnung automatisch auf 1p oder 3p. Das ist implementiert worden, damit man, je nach Situation, nicht immer die yaml anpassen muss.
Das ist in dem Fall jetzt kontraproduktiv.
Du könntest den Shelly als custom Meter einbinden und nur Power abfragen. Dann wirkt die yaml Einstellung.

andig, 2023-11-04:
> Wenn die Phasenspannungen gemeldet werden, schaltet evcc intern für die Berechnung automatisch auf 1p oder 3p. Das ist implementiert worden, damit man, je nach Situation, nicht immer die yaml anpassen muss.

@premultiply gilt das hier? Ich bin eigentlich schon davon ausgegangen, dass die 1p/3p Einstellung weiter gilt. Das ist natürlich... unglücklich?

andig, 2023-11-04:
@Abyssion99 warum klemmst Du nicht einfach 2 Phasen vom Shelly ab wenn dahinter eh nur 1p angeschlossen ist?

Abyssion99, 2023-11-04:
Die Anlage hängt an allen 3 Phasen. Aber meistens reicht die Leistung einer Phase für den Verdichter oder den Heizstab aus. Ich habe in der YAML auch nur 1 Phase stehen, weil ich mich entsinne irgendwo mal gelesen zu haben, dass einfache schaltbare Leistungsmesser 1 Phasig eingebunden werden müssen. Aber wahrscheinlich sind damit eher schaltbare Steckdosen gemeint. Ich habe kein Problem damit 3 Phasen in die YAML zu schreiben und die Stromstärke anzupassen. Habe schon in der GUI einfach auf 3A min current reduziert. Dann schaltet er in etwa bei 2.1kW Überschuss. Das passt dann. Alles gut. Ich wusste einfach nur nicht, dass der Controller von selbst erkennt, dass am 3em 3 Phasen sind. Ich wollte von Prinzip nur vorgeben, ab welchem Überschuss der Shelly schalten soll. Könnte man lösen, in dem man in Zukunft direkt die Überschussleistung als Parameter angibt, ab der geschalten werden soll. Diesen Grenzwert setzt man für ein Meter. Ob das gerade 1 Phasig oder 3 Phasig arbeitet wäre egal.

closed 2023-11-04 by andig