evcc-io/evcc #10632: Idee: smartCostLimit per Loadpoint definieren

VolkerK62, 2023-11-03:
smartCostLimit kann man nur "global" einstellen.
Idee ist, dass für jeden Loadpoint einstellen zu können. Im Sommer mit viel Sonne ist mir der Gedanke nicht gekommen.
Mein aktueller use case: 
Mein Auto hängt bei dem aktuellen Wetter den ganzen Tag an der Wallbox und soll übers Wochenende nur "saubere Energie" nuckeln, damit ich am Montag wieder genug im Akku habe. 
Mein eBike hängt im Modus PV in der Garage immer am Ladegerät. Nur wenn Überschuss vorhanden ist, soll geladen werden (nach dem Motto: "nice to have")
Leider wird das eBike dann aber auch geladen.

mailkluge, 2023-11-04:
😊 die Idee ist mir auch gestern Abend gekommen, als ich mich mit PV-Überschuss laden im Winter beschäftigt habe. Einen Namen dafür habe ich mir aber nicht überlegt 😉 Ich würde gerne einen hohen Anteil von PV Energie nutzen. Die Einstellung über enable und disable habe ich gelesen und werde ich auch versuchen. Aber durch die Inspiration der dynamischen Tarife und die mögliche Steuerung im PV Modus unter einer definierten Preisgrenze schnell zu laden, kam die Idee auf, dies auch für einen PV-Grid-Mix über die Kosten des Ladevorgangs zu ermöglichen. Idee ist, dass man eine Kostenobergrenze zu definieren für den Solar-Grid-Mix, die bereits heute über die Kosten je kWh des Ladepunkts ausgewiesen werden. Also vergleichbar zu den dynamischen Tarifen wenn man unterhalb einer reinen Netzbezuggrenze laden möchte im PV Modus, jedoch mit der Möglichkeit dynamisch zu steuern bzw. zu regeln.

andig, 2023-11-04:
@naltatis das würde auch Deinen "Heizstab nur wenn Strom günstiger als Pellets" Case adressieren?

VolkerK62, 2023-11-04:
>"Heizstab nur wenn Strom günstiger als Pellets" Case

das würde mich auch betreffen ;)

naltatis, 2023-11-04:
Ja, ich glaube auch, dass es sinnvoll ist, das heutige smartCostLimit an den Ladepunkt zu verschieben. Damit ist es auch implizit pro Ladepunkt ein-/auschaltbar.

Einziger Nachteil zur heutigen Lösung: Hat man viele Ladepunkte und möchte die gleich steuern muss man ggf. öfters klicken. Ist aus meiner Sicht aber eher ein Grenzfall und ließe sich durch ne separate Aktion (bspw Button `für alle übernehmen`) auch einfach nachrüsten, wenn der Bedarf groß ist.

naltatis, 2023-11-04:
Ich mach mir mal Gedanken dazu, wie wir die Ladepunkt-Settings gut aufbauen können. Punkte wie enable/disableThreshold und später ggf. Battery-Boost werden da ja über kurz oder lang auch rein wandern müssen.

ogheinz77, 2023-12-09:
Die Idee finde ich sehr gut!!!
Könnte ich gut gebrauchen (Heizstab Besitzer)

L-Online, 2024-01-27:
Hallo, das wäre für mich auch sinnvoll.
Außerdem: separater Ziel SoC für jedes Fahrzeug (und Heizstab) bei smartem Netzladen.
z.B. billiger Netzstrom bis 70% SoC und wenn die Sonne Scheint bis 90% oder 100%. Dann hat man immer noch platz für PV-Überschuss im Akku. Oder gibt es das etwa schon?
Schöne Grüße
Ludwig

andig, 2024-01-27:
> Oder gibt es das etwa schon?

Nein. Ich verstehe die Idee, aber das wäre dann das 4. Limit das zu berücksichten wäre. Wie sollen wir das dem Anwender klar machen?

L-Online, 2024-01-27:
Ich könnte es mir so ähnlich vorstellen wie bei den SoC in den Batterieeinstellungen. Das ist sehr übersichtlich. 
Man bräuchte dann diese beiden Ziel-SoCs und den für die Sofortladung bei Ankunft.
Wenn kein variabler Tarif eingestellt ist, würde ich den Zugehörigen SoC auch ausblenden. Sonst ist sorgt es nur für Verwirrung.
Man könnte auch den PV SoC nicht starr mit PV-Überschuss verknüpfen sondern mit einem Strompreis kleiner-gleich der Einspeisevergütung.

naltatis, 2024-01-27:
Ich wäre da bei @andig. Lass uns das erstmal simple halten. "Platz für PV Strom lassen" klingt ja auch so, als ob das eigentlich ne Aufgabe für den Planungsalgorithmus ist. Für den Anwendungsfall sollten wir eher auf "wiederholende Ladepläne" #5492 und "PV-Forecast" #9454 als auf mehr manuelle Kontrolle setzen.

Marty56, 2024-02-03:
Wäre es ein Ansatz, wenn man Docker hat, zwei EVCC zweimal zu instanziieren und damit jeweils separate Yaml Files zu haben?

Ich habe gerade auch versucht, mein Auto mit Tibber Strom zu laden und gleichzeitig mein E-Bike mit PV Strom. (nur mit eine Instanz) 
Das scheint nicht zu funktionieren. Die PV Regelung war durch die Laodpoint Definition des Autos wohl nicht mehr möglich. Sprich es scheint nicht möglich zu sein, pro Loadpoint eigene Threshold zu definieren.
Eigentlich komisch, weil ich rein von der Yaml Definition des Loadpoints gedacht hätte, dass das der Fall ist.

naltatis, 2024-02-03:
> Wäre es ein Ansatz, wenn man Docker hat, zwei EVCC zweimal zu instanziieren und damit jeweils separate Yaml Files zu haben?

Klar, das spricht technisch nichts gegen. Ist halt doof, weil dann zwei Energiemanager parallel um "Überschuss" kämpfen. Die Instnazen wissen ja nichts voneinander.

> Ich habe gerade auch versucht, mein Auto mit Tibber Strom zu laden und gleichzeitig mein E-Bike mit PV Strom. (nur mit eine Instanz)

PV Enable/Disable sind pro Loadpoint. Das Smart Cost Limit (Tibber Preisgrenze) ist aktuell aber global und sind wichtiger als PV-Überschuss. Heißt, die Thresholds spielen keine Rolle, wenn du in einer "günstig Phase" bist. Genau das soll ja mit diesem Issue geändert werden.

Marty56, 2024-02-03:
Danke für die Rückmeldung.
Dann werde ich mich über das kommende Update freuen :-)

PR #12135 (2024-02-24): Smart grid charging per loadpoint (BC)
fixes #10632
introduces **battery control: grid charging** 🎉

- 🔌 ~~add "cheap/clean grid charging" section in loadpoint modal (if co2 forecast or dyn. prices are configured)~~ (later)
- 🔋 add "cheap/clean grid charging" section in battery settings modal (if battery control is available and see above)
- 🗑️ remove global "smart grid charging" modal (top navi, price/co2 link)

 **Breaking API Change**

- replaced `POST /api/smartcostlimit/:limit` endpoint with `POST /api/loadpoints/:id/smartcostlimit/:limit` and `POST /api/batterysmartcostlimit/:limit`

**T [... cut]