evcc-io/evcc #10638: Zielladen berechnet Anfangspunkt völlig falsch

christian-moeller, 2023-11-04:
### Describe the bug

Habe gerade beim Zielladen "morgen, 10:00 Uhr Abfahrt" eingestellt. Und evcc will schon heute um kurz nach 10:00 Uhr mit dem Zielladen beginnen (und tut es auch!). Das ist natürlich totaler Quark...

### Steps to reproduce

Wähle einen Zeitpunkt "morgen", der mehr als 24 Stunden in der Zukunft liegt.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local 
  port: 7070

log: info
levels:
  cache: error

interval: 10s 

sponsortoken: ################################

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3389

  feedin:
    type: fixed
    price: 0.1732

  co2:
    type: grünstromindex
    zip: 31547

meters:
- type: template
  template: volkszaehler-importexport
  usage: grid
  host: ##########
  port: ####
  importuuid: ##########
  exportuuid: ##########
  name: grid1

- name: Kostal_Piko
  type: template
  template: kostal-piko-legacy
  usage: pv
  host: ##########
  user: ##########
  password: ##########

chargers:
- type: template
  template: go-e-v3
  host: ##########
  name: go-e-wallbox

loadpoints:
- title: Carport
  charger: go-e-wallbox
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  vehicle: mokkae
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true

vehicles:
- name: mokkae
  type: template
  template: opel
  title: MokkaE
  user: ##########
  password: ##########
  vin: ##########
  capacity: 50

site:
  title: Möller-Loccum
  meters:
    grid: grid1
    pv: Kostal_Piko
```

### Log details

```text
[main  ] INFO 2023/11/04 11:31:03 evcc 0.121.5
[main  ] INFO 2023/11/04 11:31:03 using config file: /home/pi/evcc.yaml
[main  ] INFO 2023/11/04 11:31:03 starting ui and api at :7070
[db    ] INFO 2023/11/04 11:31:04 using sqlite database: /home/pi/.evcc/evcc.db
[lp-1  ] DEBUG 2023/11/04 11:31:05 charge total import: 4081.446kWh
[site  ] INFO 2023/11/04 11:31:05 site config:
[site  ] INFO 2023/11/04 11:31:05   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/11/04 11:31:05     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/04 11:31:05     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/04 11:31:05   vehicles:
[site  ] INFO 2023/11/04 11:31:05     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/11/04 11:31:05 loadpoint 1:
[lp-1  ] INFO 2023/11/04 11:31:05   mode:        pv
[lp-1  ] INFO 2023/11/04 11:31:05   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/11/04 11:31:05   meters:      charge ✓
[lp-1  ] INFO 2023/11/04 11:31:05     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/11/04 11:31:05 phase timer inactive
[lp-1  ] DEBUG 2023/11/04 11:31:05 pv timer inactive
[lp-1  ] DEBUG 2023/11/04 11:31:05 guard timer inactive
[lp-1  ] INFO 2023/11/04 11:31:05 vehicle updated: unknown -> MokkaE
[lp-1  ] DEBUG 2023/11/04 11:31:06 max charge current: 6A
[site  ] WARN 2023/11/04 11:31:06 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/11/04 11:31:06 ----
[lp-1  ] DEBUG 2023/11/04 11:31:06 charge power: 5590W
[site  ] DEBUG 2023/11/04 11:31:06 pv power: 6471W
[site  ] DEBUG 2023/11/04 11:31:06 grid power: -69W
[site  ] DEBUG 2023/11/04 11:31:06 site power: -69W
[lp-1  ] DEBUG 2023/11/04 11:31:07 vehicle odometer: 20384km
[lp-1  ] DEBUG 2023/11/04 11:31:07 charge currents: [7.3 7.4 7.3]A
[lp-1  ] DEBUG 2023/11/04 11:31:07 charge total import: 4081.449kWh
[lp-1  ] DEBUG 2023/11/04 11:31:07 charger status: C
[lp-1  ] INFO 2023/11/04 11:31:07 car connected
[lp-1  ] DEBUG 2023/11/04 11:31:07 pv timer elapse
[lp-1  ] DEBUG 2023/11/04 11:31:07 pv timer inactive
[lp-1  ] DEBUG 2023/11/04 11:31:07 charger: guard elapse
[lp-1  ] DEBUG 2023/11/04 11:31:07 guard timer inactive
[lp-1  ] DEBUG 2023/11/04 11:31:07 charge total import: 4081.449kWh
[lp-1  ] INFO 2023/11/04 11:31:07 start charging ->
[lp-1  ] DEBUG 2023/11/04 11:31:07 wake-up timer: stop
[lp-1  ] DEBUG 2023/11/04 11:31:07 vehicle status: B
[lp-1  ] DEBUG 2023/11/04 11:31:07 vehicle soc: 52%
[lp-1  ] DEBUG 2023/11/04 11:31:07 vehicle range: 134km
[lp-1  ] DEBUG 2023/11/04 11:31:07 available power 5659W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/11/04 11:31:07 start phase scale3p timer
[lp-1  ] DEBUG 2023/11/04 11:31:07 phase scale3p in 1m0s
[lp-1  ] DEBUG 2023/11/04 11:31:07 pv charge current: 6.1A = 6A + 0.1A (-69W @ 3p)
[lp-1  ] DEBUG 2023/11/04 11:31:07 pv timer reset
[lp-1  ] DEBUG 2023/11/04 11:31:07 pv timer inactive
[site  ] DEBUG 2023/11/04 11:31:16 ----
[lp-1  ] DEBUG 2023/11/04 11:31:16 charge power: 4170W
[site  ] DEBUG 2023/11/04 11:31:16 pv power: 6536W
[site  ] DEBUG 2023/11/04 11:31:16 grid power: -1533W
[site  ] DEBUG 2023/11/04 11:31:16 site power: -1533W
[lp-1  ] DEBUG 2023/11/04 11:31:16 vehicle odometer: 20384km
[lp-1  ] DEBUG 2023/11/04 11:31:16 charge currents: [5.8 5.8 5.8]A
[lp-1  ] DEBUG 2023/11/04 11:31:16 detected active phases: 3p
[lp-1  ] DEBUG 2023/11/04 11:31:16 charge total import: 4081.458kWh
[lp-1  ] DEBUG 2023/11/04 11:31:16 charger status: C
[lp-1  ] DEBUG 2023/11/04 11:31:16 soc estimated: 52.02% (vehicle: 52.00%)
[lp-1  ] DEBUG 2023/11/04 11:31:16 vehicle soc: 52%
[lp-1  ] DEBUG 2023/11/04 11:31:16 vehicle range: 134km
[lp-1  ] DEBUG 2023/11/04 11:31:16 available power 5703W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/11/04 11:31:16 phase scale3p in 51s
[lp-1  ] DEBUG 2023/11/04 11:31:16 pv charge current: 8.22A = 6A + 2.22A (-1533W @ 3p)
[lp-1  ] DEBUG 2023/11/04 11:31:16 max charge current: 8A
[site  ] DEBUG 2023/11/04 11:31:26 ----
[lp-1  ] DEBUG 2023/11/04 11:31:26 charge power: 5540W
[site  ] DEBUG 2023/11/04 11:31:26 pv power: 6812W
[site  ] DEBUG 2023/11/04 11:31:26 grid power: -441W
[site  ] DEBUG 2023/11/04 11:31:26 site power: -441W
[lp-1  ] DEBUG 2023/11/04 11:31:26 charge currents: [7.8 7.8 7.8]A
[lp-1  ] DEBUG 2023/11/04 11:31:26 detected active phases: 3p
[lp-1  ] DEBUG 2023/11/04 11:31:26 charge total import: 4081.473kWh
[lp-1  ] DEBUG 2023/11/04 11:31:26 charger status: C
[lp-1  ] DEBUG 2023/11/04 11:31:26 soc estimated: 52.05% (vehicle: 52.00%)
[lp-1  ] DEBUG 2023/11/04 11:31:26 vehicle soc: 52%
[lp-1  ] DEBUG 2023/11/04 11:31:26 vehicle range: 134km
[lp-1  ] DEBUG 2023/11/04 11:31:26 available power 5981W > 4140W min 3p threshold
[lp-1  ] DEBUG 2023/11/04 11:31:26 phase scale3p in 41s
[lp-1  ] DEBUG 2023/11/04 11:31:26 pv charge current: 8.64A = 8A + 0.64A (-441W @ 3p)
^C[lp-1  ] DEBUG 2023/11/04 11:31:31 charge total import: 4081.480kWh
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.5

christian-moeller, 2023-11-04:
[image]

andig, 2023-11-04:
Bitte Logfile in dem ersichtlich ist, was evcc da plant. Nicht irgendein Logfile ;)

andig, 2023-11-04:
@naltatis wir zeigen aktuell im UI den Plan, aber nicht die Planungsgrundlage (z.B. benötigte Energiemenge, Ladegeschwindigkeit etc.). Wäre das eine sinnvolle Erweiterung um den Plan für den Anwender zu plausibilisieren?

[image]

naltatis, 2023-11-04:
> Planungsgrundlage (z.B. benötigte Energiemenge, Ladegeschwindigkeit etc.)

Du meinst da liegt eine Fehlkonfiguration vor? Ja, generell könnten wir das machen: `55:16m @ 11 kW`. Die Daten bekommt die UI heute schon. Die Energiemenge würd ich nicht auch noch extra mit ausweisen. Die ergibt sich aus den beiden Zahlen ja durch Multiplikation.

andig, 2023-11-04:
Die Dauer ist egtl im Chart ersichtlich- dann vllt sogar eher die fehlende Energiemenge anzeigen?

andig, 2023-11-05:
@christian-moeller Siehe auch https://github.com/evcc-io/evcc/issues/10561. Deinen Fall kann ich nicht nachvollziehen. Hier geht es- wenn sich daran nichts ändert- nur noch darum, wie wir zusätzliche Informationen zur Plausibilisierung anzeigen könnten.

PR #11038 (2023-12-07): Planner: show power in preview
fixes #10638

- show power to make estimated time more clear
- added i18n for "PREVIEW"

[image]
closed 2023-12-07 by andig