evcc-io/evcc #10643: Add recurring planned charge

m4ttb1ss, 2023-11-04:
**Is your feature request related to a problem? Please describe.**
On a daily basis I need to set the planned charging for the next working day, once I'm in PV mode. I would love to have an option to create a charge schedule to ensure my car is charged to a pre-defined target SoC in the morning of a business day, in case there wasn't enough PV surplus the day before. 

**Describe the solution you'd like**
Add an option to create a charge plan on a scheduled basis. There are various options:
* allow to select sth. like "every Mo/Tue/Wed..." with ticking a checkbox
* set a pre-defined target SoC per weekday
* offer the option for different target SoCs per weekday

The defined schedule must "survive" a disconnection and be back again when reconnected. I would suggest to configure it vehicle based. 
I can imagine the easiest option being the usage of cronjob notation in the evcc.yaml file. 
I know this is working for example in smarthomeng, which also uses yaml files and offers the possibility of creating schedules for various automated tasks. 
In a second step this for sure shall be configurable via GUI, too. 

**Describe alternatives you've considered**
No idea ;-) 

**Additional context**
I hope the context above is sufficient. 

Long story short: I want to configure on a Sunday evening the target SoC of the upcoming work week, day by day.

StefanSchoof, 2023-11-04:
Sound similar to https://github.com/evcc-io/evcc/issues/5492

m4ttb1ss, 2023-11-04:
> Sound similar to #5492

Yes. Perfekt. Genau das wie in #5492 meine ich. Sorry, hab nicht lang genug gesucht

closed 2023-11-05 by andig