evcc-io/evcc #10645: CO2 charging does not switch to Fast charging mode like Tibber

cschlipf, 2023-11-04:
### Describe the bug

When the Tibber price drops below the configured threshold, EVCC switches the charging mode to 'fast'. This switch is very useful for triggering automations, like disabling the home battery, while the car is charging to prevent it from emptying.

This is not the case for CO2 charging. In case of CO2 charging the charging mode stays in 'pv', however EVCC is essentially performing 'fast' charging.

It would be great, if the behavior would be consistent and also in CO2 charging mode a switch to 'fast' is done as long as the CO2 threshold is met.

### Steps to reproduce

1. Configure CO2 charging
2. Set the charging mode to 'pv'
3. Set the CO2 limit so that EVCC starts charging.
4. Wait for the charging to beginn
5. Charging mode stays in 'pv', although it charges in 'fast' mode while it is under the CO2limit.
...

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.227
  feedin:
    type: fixed
    price: 0.082
  co2:
    type: grünstromindex
    zip: 71126
```

### Log details

```text
[main  ] INFO 2023/11/04 16:10:56 evcc 0.121.5
[main  ] INFO 2023/11/04 16:10:56 using config file: evcc.yaml
[main  ] INFO 2023/11/04 16:10:56 starting ui and api at :7070
[db    ] INFO 2023/11/04 16:10:56 using sqlite database: /home/cschlipf/.evcc/evcc.db
[mqtt  ] INFO 2023/11/04 16:10:56 connecting evcc-1760259450 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/11/04 16:10:56 tcp://localhost:1883 connected
[lp-1  ] DEBUG 2023/11/04 16:10:58 charge total import: 4923.830kWh
[site  ] WARN 2023/11/04 16:10:58 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2023/11/04 16:10:58 site config:
[site  ] INFO 2023/11/04 16:10:58   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/11/04 16:10:58     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/11/04 16:10:58     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/11/04 16:10:58     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/11/04 16:10:58   vehicles:
[site  ] INFO 2023/11/04 16:10:58     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/11/04 16:10:58 loadpoint 1:
[lp-1  ] INFO 2023/11/04 16:10:58   mode:        pv
[lp-1  ] INFO 2023/11/04 16:10:58   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/11/04 16:10:58   meters:      charge ✓
[lp-1  ] INFO 2023/11/04 16:10:58     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/11/04 16:10:58 phase timer inactive
[lp-1  ] DEBUG 2023/11/04 16:10:58 pv timer inactive
[lp-1  ] DEBUG 2023/11/04 16:10:58 guard timer inactive
[site  ] WARN 2023/11/04 16:10:58 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/11/04 16:10:58 ----
[lp-1  ] DEBUG 2023/11/04 16:10:58 charge power: 4313W
[main  ] INFO 2023/11/04 16:10:58 new version available: 0.122.0
[site  ] DEBUG 2023/11/04 16:10:59 pv power: 0W
[site  ] DEBUG 2023/11/04 16:11:00 battery soc: 20%
[site  ] DEBUG 2023/11/04 16:11:00 battery power: 0W
[site  ] DEBUG 2023/11/04 16:11:00 grid power: 5770W
[site  ] DEBUG 2023/11/04 16:11:01 grid powers: [1920 1930 1920]W
[site  ] DEBUG 2023/11/04 16:11:01 grid currents: [8.03 8.04 8.02]A
[site  ] DEBUG 2023/11/04 16:11:02 site power: 5770W
[... 191 lines cut ...]
[site  ] DEBUG 2023/11/04 16:12:19 pv power: 0W
[site  ] DEBUG 2023/11/04 16:12:20 battery soc: 20%
[site  ] DEBUG 2023/11/04 16:12:20 battery power: 0W
[site  ] DEBUG 2023/11/04 16:12:20 grid power: 12240W
[site  ] DEBUG 2023/11/04 16:12:21 grid powers: [4080 4070 4090]W
[site  ] DEBUG 2023/11/04 16:12:21 grid currents: [17.1 17.1 17]A
[site  ] DEBUG 2023/11/04 16:12:22 site power: 12240W
[lp-1  ] DEBUG 2023/11/04 16:12:22 charge currents: [15.1 15 15.1]A
[lp-1  ] DEBUG 2023/11/04 16:12:22 charge total import: 4923.830kWh
[lp-1  ] DEBUG 2023/11/04 16:12:22 charger status: C
[lp-1  ] DEBUG 2023/11/04 16:12:22 soc estimated: 33.00% (vehicle: 33.00%)
[lp-1  ] DEBUG 2023/11/04 16:12:22 vehicle soc: 33%
[lp-1  ] DEBUG 2023/11/04 16:12:22 vehicle soc limit: 85%
[lp-1  ] DEBUG 2023/11/04 16:12:22 vehicle range: 182km
[lp-1  ] DEBUG 2023/11/04 16:12:22 !! scalePhasesIfAvailable: 3p -> 3p
[lp-1  ] DEBUG 2023/11/04 16:12:22 !! scalePhases: GetPhases 3p <> 3p wanted
```

### What type of operating system are you running?

Linux

### Version

 0.121.5

VolkerK62, 2023-11-04:
for detection you can use the MQTT topic `evcc/site/smartCostActive`

cschlipf, 2023-11-04:
> for detection you can use the MQTT topic `evcc/site/smartCostActive`

Great! Thank you very much. Note that this is missing in the documentation:  https://docs.evcc.io/docs/reference/api/#mqtt-api

VolkerK62, 2023-11-04:
most of the Doku topic are writable. There are a lot more, which are only readable. I think it is too much to list them all.

cschlipf, 2023-11-04:
Can you give a hint on how to find out, which topics are available? I am running mosquitto and I haven't found a way there how to list available topics.

StefanSchoof, 2023-11-04:
> It would be great, if the behavior would be consistent and also in CO2 charging mode a switch to 'fast' is done as long as the CO2 threshold is met.

Do you mean switch the active button in the UI?

cschlipf, 2023-11-04:
> > It would be great, if the behavior would be consistent and also in CO2 charging mode a switch to 'fast' is done as long as the CO2 threshold is met.
> 
> Do you mean switch the active button in the UI?

Yes. However with the evcc/site/smartCostActive topic, this is no longer necessary and a much better way. I am closing this therefore. It would be however great to give users some instructions on how to find our, which MQTT topics are available.

closed 2023-11-04 by cschlipf

VolkerK62, 2023-11-04:
I am using "MQTT Explorer" on my Windows Laptop. There you can see every available topic.

cschlipf, 2023-11-04:
> I am using "MQTT Explorer" on my Windows Laptop. There you can see every available topic.

This is great! Thank you very much