evcc-io/evcc #10652: Nach Update auf 0.122.0 kommt "cannot create charger 'Vestel-EVC04"

ogheinz77, 2023-11-04:
### Describe the bug

Nach Update auf 0.122.0 kommt "cannot create charger 'Vestel-EVC04"

[image]

### Steps to reproduce

1. Start EVCC
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 18s

sponsortoken:xxxxxxxxxxxxxx

meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: xxx.xxx.xxx.xxx  
  port: 502  
  name: Fronius-SmartMeter
- type: template
  template: fronius-gen24 
  usage: pv  
  host: xxx.xxx.xxx.xxx  
  port: 502  
  name: FroniusGen24
- type: template
  template: fronius-gen24 
  usage: battery  
  host: xxx.xxx.xxx.xxx  
  port: 502  
  name: BYD HVS 10.2  
  capacity: 10.2 # in kWh

chargers:
- type: template
  template: vestel 
  id: 255  
  host: xxx.xxx.xxx.xxx  
  port: 502  
  modbus: tcpip
  name: Vestel-EVC04
  
  vehicles:
- type: template
  template: mini 
  title: MINI SE  
  user: xxxxxx@.xxx   
  password: xxxxxxxxx   
  vin: xxx.xxx.xxx.xxx   
  cache: 15m
  capacity: 28.9  
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  name: MINI SE
  icon: car
  
 - title: Vestel EVC04
  charger: Vestel-EVC04
  vehicle: MINI SE
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  priority: 5
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true 
  resetOnDisconnect: true

site:
  title: Home
  meters:
    grid: Fronius-SmartMeter
    pv: FroniusGen24
    battery: BYD HVS 10.2
  residualPower: 100
```

### Log details

```text
evcc --log debug
[main  ] INFO 2023/11/04 20:47:06 evcc 0.122.0
[main  ] INFO 2023/11/04 20:47:06 using config file: /etc/evcc.yaml
[main  ] INFO 2023/11/04 20:47:06 starting ui and api at :7070
[db    ] INFO 2023/11/04 20:47:06 using sqlite database: /root/.evcc/evcc.db
[vestel] TRACE 2023/11/04 20:47:06 modbus: send 00 01 00 00 00 06 ff 04 07 d2 00 01
[vestel] TRACE 2023/11/04 20:47:06 modbus: recv 00 01 00 00 00 03 ff 84 02
[main  ] FATAL 2023/11/04 20:47:06 cannot create charger 'Vestel-EVC04': cannot create charger 'template': cannot create charger 'vestel': failsafe timeout: modbus: exception '2' (illegal data address), function '4'
[main  ] FATAL 2023/11/04 20:47:06 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.122.0

ogheinz77, 2023-11-04:
Update:
Configuration details
interval: 30s

andig, 2023-11-04:
Bitte trace log für den charger

ogheinz77, 2023-11-04:
Log kommt gleich

ogheinz77, 2023-11-04:
evcc charger --log trace
[main  ] INFO 2023/11/04 23:20:58 evcc 0.122.0
[main  ] INFO 2023/11/04 23:20:58 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/04 23:20:59 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT * FROM `settings` 7 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_configs_details\" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%       device_id       %") -1 <nil>
[db    ] TRACE 2023/11/04 23:20:59 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[vestel] TRACE 2023/11/04 23:20:59 modbus: send 00 01 00 00 00 06 ff 04 07 d2 00 01
[vestel] TRACE 2023/11/04 23:20:59 modbus: recv 00 01 00 00 00 03 ff 84 02
[main  ] FATAL 2023/11/04 23:20:59 cannot create charger 'Vestel-EVC04': cannot create charger 'template': cannot create charger 'vestel': failsafe timeout: modbus: exception '2' (illegal data address), function '4'

ogheinz77, 2023-11-04:
evcc -l debug charger
[main  ] INFO 2023/11/04 23:25:56 evcc 0.122.0
[main  ] INFO 2023/11/04 23:25:56 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/04 23:25:56 using sqlite database: /root/.evcc/evcc.db
[vestel] TRACE 2023/11/04 23:25:56 modbus: send 00 01 00 00 00 06 ff 04 07 d2 00 01
[vestel] TRACE 2023/11/04 23:25:56 modbus: recv 00 01 00 00 00 03 ff 84 02
[main  ] FATAL 2023/11/04 23:25:56 cannot create charger 'Vestel-EVC04': cannot create charger 'template': cannot create charger 'vestel': failsafe timeout: modbus: exception '2' (illegal data address), function '4'

ogheinz77, 2023-11-04:
@andig 
Bitte wieder öffnen, wenn möglich

m0nn3, 2023-11-05:
Habe den selben Fehler seit diesem Update.
Falls ich irgendwie helfen kann, sagt bescheid.

chrostek, 2023-11-05:
Habe den Fehler auch, bin jetzt zurück auf 0.121.5 - damit geht noch alles.

TMoersch, 2023-11-05:
Ja, kann das Problem ich bestätigen, werde fallback jetzt testen.

Thomas

andig, 2023-11-05:
@TMoersch es ist doch schon an Bug markiert. Warum noch ein "ich auch" Posting ohne zusätzliche Informationen? 

Das Problem wurde mit https://github.com/evcc-io/evcc/pull/10573 eingebaut:

> modbus: send 00 01 00 00 00 06 ff 04 07 d2 00 01

Ursache ist immer die gleiche: wir beheben Fehler, können das mangels Hardware aber nicht testen. Tests von Anwendern im Nightly bleiben aus. Damit landen auch die neuen Fehler dann im Release.

Fixed in https://github.com/evcc-io/evcc/commit/b8fcdbf1e29a83e43d05e14594f27dec174a3356

closed 2023-11-05 by andig

ogheinz77, 2023-11-05:
Moin,
Wenn mir jemand Bescheid gibt kann ich das gerne im Nigthly testen. Ist überhaupt kein Problem 😉

tLupus1978, 2023-11-05:
Scheinbar kann das nächste Nacht im Nightly getestet werden - ich freue mich drauf ;-)

ogheinz77, 2023-11-06:
Hi,

hab evcc Nightly 0.122.0 (f90f4ed4) im Docker installiert.
Vestel WB funktioniert wieder, sieht erstmal gut aus.

Wollte im Docker ein Charger Trace starten, dazu muss man aber EVCC stoppen.
Wie kann ich in einem Container EVCC stoppen?

MonsterinoGIT, 2023-11-07:
Guten Morgen Zusammen,
Bin hier ziemlich neu und habe mir alles zu dem Thema durchgelesen, ich verstehe jedoch nicht wie ich jetzt den Bug bei mir fixen kann. Ja Bug scheint gefixt zu sein und der Nightly ist auch durch, jedoch erhalte ich kein Update oder ähnliches. Muss ich evcc deinstallieren und neu installieren? 
Ich betreibe EVCC auf HomeAssistant der wiederum auf einer VM auf dem QNAP läuft. 
Ich habe hier nicht die Möglichkeit Programmbefehle einzugeben. 

Wäre super nett wenn mir das jemand kurz erklären könnte. 

Vielen Dank schon einmal

andig, 2023-11-07:
Bei HA musst du (leider) aufs Release warten.