evcc-io/evcc #10654: Citroen eC4 - Error "cannot create vehicle" after boot

Bleilot, 2023-11-04:
### Describe the bug

On every boot of my Raspi4 EVCC logs the error "cannot create vehicle". Thus the vehicle is shown to be offline and vehicle SOC is not available for charging algorithm. This happens with all EVCC versions I used until now (v0.122.0). If I just restart the service manually everything works right - see also discussion #9063.

Maybe it's a problem of timeout at Citroen API side? May I configure the timeout?

### Steps to reproduce

1.Reboot raspberry PI 4
2."sudo systemctl restart evcc" to solve the problem

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: [myplant]
sponsortoken: [mytoken]
interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: SB  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: SB  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: SB  
  port: 8080  
  name: battery1
chargers:
- type: template
  template: etrel 
  host: LADEBOX  
  port: 502  
  name: wallbox1

vehicles:
- type: template
  template: citroen 
  title: eC4  
  user: [myemail]  
  password: [mypassword]  
  capacity: 50  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  minSoc: 20  
  targetSoc: 50
#  maxSoc: 90 # Idee für weiteres Laden über das Ziel hinaus mit PV-Überschussstrom ...
  minCurrent: 6  
[... 153 lines cut ...]
Nov 04 20:22:29 RASPI4 evcc[2173]: [lp-1  ] DEBUG 2023/11/04 20:22:29 vehicle status: A
Nov 04 20:22:29 RASPI4 evcc[2173]: [lp-1  ] DEBUG 2023/11/04 20:22:29 vehicle soc: 85%
Nov 04 20:22:29 RASPI4 evcc[2173]: [lp-1  ] DEBUG 2023/11/04 20:22:29 vehicle range: 296km
Nov 04 20:22:38 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:38 ----
Nov 04 20:22:38 RASPI4 evcc[2173]: [lp-1  ] DEBUG 2023/11/04 20:22:38 charge power: 0W
Nov 04 20:22:38 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:38 pv power: 0W
Nov 04 20:22:38 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:38 battery soc: 18%
Nov 04 20:22:38 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:38 battery power: -3W
Nov 04 20:22:38 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:38 grid power: 537W
Nov 04 20:22:38 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:38 battery has priority at soc 18% (< 60%)
Nov 04 20:22:38 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:38 site power: 637W
Nov 04 20:22:38 RASPI4 evcc[2173]: [lp-1  ] DEBUG 2023/11/04 20:22:38 charge currents: [0 0 0]A
Nov 04 20:22:38 RASPI4 evcc[2173]: [lp-1  ] DEBUG 2023/11/04 20:22:38 charger status: A
Nov 04 20:22:38 RASPI4 evcc[2173]: [lp-1  ] DEBUG 2023/11/04 20:22:38 next soc poll remaining time: 4h59m50s
Nov 04 20:22:48 RASPI4 evcc[2173]: [site  ] DEBUG 2023/11/04 20:22:48 ----
...
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.122.0

andig, 2023-11-04:
Bitte trace log fürs Fahrzeug damit man mal sieht was los ist

closed 2023-11-04 by andig

StefanSchoof, 2023-11-04:
Evcc requires the network online target
https://github.com/evcc-io/evcc/blob/4b5d0182884bbee24b8a0f33ba1ed26539853b9c/packaging/init/evcc.service#L6

> network-online.target is a target that actively waits until the nework is "up", where the definition of "up" is defined by the network management software.

https://www.freedesktop.org/wiki/Software/systemd/NetworkTarget/

Maybe your network management software is signal this target before internet is available.

Bleilot, 2023-11-05:
Herzlichen Dank für die prompte Antwort.

 

Ich habe die Datei "evcc.service" in "/etc/systemd/system" mit dem Inhalt aus dem unten genannten Github-Link angelegt. Da war nämlich keine solche Datei. Dann Reboot, aber das Fahrzeug ist wieder offline wie gehabt. Erst nach "reatart evcc" ist es online :-(

 

Für mich (Linux-DAU) waren das zwei Stunden Google-Suche, weil ich einfach nicht kapiert habe, wo ich was tun muss. Vermutlich fehlt noch etwas oder was ich gemacht habe war Unsinn?

 

Sofern möglich bitte nochmals Hilfe!

 

Danke und Gruß

Holger Dickert
 

 
 

Gesendet: Samstag, 04. November 2023 um 22:20 Uhr

 

https://github.com/evcc-io/evcc/blob/4b5d0182884bbee24b8a0f33ba1ed26539853b9c/packaging/init/evcc.service#L6

https://www.freedesktop.org/wiki/Software/systemd/NetworkTarget/

StefanSchoof, 2023-11-05:
@andig Kannst du das bitte in eine Diskussion umwandeln.

Was hast du für ein Netzwerk an deinem raspi? LAN oder WLAN?

Bleilot, 2023-11-05:
Hallo, der Raspi läuft mit WLAN. Habe aufgrund Deiner Frage mal das LAN angeschlossen - gleiches Verhalten.Andig meinte ich soll noch ein anderes Log machen. Das versuche ich grade.Danke schon mal!Gruß Holger Gesendet mit der GMX Mail AppAm 05.11.23 um 19:43 schrieb StefanSchoof
                
            
            
                
                    Von: "StefanSchoof" ***@***.***>Datum: 5. November 2023An: "evcc-io/evcc" ***@***.***>Cc: "Bleilot" ***@***.***>,"Author" ***@***.***>Betreff: Re: [evcc-io/evcc] Citroen eC4 - Error "cannot create vehicle" after boot (Issue #10654)
                
                
@andig Kannst du das bitte in eine Diskussion umwandeln.
Was hast du für ein Netzwerk an deinem raspi? LAN oder WLAN?