evcc-io/evcc #10663: Add Deye 1p template

premultiply, 2023-11-05:
Hallo nochmal,

ich habe jetzt die Verbindung hinbekommen. Jedoch sind die Abfragen wohl nicht erfolgreich, bzw. die Werte falsch. Die Daten für die Meter aus der Doku sind wohl für die dreiphasigen Modell, ich habe jedoch einen einphasigen. Kann es sein, dass die Modbus Register anders sind, und daher die Abfrage nicht klappt, bzw. falsche Werte liefert? Es geht um die Meter PV1 und den Batteriespeicher. Dieser ist aktuell zu 100 Prozent geladen. 

Meine Einstellungen in der evcc.yaml

```
meters:
- type: template
  template: deye-hybrid
  usage: pv
  modbus: tcpip
  id: 1
  host: 192.168.0.154
  port: 502
  name: pv1
- name: my_battery
  type: template
  template: deye-hybrid
  usage: battery
  modbus: tcpip
  id: 1
  host: 192.168.0.154 # Hostname
  port: 502 # Port

```

ergibt folgende Meldung

```
[main  ] INFO 2023/11/04 17:27:02 evcc 0.121.5
[main  ] INFO 2023/11/04 17:27:02 using config file: /etc/evcc.yaml
[db    ] INFO 2023/11/04 17:27:02 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/11/04 17:27:02 connecting evcc-1915430670 at tcp://192.168.0.64:1883
pv1
---
Power:  add[0]: read failed: modbus: exception '2' (illegal data address), function '3'
Energy: 0.0kWh

pv2
---
Power:  1W
Energy: 55.3kWh

grid1
-----
Power:          463W
Energy:         10230.1kWh
Current L1..L3: 0.88A 0.53A 1.85A
Voltage L1..L3: 227V 230V 229V
Power L1..L3:   127W 65.7W 270W

my_battery
----------
Power:  -1W
Energy: 0.0kWh
Soc:    -1%

```

Was muss ich ändern?

_Originally posted by @Tudo2000 in https://github.com/evcc-io/evcc/discussions/10604#discussioncomment-7475191_

premultiply, 2023-11-05:
https://github.com/kellerza/sunsynk/blob/main/src/sunsynk/definitions.py

Tudo2000, 2023-11-06:
Ich habe den Deye Support angeschrieben und um das Protokoll für meinen Wechselrichter Deye SUN-5K-SG03-LP1 EU 48V EU gebeten. Sie haben mir folgendes Dokument zugeschickt:

[Modbus__-__-______V118(1).pdf](https://github.com/evcc-io/evcc/files/13262950/Modbus__-__-______V118.1.pdf)

Tudo2000, 2023-11-06:
Ich hoffe, dass ich hier richtig poste. Ich habe heute versucht, über den IO-Broker per modbus auf den Wechselrichter zuzugreifen. Die Register aus dem Dokument mit den Faktoren scheinen zu stimmen. Ich habe mir relativ wahllos Register mit den Faktoren herausgesucht und diese im IO-Broker eingetragen:

[image]

Die Ergebnisse passen

[image]

Daher kann ich gerne die erforderlichen Register testen (keine Ahnung was gebraucht wird). Ich würde mich sehr freuen, wenn das klappt und bitte um Nachsicht, da ich nicht so bewandert bin.

Vielleicht kann ja jemand auflisten, welche Werte benötigt werden und ich sehe mal zu, welche  Register und Faktoren passen.

Tudo2000, 2023-11-07:
Das Schreiben in die Register habe ich heute auch hinbekommen. Dazu musste ich im IO-Broker folgende Einstellungen wählen

[image]

Der IO-Broker hat dann automatisch die Register geändert. Vorweg kommt eine 40 und dann zählt er einen höher. Aus Register 40 wird also z.B. 40041

[image]

Da ich grundsätzlich kaum Ahnung habe und das meiste durch Ausprobieren herausfinde, kann ich dazu natürlich rein garnichts sagen. Ich hoffe es hilft bei der Template Erstellung.

premultiply, 2023-11-07:
> Der IO-Broker hat dann automatisch die Register geändert. Vorweg kommt eine 40 und dann zählt er einen höher. Aus Register 40 wird also z.B. 40041

Das ist eine absolut nutzlose, sinnlose, überflüssige iobroker-Krankheit. :(

premultiply, 2023-11-07:
[Modbus__-__-______V118.1 2.pdf](https://github.com/evcc-io/evcc/files/13284818/Modbus__-__-______V118.1.2.pdf)

PR #10692 (2023-11-11): Deye: Add single phase inverters
Adds Deye (and some of its brands) string, micro and storage (hybrid) inverter series.

Fixes #10663
closed 2023-11-11 by andig