evcc-io/evcc #10672: Fehlende Phasenumschaltoption

blaBla123BlaBla, 2023-11-06:
Hallo, Seit der letzten Version 0.121.5 ist die Einstellung 1-phasig, 3-phasig und automatische Umstellung nicht mehr verfügbar! Bitte wieder einfügen
MfG

VolkerK62, 2023-11-06:
hast du die "experimentellen features" eingeschaltet?
Im UI oben rechts unter Einstellungen

closed 2023-11-06 by premultiply

blaBla123BlaBla, 2023-11-06:
Danke, hab ich gerade gemacht. Ist jetzt ein bisschen anders. Zwischenzeitlich war bei mir nur die Ampere Einstellung verfügbar. Vielen Dank.Ralf SchröderGesendet mit der 1&1 Mail AppAm 06.11.23 um 10:55 schrieb VolkerK62
                
            
            
                
                    Von: "VolkerK62" ***@***.***>Datum: 6. November 2023An: "evcc-io/evcc" ***@***.***>Cc: "Author" ***@***.***>,"blaBla123BlaBla" ***@***.***>Betreff: Re: [evcc-io/evcc] Fehlende Phasenumschaltoption (Issue #10672)
                
                
hast du die "experimentellen features" eingeschaltet?
Im UI oben rechts unter Einstellungen