evcc-io/evcc #10695: list of charging sessions (Ladevorgänge)

PeterPablo, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
In our evcc installation we do not only track charging ov (B)EVs but also monitor consumption of flats (Wohnungen) and heat pump (Wärmepumpe) using shelly 3PM power meters. This works very well lin principle. During the month change from september to october evcc did not realease new versions for several weeks and thus the instance and the related "charging sessions" continued for many many days.

Initially, we wondered why the october overview of sessions did seemingly miss the consumed power for the first 1-2 weeks. Later, I noted that the data was not lost but merely only depcited in the september timeframe, as this is where the charging sessions started.

This was unexpected behavior to us, but certainly works "as designed" right now.

**Describe the solution you'd like**
Can I propose to engineer some method to split charging sessions (per loadpoint configuration?) on a given interval (e.g. 1d, 7d, month)? This way the above stated scenario would be avoided and furthermore the session information could easily post-processed to analyzed for example daily/weekly consumption of such a special load point (e.g. heat pump).

**Describe alternatives you've considered**
As a user, understand that prolonged charging sessions can be found in the month they started.

**Additional context**
On request, I can supply a screenshot of the described situation.

PeterPablo, 2023-11-08:
[image]

VolkerK62, 2023-11-08:
switchsockets are always connected. Because of that, the session never ends, except by a restart of evcc.
I do the restart every night with a cron job,

PeterPablo, 2023-11-09:
What you write is of course correct and the restart cron job is certainly a workaround. I'm not sure if it could cause unwanted side effects for ongoing car charging sessions. At least my e-Golf does (to my understanding) not support to start another charge session without re-plugging the plug.
I'm posting this feature request to showcase this slightly problematic use case of evcc in combination with permanently connected switchsockets.