evcc-io/evcc #10696: Feature Wunsch: zusätzliches SoC PV Laden nachdem zwangsgeladen wurde.

horst789github, 2023-11-08:
Problem: 
Ich lade z.b. auf 50% durch MIN+PV Modus, wenn das erreicht ist, habe ich das Problem, dass überschüssiger strom (falls vorhanden) nicht ins auto geladen wird, weil ja der SoC schon erreicht ist. 

Mögliche Lösung:
Gute wäre eine zusatz funktion, die bei "Min+PV" bzw "Schnell" eine Option anbietet, dass wenn der SoC erreicht ist und er quasi nicht mehr lädt, man dann automatisch auf PV Modus umschaltet und einen neuen ziel SoC einstellt z.b. 100%. So das er erst das Mindest ziel lädt und danach in PV modus wechselt und den rest versucht auf 100% zuladen.

Beispiel Bild:
[image]

TobiasHuber1980, 2023-11-09:
siehe #9235

andig, 2023-11-09:
Das wird durch https://github.com/evcc-io/evcc/pull/10335 gelöst. Ich bin so frei und mache hier schonmal zu als Duplikat.

closed 2023-11-09 by andig