evcc-io/evcc #10697: Anzeige LEISTUNG, GELADEN, RESTZEIT fehlerhaft

Mikka20, 2023-11-08:
### Describe the bug

Hallo,

die Anzeige der Leistung, Geladen und Restzeit ist fehlerhaft. Siehe Screenshot.

VG
[image]

### Steps to reproduce

1. Auto an Wallbox Heidelberg Energy Control anschließen
2. Laden auf Schnell stellen
3.
...

### Configuration details

```yaml
Nicht relevant
```

### Log details

```text
Nicht relevant
```

### What type of operating system are you running?

Docker container

### Version

0.122.0

premultiply, 2023-11-09:
Log fehlt

closed 2023-11-09 by premultiply

Mikka20, 2023-11-13:
Hallo,

habe nun die Konfiguration und den letzen Ladevorgang als Log angehangen

Mikka20, 2023-11-13:
@premultiply kannst du dieses issue wieder öffnen bitte oder soll ich ein neues erstellen?

premultiply, 2023-11-13:
In deinem Log sind ganz viele offensichtliche Fehler direkt sichtbar.
Das externe Chargemeter (warum?)  liefert keine sinnvollen Daten, es gibt Timeouts mit der Wallbox und das Gridmeter zeigt teilweise riesige Ströme aber dazu unpassende Leistung usw.

Ich denke hier ist erstmal etwas Verschlankung und schrittweises Testen der einzelnen Geräte sinnvoll. Denn in diesem Zustand kann evcc nichts sinnvolles anzeigen.

Mikka20, 2023-11-15:
Danke für deine Rückmeldung. Die Fehler konnte ich nun scheinbar eliminieren, indem ich am Protoss PE11 eine Option geändert habe. Ob diese das Problem generell lösen wird, wird man bei der nächsten Ladung des Autos sehen. Melde mich also erneut.

Der externe Chargemeter wurde damals eingebaut, da die Wallbox keinen besitzt. Mit diesem kann ich in anderen Systemen wie Fhem die Daten weiterverarbeiten. In Fhem wird auch alles korrekt angezeigt, was der Chargemeter auch liefert.

Mikka20, 2023-11-15:
@premultiply: Mein Problem hat sich erledigt, es funktioniert alles so wie es soll.
Habe am Protoss-PE11 unter Serial Port Settings -> Protocol von None auf Modbus umgestellt. 
Meine Vermutung ist, die neuere Firmware 1.40.4 auf dem Protoss. Hatte zuvor v1.40.1 bzw. v1.40.2 wo alles wie jetzt funktioniert hat.

premultiply, 2023-11-16:
Klar, wenn evcc und die der Konverter verschiedene Sprachen sprechen kann es nicht funktionieren.