evcc-io/evcc #10703: Ladeplanung im PV Modus: Zielladen lädt zu früh, wenn Abfahrtszeit nach vorne verschoben wird

All-Ex-U, 2023-11-08:
### Describe the bug

Die Ladeplanung ist aktiv, das Ladeziel beträgt 100% und die Abfahrtszeit ist in einer Stunde. Daher lädt das Auto (wie zu erwarten).
Jetzt wird die Abfahrtszeit auf morgen verschoben. Das Auto lädt weiter, obwohl der Ladeplanung die Ladung erst morgen anzeigt.

### Steps to reproduce

**1. evcc nightly 0.122.0 (cdd9e856) als Demo starten (d.h. ohne evcc.ymal)**
Es ist 21:00 Uhr (ggf. Zeiten unten anpassen)

**2. Folgendes einstellen:**
Smartes Netzladen / CO2-Grenze: keine
Garage: Gastfahrzeug, Modus Schnell (lädt mit 11 kW, das ist wichtig, damit es auf jeden Fall Netzbezug gibt)
Carport: blauer E-Golf, keine Ladeplanung, Modus: PV, Ladeziel: 100%
[image]

**3. Carport: Ladeplanung aktivieren auf morgen 22:00**
[image]

**=> lädt nicht
=> Zielladung startet morgen 08:44**

[image]

**==> alles OK**

**4. Carport: Ladeplanung aktivieren auf heute 22:00**
[image]

**=> lädt** 

[image]

**==> alles OK**

**5. Carport: Ladeplanung aktivieren auf morgen 22:00 (genau wie in Schritt 3)**
[image]

**=> lädt** 

[image]

**==> FEHLER: Das Auto muss sich wie in Schritt 3 verhalten, d.h. darf nicht weiter laden.** 
 

### Configuration details

```yaml
siehe demo
```

### Log details

```text
siehe demo
```

### What type of operating system are you running?

Linux

### Version

nightly 0.122.0 (cdd9e856)

andig, 2023-11-09:
SUPER Fehlerbeschreibung, vielen Dank! Das schaue ich mir in Ruhe an.

GrimmiMeloni, 2023-12-02:
Das scheint "by design" zu sein. Ein aktiver Slot wird nicht abgebrochen. 

https://github.com/evcc-io/evcc/blob/66143c2c614fbf841362dc1d3e0400c35ad3ee97/core/loadpoint_plan.go#L127

Im Log sehe ich auch eine entsprechende Meldung das der Slot noch zu Ende geladen wird.

andig, 2023-12-02:
> Das scheint "by design" zu sein. Ein aktiver Slot wird nicht abgebrochen.

Im Falle einer Planänderung könnte man das natürlich machen. Aber lohnt das? Im Zweifel wird hier 1h verschenkt.

All-Ex-U, 2023-12-02:
Das aktuelle Verhalten ist nicht konsistent und entspricht nicht dem, was der User erwartet. In der Ladeplanung wird die aktuelle Stunde nicht grün sondern grau angezeigt, d.h. laut Ladeplanung wird nicht geladen, tatsächlich lädt das Auto jedoch. Es ist nicht nicht klar, warum evcc mal so und mal anders reagiert. 

Es geht mir nicht um die eine Stunde, sondern darum, dass das aktuelle Verhalten zu dem Eindruck führen kann, dass evcc fehlerhaft arbeitet und man der Software nicht vertraut. M.E. sollte das Verhalten immer konsistent und nachvollziehbar sein. Außerdem sollte die Darstellung in der Ladeplanung dem tatsächlichen Stand entsprechen. 

Daher lohnt sich ein Fix meiner Meinung nach schon: Die Reputation von evcc würde dadurch steigen.

andig, 2023-12-03:
Ich denke wir könnten das akzeptieren. Dürfte nicht zu häufig vorkommen.

@GrimmiMeloni oder fällt Dir ein gutes Kriterium ein, dennoch den Slot abzubrechen? Wie wäre es z.B., wenn der neue Plan seinen ersten Slot überhaupt erst in der Zukunft nach dem gerade laufenden Slot hat? Das tut vllt. nicht weh...

andig, 2023-12-04:
Das war eine der Maßnahmen um flattern beim Planner zu eliminieren, z.b. weil der Estimator mal springt. 

/cc @schenlap was meinst du?

schenlap, 2023-12-04:
> Das war eine der Maßnahmen um flattern beim Planner zu eliminieren, z.b. weil der Estimator mal springt.

Ja, dafür war es notwendig. Ich würde es auch nicht wieder rausnehmen, das macht uns sonst wieder einige Probleme. Was man schon bedenken muss ist dass ein Slot nicht nur 1h sein muss, je nach Tarif kann es auch länger sein (oder werden die dann in Stundenslots geteilt?)

Unschön ist natürlich dass es der Planer nicht anzeigt. Also gibt haben wir zwei Möglichkeiten:
- der Planer zeigt den Slot auch aktiv an. Wobei dann die Frage auftaucht warum er da lädt wenn es teurer ist
- Alles zurücksetzen wenn bei Planänderung. Wäre wohl die sauerberste Lösung. Berücksichtigen muss man vielleicht noch dass die Zeit über API immer wieder mal auf die selbe Zeit gesetzt wird, da sollte es dann nicht greifen, also nur bei tatsächlicher Änderung der Zielzeit oder vom Ziel-SoC.

andig, 2023-12-10:
Depends on https://github.com/evcc-io/evcc/issues/11044

andig, 2024-02-03:
Gem. https://github.com/evcc-io/evcc/issues/10703#issuecomment-1837192161 aktuell keine Priorität. Workaround: Ladeplan deaktivieren und erst dann neuen Plan definieren.

closed 2024-02-03 by andig