evcc-io/evcc #10704: MinSoc PV laden stoppt nicht nach erreichen des Soc und fehlender PV Leistung

panzerdev, 2023-11-09:
Wenn ein minSoc gewählt ist im PV Modus wird nach erreichen des gewünschten nahtlos im PV Modus weitergeladen. Dann kommen die normalen disable timer zum einsatz und es dauert im maximalen Fall 2x disable Timer bis das Laden aufhört. 

Vorschlag wäre das Laden bei erreichen des minSoc entweder direkt zu beenden oder den normalen Prüfprozess fürs das starten des Ladens zu durchlaufen und anhand dessen zu entscheiden ob weiter geladen werden soll.

GrimmiMeloni, 2023-11-09:
Technisch verstehe ich das Problem. Aber machen 2 Minuten mehr Netzbezug jetzt wirklich einen Unterschied, wenn davor sowieso schon länger aus dem Netz geladen wurde?

panzerdev, 2023-11-09:
@GrimmiMeloni In meinem Fall sind es 20 minuten (2x 10 minuten pro Phase) und dazu kommt, dass der Akku ab erreichten minSoc wieder aktiviert wird und dann unnötig entleert wird. Die Frage ist halt wieso er in diesem Spezialfall (minSoc Laden) einfach direkt in den PV Modus übergeht als ob die enable Rahmenbedingungen gepasst hättten?

GrimmiMeloni, 2023-11-09:
> Die Frage ist halt wieso er in diesem Spezialfall (minSoc Laden) einfach direkt in den PV Modus übergeht als ob die enable Rahmenbedingungen gepasst hättten?

Das liegt daran, dass der Übergang von den Lade Modi nicht berücksichtigt wird. Das Regel Intervall weiß gar nicht, was noch im letzten Intervall für ein Modus vorlag, sondern entscheidet einfach basierend auf dem jetzigen Stand, was das Beste ist. In deinem Fall ist ein Ladevorgang in Gange und der PV Modus ist aktiv, und die Ladung reicht nicht mehr. Es verhält sich daher genauso, als würde einfach die Sonne untergehen/ Wolken kommen.

andig, 2023-11-12:
Grad angeschaut. Da wir nur auf den Zustand (minsoc erreicht) aber nicht auf einen Übergang (von nicht erreicht auf erreicht) schauen lässt sich das ohne weitere Zustandsvariablen nicht lösen -> das sollen wir nicht.

Ich sehe grade- steht ja schon alles da: https://github.com/evcc-io/evcc/issues/10704#issuecomment-1803742057

closed 2023-11-12 by andig

panzerdev, 2023-11-12:
Ja gut der comment erklärt nur das es nicht Doll implementiert ist und nicht warum das gewünschtes Verhalten sein soll. Muss ich halt wieder von außen drum herum arbeiten. Schade

andig, 2023-11-12:
> Muss ich halt wieder von außen drum herum arbeiten. Schade.

Du bist natürlich herzlichst eingeladen, in evcc eine bessere Lösung zu finden die wir gerade nicht sehen. Die Schwierigkeiten hab ich oben beschrieben. Dafür- als Go Entwickeler- Daumen runter zu geben, erschließt sich mir nicht.

panzerdev, 2023-11-12:
Der Daumen bezieht sich darauf das ein Implementierungsdetail (hier fehlender state) über die grundsätzliche Frage gestellt wird ob das Verhalten das Richtige ist. Wenn das Verhalten aktuell so ist wie gewünscht, absolut fine. Ich habe das Ticket hier nur auf gemacht weil du mich darum gebeten hast.

andig, 2023-11-12:
...und nach rein schauen zeigt sich dass das nicht so einfach ist. Daher wontfix.