evcc-io/evcc #10707: mqtt topic kann nicht mit einem / (slash) beginnen

bawe, 2023-11-09:
### Describe the bug

mqtt ausgabe:

evcc/updated 1699521330
evcc/site/smartCostActive false
evcc/loadpoints/1/chargePower 0

### Steps to reproduce

1. topic mit / (slash) beginnen lassen
2. evcc starten
3. mit prüfen z.b. mosquitto_sub -h docker -v -t '#' 

### Configuration details

```yaml
# mqtt message broker
mqtt:
  broker: 192.168.0.123:1883
  topic: /evcc
```

### Log details

```text
root@evcc ~ # evcc --log debug
[main  ] INFO 2023/11/09 10:23:42 evcc 0.121.5
[main  ] INFO 2023/11/09 10:23:42 using config file: /root/evcc.yaml
[main  ] INFO 2023/11/09 10:23:42 starting ui and api at :80
[db    ] INFO 2023/11/09 10:23:42 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/11/09 10:23:42 connecting evcc-890075594 at tcp://docker.mshome.local:1883
[mqtt  ] DEBUG 2023/11/09 10:23:42 tcp://docker.mshome.local:1883 connected
[lp-1  ] WARN 2023/11/09 10:23:42 poll mode '{always 1m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2023/11/09 10:23:42 locking phase config to 3p for switchable charger
[lp-1  ] DEBUG 2023/11/09 10:23:42 charge total import: 2020.764kWh
[site  ] INFO 2023/11/09 10:23:42 site config:
[site  ] INFO 2023/11/09 10:23:42   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2023/11/09 10:23:42     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/09 10:23:42   vehicles:
[site  ] INFO 2023/11/09 10:23:42     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/11/09 10:23:42 loadpoint 1:
[lp-1  ] INFO 2023/11/09 10:23:42   mode:        now
[lp-1  ] INFO 2023/11/09 10:23:42   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/11/09 10:23:42   meters:      charge ✓
[lp-1  ] INFO 2023/11/09 10:23:42     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/11/09 10:23:42 phase timer inactive
[lp-1  ] DEBUG 2023/11/09 10:23:42 pv timer inactive
[lp-1  ] DEBUG 2023/11/09 10:23:42 guard timer inactive
[lp-1  ] INFO 2023/11/09 10:23:42 vehicle updated: unknown -> BMW i3 (via MQTT)
[lp-1  ] DEBUG 2023/11/09 10:23:42 set min current: 6
[lp-1  ] DEBUG 2023/11/09 10:23:42 set target soc: 80
[lp-1  ] DEBUG 2023/11/09 10:23:42 set max current: 11
[site  ] DEBUG 2023/11/09 10:23:42 ----
[lp-1  ] DEBUG 2023/11/09 10:23:42 charge power: 0W
[main  ] INFO 2023/11/09 10:23:42 new version available: 0.122.0
[site  ] DEBUG 2023/11/09 10:23:42 grid power: 1134W
[site  ] DEBUG 2023/11/09 10:23:42 pv power: 0W
[site  ] DEBUG 2023/11/09 10:23:42 site power: 1134W
[lp-1  ] DEBUG 2023/11/09 10:23:42 vehicle odometer: 39948km
[lp-1  ] DEBUG 2023/11/09 10:23:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/09 10:23:42 charge total import: 2020.764kWh
[lp-1  ] DEBUG 2023/11/09 10:23:42 charger status: A
[lp-1  ] INFO 2023/11/09 10:23:42 car disconnected
[lp-1  ] DEBUG 2023/11/09 10:23:42 set charge mode: now
[lp-1  ] DEBUG 2023/11/09 10:23:42 set min current: 6
[lp-1  ] DEBUG 2023/11/09 10:23:42 set max current: 16
[lp-1  ] DEBUG 2023/11/09 10:23:42 set priority: 0
[lp-1  ] DEBUG 2023/11/09 10:23:42 set min current: 6
[lp-1  ] DEBUG 2023/11/09 10:23:42 set target soc: 80
[lp-1  ] DEBUG 2023/11/09 10:23:42 set max current: 11
[lp-1  ] DEBUG 2023/11/09 10:23:42 vehicle soc: 77%
[lp-1  ] DEBUG 2023/11/09 10:23:42 vehicle range: 156km
[site  ] DEBUG 2023/11/09 10:23:42 ----
[lp-1  ] DEBUG 2023/11/09 10:23:42 charge power: 0W
[site  ] DEBUG 2023/11/09 10:23:42 grid power: 1134W
[site  ] DEBUG 2023/11/09 10:23:42 pv power: 0W
[site  ] DEBUG 2023/11/09 10:23:42 site power: 1134W
[lp-1  ] DEBUG 2023/11/09 10:23:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/09 10:23:42 charge total import: 2020.764kWh
[lp-1  ] DEBUG 2023/11/09 10:23:42 charger status: A
[lp-1  ] DEBUG 2023/11/09 10:23:42 next soc poll remaining time: 59s
[site  ] DEBUG 2023/11/09 10:24:12 ----
[lp-1  ] DEBUG 2023/11/09 10:24:12 charge power: 0W
[site  ] DEBUG 2023/11/09 10:24:12 grid power: 1140W
[site  ] DEBUG 2023/11/09 10:24:12 pv power: 0W
[site  ] DEBUG 2023/11/09 10:24:12 site power: 1140W
[lp-1  ] DEBUG 2023/11/09 10:24:12 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/09 10:24:12 charge total import: 2020.764kWh
[lp-1  ] DEBUG 2023/11/09 10:24:12 charger status: A
[lp-1  ] DEBUG 2023/11/09 10:24:12 next soc poll remaining time: 29s
[site  ] DEBUG 2023/11/09 10:24:42 ----
[lp-1  ] DEBUG 2023/11/09 10:24:42 charge power: 0W
[site  ] DEBUG 2023/11/09 10:24:42 grid power: 1134W
[site  ] DEBUG 2023/11/09 10:24:42 pv power: 0W
[site  ] DEBUG 2023/11/09 10:24:42 site power: 1134W
[lp-1  ] DEBUG 2023/11/09 10:24:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/09 10:24:42 charge total import: 2020.764kWh
[lp-1  ] DEBUG 2023/11/09 10:24:42 charger status: A
[lp-1  ] DEBUG 2023/11/09 10:24:42 soc estimated: 77.00% (vehicle: 77.00%)
[lp-1  ] DEBUG 2023/11/09 10:24:42 vehicle soc: 77%
[lp-1  ] DEBUG 2023/11/09 10:24:42 vehicle range: 156km
[site  ] DEBUG 2023/11/09 10:25:12 ----
[lp-1  ] DEBUG 2023/11/09 10:25:12 charge power: 0W
[site  ] DEBUG 2023/11/09 10:25:12 grid power: 1134W
[site  ] DEBUG 2023/11/09 10:25:12 pv power: 0W
[site  ] DEBUG 2023/11/09 10:25:12 site power: 1134W
[lp-1  ] DEBUG 2023/11/09 10:25:12 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/11/09 10:25:12 charge total import: 2020.764kWh
[lp-1  ] DEBUG 2023/11/09 10:25:12 charger status: A
[lp-1  ] DEBUG 2023/11/09 10:25:12 next soc poll remaining time: 29s
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.121.5

goebelmeier, 2023-11-09:
Das entspricht dem Standard. MQTT Base Topics fangen nicht mit einem Slash an. Demnach Works as designed in meinen Augen.

GrimmiMeloni, 2023-11-09:
Und was ist denn jetzt funktional das Problem? Welcher Usecase funktioniert deshalb nicht?

bawe, 2023-11-09:
Es passt nur nicht zu meinem lokalen topic's.
Ich sehen keinen Grund warum führende "/" entfernet werden müssen, denke es sollen folgende entfernt werden.
Habe mittlerweile den [code](https://github.com/evcc-io/evcc/blob/0242dad7e9ccb6d2b918ddf785b48d85ef5f427f/cmd/root.go#L207) dafür gefunden denke ich.

andig, 2023-11-09:
Siehe https://github.com/evcc-io/evcc/issues/10707#issuecomment-1803534372. Wontfix.

closed 2023-11-09 by andig