evcc-io/evcc #10716: Home battery discharge control: SolarMax MAX.STORAGE / MAX.STORAGE Ultimate

thebrainkafka, 2023-11-09:
Is your feature request related to a problem? Please describe.

Add support for SolarMax MAX.STORAGE / MAX.STORAGE Ultimate for https://github.com/evcc-io/evcc/issues/10379
https://github.com/evcc-io/evcc/issues/10553

Sorry Implementierung kann ich nicht unterstützen nur den Wunsch dazu äußern.

_Originally posted by @thebrainkafka in https://github.com/evcc-io/evcc/discussions/10713_

thebrainkafka, 2023-11-09:
@andig wie wird denn 10716 weiterverfolgt wenn auf closed gesetzt?
Hatte deinen post dazu im 10379 gelesen

premultiply, 2023-11-09:
Siehe https://github.com/evcc-io/evcc/discussions/10713

andig, 2023-11-09:
Das geht mir etwas zu schnell, zumal die Ausprägung im custom meter auch noch fehlt?

andig, 2023-11-09:
@GrimmiMeloni hast Du schon eine Idee, wie wir von Modus zu konkretem Modbus Register kommen? Wie wäre es mit sowas wie einem `map` Plugin, das aus einem Wert (dem Modus) x Plugins, eines je Wert, macht? Also fan-out? Willst Du es probieren? Alternativ `go` Plugin- das wäre mal eine schöne Anwendung, zumal das Lesen des Modus auch ein wenig Logik braucht für die sich `map` wohl nicht gut eignen würde.

thebrainkafka, 2023-11-12:
@GrimmiMeloni.
Ich würde mich freuen wenn du hier mal näher reinschauen könntest wenn es deine Zeit zulässt.

Danke

GrimmiMeloni, 2023-11-12:
Für Hardware die ich selbst nicht besitze, werde ich keine Implementierung liefern. Wenn es jetzt nur um spezifische Register ginge wäre das vielleicht noch denkbar, aber Ich habe keinerlei Modbus Hardware im Haus - daher ist das für mich nicht realistisch hier an der spezifischen Implementierung mitzuwirken.

Mit Andig bin ich so erstmal so verblieben, das wir darauf schauen eine erste Modbus basierte Implementierung zu haben, und dann können wir von dort ausgehen schauen welche generischen Bausteine ggf. helfen. Da kann ich dann auch gerne mit anpacken.

thebrainkafka, 2023-11-12:
Alles klar. Danke dir/euch!

Gifford47, 2023-11-13:
> Für Hardware die ich selbst nicht besitze, werde ich keine Implementierung liefern. Wenn es jetzt nur um spezifische Register ginge wäre das vielleicht noch denkbar, aber Ich habe keinerlei Modbus [...]

@GrimmiMeloni @thebrainkafka 
Für solaredge habe ich eine entsprechende diskussion #8498 eröffnet. 
Die Adresse für den Storage Control Mode (Laden/ Entladen) ist E004. Size, type, Range etc. findet man [hier](https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/) auf Seite 18.
Ich habe bei mir schon eine entsprechende manuelle Automatisation vorgenommen. Für HA gibt es auch schon entsprechende Addons die das ebenfalls anbieten (links sind in #8498).

Falls noch Hilfe gewünscht ist, könnte ich da unterstützen.

thebrainkafka, 2023-11-13:
@Gifford47 
Vielen Dank. Aus meiner Sicht ist Unterstützung herzlich angekommen.

andig, 2023-11-13:
Wir lösen jetzt bitte erstmal einen einzelnen Case. Dann schauen wir weiter. Dieser Case ist aktuell Victron.

closed 2023-11-13 by andig