evcc-io/evcc #10723: Extend Sofar template

premultiply, 2023-11-10:
Hi all,

In the past I had contributed to the Sofar Solar HYD x KTL support of EVC with the active help of @premultiply . I had faced several challenges connecting the inverter to an RS485 USB Adapter to my Raspberry PI. Not every adapter was reliable and in the end we had to add an delay of 1s between requests to get it working up to a 'stable enough' state.

However this has some drawbacks:
- Reading is slower (of course)
- Values of each read round diverge - especially when conditions are changing rapidly (cloud uncovers sun). Therefore it can take a while in such circumstances until the system balances itself.
- There are still read errors from time to time

With the help from a Sofar support engineer I have a much better solution now: Instead of using the LSW-3 Wifi Logger Stick, get an LSE-3 Ethernet Logger Stick:
[image]

This one is talking ModBus TCP natively over the port 8899. And yes, this port is also open on the LSW-3, but here it does not work. With the LSE-3 it works perfectly fine over ethernet. My meters config:

```
meters:
  - name: sofar-grid
    type: custom
    power:
      source: modbus
      id: 1
      uri: <LoggerIPAddress>:8899
      register:
        address: 0x0488 # Total PCC active power. Selling electricity is positive, buying electricity is negative
        type: holding
        decode: int16
      scale: -10
    energy:
      source: modbus
      id: 1
      uri: <LoggerIPAddress>:8899
      register:
        address: 0x068E # Energy_Purchase_Total
        type: holding
        decode: uint32
      scale: 0.1
    currents:
      - source: modbus
        id: 1
        uri: <LoggerIPAddress>:8899
        register:
          address: 0x0492 # R phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
      - source: modbus
        id: 1
        uri: <LoggerIPAddress>:8899
        register:
          address: 0x049D # S phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
      - source: modbus
        id: 1
        uri: <LoggerIPAddress>:8899
        register:
          address: 0x04A8 # T phase PCC current
          type: holding
          decode: uint16
        scale: 0.01
    powers:
      - source: modbus
        id: 1
        uri: <LoggerIPAddress>:8899
        register:
          address: 0x0493 # ActivePower_PCC_R
          type: holding
          decode: int16
        scale: -10
      - source: modbus
        id: 1
        uri: <LoggerIPAddress>:8899
        register:
          address: 0x049E # ActivePower_PCC_S
          type: holding
          decode: int16
        scale: -10
      - source: modbus
        id: 1
        uri: <LoggerIPAddress>:8899
        register:
          address: 0x04A9 # ActivePower_PCC_T
          type: holding
          decode: int16
        scale: -10

  - name: sofar-pv
    type: custom
    power:
      source: modbus
      id: 1
      uri: <LoggerIPAddress>:8899
      register:
        address: 0x05C4 # Total PV power
        type: holding
        decode: uint16
      scale: 100
    energy:
      source: modbus
      id: 1
      uri: <LoggerIPAddress>:8899
      register:
        address: 0x0686 # PV_Generation_Total
        type: holding
        decode: uint32
      scale: 0.1
      
  - name: sofar-battery-1
    type: custom
    power:
      source: modbus
      id: 1
      uri: <LoggerIPAddress>:8899
      register:
        address: 0x0606 # The charging and discharging power of the first battery pack. Charge is positive, discharge is negative
        type: holding
        decode: int16
      scale: -10
    energy:
      source: modbus
      id: 1
      uri: <LoggerIPAddress>:8899
      register:
        address: 0x069A # Bat_Discharge_Total
        type: holding
        decode: uint32
      scale: 0.1
    soc:
      source: modbus
      id: 1
      uri: <LoggerIPAddress>.13:8899
      register:
        address: 0x0608 # No. 1 battery pack SOC
        type: holding
        decode: uint16
    capacity: 10
```

Set up your logger so that it receives a static IP address and replace `<LoggerIPAddress>` above with its IP address.

Advantages:
- Works absolutely reliable without any delay
- You can access the ModBus from additional devices (e.g. Home Assistant) without the need to configure a ModBus proxy or other tricks
- You do not need an USB RS485 Adapter
- You do not need to open the COM connector, find the correct ports and connect the wires.
- You do not need a physical RS485 connection and may be able to place your EVCC installation everywhere else as long as it's on the same network

@premultiply , @andig  - can you add a recommendation to the documentation? I strongly recommend using the LSE-3 ethernet logger stick over any other solution. Also you might want to create a template without the delay for those using the LSE-3 ethernet logger.

Hope this helps.

_Originally posted by @cschlipf in https://github.com/evcc-io/evcc/discussions/9598_

premultiply, 2023-11-12:
@cschlipf Schau mal: Das ist doch alles schon in diesem Template so drin, oder?
https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/sofarsolar-g3.yaml

Vielleicht hier nur noch ein paar kleine Korrekturen in den Metadaten wie z. B. Port 8899 statt 502 oder Hinweise auf den richtigen Stick?

cschlipf, 2023-11-12:
@premultiply Ja, im Prinzip ist alles drin. Man könnte es aber spezifischer machen: Delay raus (der Hauptgrund für den LSE-3), Modbus immer tcpip, Port fest auf 8899.

Wie ich @andig verstanden habe hat er ein Extra -lse3 template in erster Linie deswegen vorgeschlagen, weil man es so getrennt dokumentieren kann und diesen Ansatz ausdrücklich empfehlen kann.

premultiply, 2023-11-12:
Das Template unterstützt sowohl RS485 als auch Modbus TCP. Die Register sind ohnehin gleich.
Eine Dopplung wäre IMHO nicht so sinnvoll.

Geht dieser Stick nur mit diesem WR oder ist der für alle Solax-WR mit deren jeweiligen Registersätzen?

cschlipf, 2023-11-13:
Fuer alle Sofar (nicht Solax) Wechselrichter, mit diesen Registersaetzen, welche eben die HYD Reihe ist. Sollte genauso auch fuer ZCS Azzuro gelten. Kann ich aber nicht testen.

Man kann natuerlich auch dasselbe Template nehmen. Sollte dann aber eben dokumentieren, dass der LSE-3 stark empfohlen wird und man dann den Port auf 8899 und den Delay auf 0 setzen sollte. Ich verstehe Deinen Punkt - wie gesagt, der Vorschlag eines weiteren Templates kam von @andig um den empfohlenen LSE-3 Adapter besser dokumentieren zu koennen.

PeterPablo, 2023-11-15:
Lässt sich das vorhandene Template überladen? Ich fände es für Neulinge besser, wenn über einen Alias direkt die richtigen Einstellungen für den LSE-3 gesetzt werden.

PeterPablo, 2023-11-20:
In Vorbereitung auf den Test des Pull Request #10847 habe ich die Konfiguration von der von @cschlipf geteilten "manuellen Modbus-TCP/IP-Konfiguration" (s. oben) auf das existierende Template umgestellt (IP sowie Port angepasst, Delay explizit auf 0s gestellt). Das hat prompt sowohl für die Typen "grid", als auch "PV" funktioniert. Danke @premultiply, aus der Dokumentation ging das für mich nicht ausreichend klar hervor. Insofern möchte ich bekräftigen entweder die Dokumentation diesbezüglich ausführlicher zu formulieren oder eben ein abgeleitetes Template zu erzeugen, dass die Anpassungen von Port und Delay überflüssig macht.

cschlipf, 2023-11-20:
Ich habe noch einen Vorschlag. Ich habe mittlerweile sehr viel Feedback, dass der LTE-3 definitiv der bessere Ansatz ist und höre sehr oft von den RS485 Problemen mit Adaptern. Wie wäre es das allgemeine Template zu entfernen und nur noch den LSE-3 zu unterstützen? Zusätzliche Hardware muss eh gekauft werden - dann kann man sich auch gleich den LSE-3 kaufen und der funktioniert eben einfach. Damit promoted EVCC keine schlecht funktionierende Lösung mit Workarounds wie dem Delay, das der Regelung nur schadet (längeres einpendeln der Werte bei schnellen Änderungen).

Wer unbedingt RS485 Adapter nutzen will, kann sich auch manuell eine meters Config erstellen, bzw diese sich leicht aus der generierten Config ableiten.

premultiply, 2023-11-21:
Warum diese Empfehlung nicht einfach ins Template schreiben und alles andere konfigurierbar lassen?

cschlipf, 2023-11-21:
Auch OK. Das war ja mein ursprünglicher Vorschlag, worauf dann von @andig aber der Einwand kam, dass ein separates Template besser wäre.

Wir haben also 3 Vorschläge:

- Existierendes Template behalten und Dokumentation um eine Empfehlung des LSE-3 Logger Sticks erweitern mit den entsprechenden Einstellungen (<IP>, Port: 8899, Delay: 0s)
- Ein LSE-3 Template hinzufügen
- Das existierende Template in ein LSE-3 Template umwandeln und das allgemeine, nicht ganz optimal funktionierende Template nicht mehr unterstützen.

PR #11222 (2023-12-23): Sofar: Modify template for LSE-3 logger stick
Fixes #10723

/cc @cschlipf
closed 2023-12-23 by premultiply