evcc-io/evcc #10736: "Failed to start add-on" in Homeassistant

Deinjo, 2023-11-11:
### Describe the bug

Beim Update auf 0.122.1 erhalte ich beim Start des Add-On in HomeAssistant folgende Fehlermeldung:

Can't create container from addon_49686a9f_evcc: 400 Client Error for http+docker://localhost/v1.43/containers/create?name=addon_49686a9f_evcc: Bad Request ("invalid mount config for type "bind": bind source path does not exist: /mnt/data/supervisor/addon_configs/49686a9f_evcc")

HomeAssitant und alle andere AddOn's etc. sind alle aktualisiert und auf dem neusten Stand

### Steps to reproduce

1. Meldung kommt bei jedem Startversuch des AddOn

### Configuration details

```yaml
Evcc vor dem Update schon länger im Einsatz mit funktionierender evcc.yaml.
Wenn ich die Fehlermeldung richtig verstehe liegt das Problem tendenziell darin, dass die Config nicht gefunden wird.
```

### Log details

```text
Kann in HA leider nicht erzeugt werden.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.122.1

chrispfarr, 2023-11-11:
Hab das gleiche Problem, liegt aber vllt auch am Update von Tesla?

jayjay982, 2023-11-11:
Ist bei mir auch seit dem 0.122.1 so.

alexonbandit, 2023-11-11:
Kann ich bei mir NICHT bestätigen! 
Habe auch evcc 122.1 drauf. Alles funktioniert astrein. Mein Tesla Model 3 lädt wie es sein sollte.
Allerdings habe ich mein HASS noch auf 2023.10.5 und mein Tesla auf 2023.32.9

mruettgen, 2023-11-11:
War bei mir auch so – habe kurzerhand neu installiert und nun kommt die Fehlermeldung...
"Using config file: /config/evcc.yaml
Config not found. Please create a config under /config/evcc.yaml.
For details see evcc documentation at https://github.com/evcc-io/evcc#readme."

Deinjo, 2023-11-11:
> War bei mir auch so – habe kurzerhand neu installiert und nun kommt die Fehlermeldung... "Using config file: /config/evcc.yaml Config not found. Please create a config under /config/evcc.yaml. For details see evcc documentation at https://github.com/evcc-io/evcc#readme."

Aber dann wären doch auch die Log-Daten weg, oder?

mruettgen, 2023-11-11:
> > War bei mir auch so – habe kurzerhand neu installiert und nun kommt die Fehlermeldung... "Using config file: /config/evcc.yaml Config not found. Please create a config under /config/evcc.yaml. For details see evcc documentation at https://github.com/evcc-io/evcc#readme."
> 
> Aber dann wären doch auch die Log-Daten weg, oder? Und es hat nicht zum Erfolg geführt, richtig?

Genau, auch nach dem dritten Versuch gleiches Meldung.
Die Log-Meldung wird ja nach der Installation gezeigt... Nachdem ich "nur" das Update gemacht hatte, kam auch die gleiche Fehlermeldung wie oben beschrieben.

HubsiKU, 2023-11-11:
Tatsächlich habe ich den aktuellen Changelog durchgeführt und einen Ordner über Samba Share erstellt und es wieder zum laufen gebracht. (ohne neuinstallation) leider hats aber die datenbank nicht mehr wiederhergestellt. Wisst ihr wo die bei den älteren Versionen gespeichert werden damit ich sie uns neue Verzeichnis kopieren kann. oder evtl über das backup?

!! BREAKING CHANGES !!

evcc.yaml is now in "/addon_configs/<your addon's slug>" ("/addon_configs/<ID>_evcc")
evcc.db could redirekt to "/addon_configs/<your addon's slug>" ("/addon_configs/<ID>_evcc"). Change /data/evcc.db to /config/evcc.db in addon configuration.
https://developers.home-assistant.io/blog/2023/11/06/public-addon-config/
To access /addon_configs install at least 11.0.0 Samba share (Home Assitant Addon)
evcc will be backed up with the add-on. So, backups of that add-on do include all the necessary files to run it after restore.
https://github.com/evcc-io/evcc/releases

VG

chrispfarr, 2023-11-11:
@HubsiKU irgendwie klappt es bei mir nicht bzw. ich verstehe nicht ganz, wie ich es anstellen soll.
Ich habe unter /addon_configs/ einen Ordner mit xxx_evcc erstellt und dort meine evcc.yaml abgelegt.
In der Konfiguration in HA habe ich den Pfad auf /addon_configs/xxx_evcc gelegt.
Dennoch bekomme ich die folgende Fehlermeldung:

```
Using config file: /addon_configs/49686a8f_evcc.yaml
Config not found. Please create a config under /addon_configs/49686a8f_evcc.yaml.
For details see evcc documentation at https://github.com/evcc-io/evcc#readme.
```

Kannst du mir mal sagen, wie du es gemacht hast?
Vielen lieben Dank