evcc-io/evcc #10737: Home battery control: Victron

Hofyyy, 2023-11-11:
**Is your feature request related to a problem? Please describe.**
It would be cool if the victron battery which is already supported via "victron-energy",
could be activated for batteryDischargeControl.

**Describe the solution you'd like**
Like described in the official victron documentation, their are 3 different ESS modes.
Mode 1 and 2 are interesting for this request.
see: https://www.victronenergy.com/live/ess:ess_mode_2_and_3

My first idea was to use mode 1 which is the default on every ESS installation, because its automatic.
In this mode we need to the the minSOC value to (low, equal, or max) corresponding to the current battery SOC,
to control the charging, discharging indirectly.
Like discussed here (generic): https://github.com/evcc-io/evcc/discussions/10711

But mode 2 is mode 1 + additional features like stop discharging.
Maybe that is what we want at the end, because it does not change "real" configuration.

For me as a user it would be cool to have a nativ implementation, so that I would only need to activate batteryDischargeControl in the site. Because Victron as a battery is already configured.

I can also test or give remote access to my storage.

Further more here are some register and links for minSOC
MQTT (for mode 1): 
W/IDIDIDIDIDIDID/settings/0/Settings/CGwacs/BatteryLife/MinimumSocLimit with { value:NUMBER }
>> Victron itself allows to set this in 5% steps 0,5,10, ... 100. 

**Additional context**
Links:
https://www.victronenergy.com/live/ess:ess_mode_2_and_3
https://github.com/victronenergy/venus/wiki/dbus-api

Hofyyy, 2023-11-11:
A positive point for mode 1 ist, that the user does see the minSOC value and change in the Victron UI. That means it is transparent what evcc is doing. 

Mode 2 register are not visible for the user. and if we have a bug or evcc crashes in a stop situation. i would have no idea how to reset the register by myself

andig, 2023-11-11:
Also was soll jetzt gesetzt werden?

Hofyyy, 2023-11-11:
Es ist die Frage wie es am besten implementiert werden soll. Ok wir können ja einfach starten. Angehängt ist die Registerspec. https://www.victronenergy.com/upload/documents/CCGX-Modbus-TCP-register-list-3.10.xlsx

Ich schlage vor ein min und ein max festzulegen.
min = 20% -> Setzen bei darf entladen
max = 95%  -> Setzen bei soll laden
Bei laden stop soll der aktuelle SOC gesetzt werden, ohne commastelle. Das teste ich gleich einmal per MQTT, wie sich victron bei verschiedenen werten verhält.

Register für minsoc ist:
2901 (excel zeile 369) für Leute die sich auskennen, bitte noch einmal nachlesen und mit der aktuellen Victron implementation abgleichen.
[CCGX-Modbus-TCP-register-list-3.10.xlsx](https://github.com/evcc-io/evcc/files/13326005/CCGX-Modbus-TCP-register-list-3.10.xlsx)

Hofyyy, 2023-11-11:
Ich habe das gerade einmal mit MQTT durchgetestet, da ich mich mit Modbus nicht auskenne. Das sollte intern aber auf das selbe Register gehen.

Schreibregister
W/IDIDIDIDIDID/settings/0/Settings/CGwacs/BatteryLife/MinimumSocLimit

- Set: {„value“: 65.0} -> es wird auch {„value“: 65.0} gespeichert und 65% im Victron UI angezeigt (ohne Kommastelle)
- Set: {„value“: 62.6} -> es wird auch {„value“: 62.6} gespeichert und 65% im Victron UI angezeigt (ohne Kommastelle)
- Set: {„value“: 62.5} -> es wird auch {„value“: 62.5} gespeichert und 60% im Victron UI angezeigt (ohne Kommastelle)
- Set: {„value“: 62.0} -> es wird auch {„value“: 62.0} gespeichert und 60% im Victron UI angezeigt (ohne Kommastelle)

Ich denke also wir können Gefahrlos mit einer Nachkommastelle arbeiten und sollten das auch beim "ladestopp" und der Übernahme des Ist-SoC. Allerdings sollten wir in die Doku schreiben wie Victron rundet, sonst ist man echt verwirrt, weil man nicht den echten Wert sieht.

Anbei noch ein Screenshot vom Victron UI, der nur den SOC ohne Nachkommastelle und in 5% Schritten anzeigt.
[image]

Hofyyy, 2023-11-11:
Es scheint doch schwieriger als gedacht wenn wir es ordentlich haben wollen. Meinetwegen baue ich es mit MQTT vor. Doch noch mal eine Diskussion -> ? oder ok so?

Also:
- Prüfen ob auf dem Victron ein ESS läuft
- Prüfen ob das DCC System aktiv ist
- MinSOC beim Start als Referenz merken und per Log ausgeben (wäre bei mir 20%)
- Maximale Entladegeschwindigkeit (Wechselrichter Leistung) merken und per Log ausgeben (wäre bei mir 800W)

1. Entladen erlauben (normaler zustand)
- MinSOC auf gespicherten Default setzen
- Entladegeschwindigkeit auf Defautl setzen 

2. Laden erzwingen auch bei keiner Sonne (Tibber)
- MinSOC auf 95%
- Entladegeschwindigkeit auf Defautl setzen 

3. Entladen verbieten
- MinSOC auf gespicherten Default setzen
- Entladegeschwindigkeit 0 Watt setzen

andig, 2023-11-11:
Was hat 3 mit minSoc zu tun?

Hofyyy, 2023-11-11:
> Was hat 3 mit minSoc zu tun?

Also es gibt ein DVCC Modul, siehe Screenshoot. Wenn das aktiv ist (was wir prüfen müssten), bei mir ist es aktiv, dann kann man da die Wechselrichterleistung auf 0 drehen.

Wenn das nicht aktiv ist, oder wir es erstmal ganz simpel machen wollen, können wir den Mindest SOC denn das ESS System halten soll, auf den aktuellen Ladestand der Batterie setzen. Damit wäre dann automatisch das Entladen verboten. Also wenn die Batterie 60% geladen ist udn ich ihm sage er soll mindestens 60% geladen lassen dann ...

Passt?

Hofyyy, 2023-11-11:
Der minSOC ist bei mir aktuell auf 60%, da ich es im Winter wegen Notstrom höher lasse, da ich nicht mehr so viel Akku zum pumpen brauche. Jetzt nicht verwirren lassen. Der Wert wäre eigentlich bei genug solar immer bei +- 20%.

Hofyyy, 2023-11-11:
Ok register 2740 funktioniert prächtig. Maximale Entladegeschwindigkeit in Watt. Null heisst garnichts. Dann schaltet das ESS in den INFO Zustand "7" was heisst entladen verboten. Ist also genau das was wir benötigen. Wir müssen nur den original Wert vorher speichern.

andig, 2023-11-12:
Das ist schlecht. Ich würde da einfach MaxInt rein schreiben?

Hofyyy, 2023-11-12:
> Das ist schlecht. Ich würde da einfach MaxInt rein schreiben?

Das tut mir leid, ich verstehe deinen Satz nicht. Was meinst du?

In Register 2740 ist in Watt und -1 heisst keine Limitierung. Also könnten wir zwischen 0 und -1 toggelt (0 aus, -1 erlaubt).
Das Problem ist halt das ein Kunde hier wie ich 800W als default eingetragen haben könnte. Daher müssen wir den Wert beim EVCC Start einmal holen.

andig, 2023-11-12:
In dem Fall kann man die Automatik nicht nutzen und muss eben von Hand steuern. Also in dem Fall kein Maximum sondern -1.

Mal eine Beobachtung: wir haben jetzt 10 Posts gebraucht um dieses simple Register festzulegen. Ich bin noch nicht überzeugt, dass wir diesen Zauber für jeden Akku machen wollen. Zumal die Steuerung ja auch extern wunderbar möglich ist.

Hofyyy, 2023-11-12:
Naja das liegt wohl an mir das ich nicht einfach ein PR hinbekomme. Sorry
Und das die meisten es über minSOC ableiten und es mehrere Möglichekiten gibt das Verhalten herbeizuführen bei Victron. Die Frage ist ja welche ist die Risikoärmste und beste.

ISt es denn schwierig sich beim Start von EVCC einmal per Modbus den Wert für Entladespeed und minSOC zu merken? im Victron Battery template und den auszugeben? Dann scheint mir die Logic jetzt einfach.

andig, 2023-11-12:
Die Templates merken sich nichts und es gibt keine Garantie, dass evcc einen Wert wieder her stellen könnte. Wenns das braucht dann gehts einfach nicht.

Hofyyy, 2023-11-12:
Das Problem ist das man in einer Batteriesteuerung pro Kunde unterschiedliche Settings hat und es nicht in sich autom. Konsistent ist wie bei der Wallbox. Ich denke man wird nicht daran vorbeikommen für die Werte an denen man rumstellt einen Default zu haben. wenn wir uns ihn nicht merken können, dann bräuchten wir denke ich ein Dfault setting. Also ich würde dann in der Victron Battrey Config Entladespeed auf 800W und MinPV auf 20% setzen. Und es eben über extern (EVCC) festlegen ud nicht mehr über Victron selber. Dann wäre man safe.

andig, 2023-11-12:
Gibt es irgendeinen Grund die Entladeleistung zu limitieren? Falls nein: OT und evcc kann für diesen Case halt nicht genutzt werden.

Hofyyy, 2023-11-12:
Stimmt muss man nicht unterstützen. Gründe können elektrischer natur sein. dünne leitung. oder gesetzlich. max 800w als balkon solar. wichtig denke ich nur das der kunde weiss das es auf max gesetzt würde.

andig, 2023-11-12:
Hier mal ein gedankliches Modell wie man `SetBatteryMode()` mit den beiden Inputs:

- misoc
- maxdischargepower

realisieren kann:

```go
switch mode {
case api.BatteryNormal:
	// allow discharge
	err := dischargerPowerS(m.MaxDischargePower)
	if err != nil {
		return err
	}

	// default minsoc
	return minSocS(m.MinSoc)

case api.BatteryLocked:
	// lock discharge
	err := dischargerPowerS(0)
	if err != nil {
		return err
	}

	// default minsoc
	return minSocS(m.MinSoc)

case api.BatteryCharge:
	// lock discharge
	err := dischargerPowerS(m.MaxDischargePower)
	if err != nil {
		return err
	}

	// full charge
	return minSocS(100)

default:
	return api.ErrNotAvailable
}
```

Im Falle Victron müsste die `maxDischargePower` gem. oben mit -1 definiert werden.

/cc @GrimmiMeloni das API ist maximal schwer zu benutzen, aber für ein Experiment können wir ja mal schauen wie weit wir kommen.

andig, 2023-11-12:
@Hofyyy das könntest du mal testen?

GrimmiMeloni, 2023-11-12:
> /cc @GrimmiMeloni das API ist maximal schwer zu benutzen, aber für ein Experiment können wir ja mal schauen wie weit wir kommen.

@andig Ich fand das jetzt ziemlich straight forward. Ein switch block, fertig. Was genau würdest Du denn ändern wollen?

andig, 2023-11-12:
Noch nicht sicher. Das passt fürs Template ja nur weil ich die Option Entladeleistung/MinSoc fest ausimplementiert habe. Der Getter ist auch kritisch. Woran würde man fest machen in welchem Modus sich ein System mit diesen Parametern befinden würde?

GrimmiMeloni, 2023-11-12:
> Noch nicht sicher. Das passt fürs Template ja nur weil ich die Umsetzung mittels Entladeleistung/MinSoc fest ausimplementiert habe. 

Habe ich in #10758 aber auch so gelöst. Scheint mir auch erstmal so das erste "Pattern" zu sein was sich herauskristalisiert?

> Der Getter ist auch kritisch. Woran würde man fest machen in welchem Modus sich ein System mit diesen Parametern befinden würde?

Aktuell habe ich mir das in einem lokalen Member gemerkt. Theoretisch könnte ich den auch loswerden und auf minSoc 100% prüfen, erschien mir aber auch nicht all zu kritisch.

andig, 2023-11-13:
@Hofyyy spräche eigentlich etwas dagegen, die Steuerung analog https://github.com/evcc-io/evcc/discussions/10711 nur mittels `minsoc` zu machen? Das wäre deutlich einfacher umzusetzen ;)

Hofyyy, 2023-11-13:
> @Hofyyy spräche eigentlich etwas dagegen, die Steuerung analog #10711 nur mittels `minsoc` zu machen? Das wäre deutlich einfacher umzusetzen ;)

Ne kann man machen. Ist halt mehr indirekt. Geht aber genau so. Aber bitte bei victron nur 95% als max oder auch einstellbar.

Hofyyy, 2023-11-15:
Test:
- testing docker installier
- Config hinzugefügt: batteryDischargeControl: true
- Restart des dockers

Erste Ergebnisse:
- Ich sehe kein logging, daher weiss ich nicht ob es im testing auch wirklich drin ist.
- Wenn ja wird der Batteriemode initial nicht gesetzt. Meine Voreingestellten 60% (im Victron), werden nicht auf die 20% default gezogen.

closed 2023-11-16 by andig