evcc-io/evcc #10742: vzlogger liefert jq: empty result, wenn uuid des channels mit 0 beginnt

frandess, 2023-11-11:
### Describe the bug

Wenn die uuid des vzlogger channels mit einer 0 beginnt, funktioniert anscheinend das select im jq nicht mehr. Ändert man die uuid, sodass das erste Zeichen nicht mehr 0 ist, verschwindet der Fehler.

Fehlermeldung: js: empty result

Was mir beim Debuggen aufgefallen ist:
Wenn man beim Abruf der Daten bei vzlogger nicht über den index, sondern über die Ressource direkt gehen würde
(z.B. http://raspi.local:8082/a8232ee0-a4ab-11ec-8d36-211f6b082dc8 und dann direkt data[0].tuples[0][1]), bräuchte man das select gar nicht mehr, und der Fehler müsste sich erübrigen.

### Steps to reproduce

1. im vzlogger eine channel-uuid beginnend mit einer '0' konfigurieren

### Configuration details

```yaml
geht:
---------------
- name: hausanschluss
  type: template
  template: vzlogger
  id: 2
  usage: grid
  host: 192.168.179.57
  port: 8082
  uuid: "a8232ee0-a4ab-11ec-8d36-211f6b082dc8"
----------------

geht nicht:
----------------
- name: hausanschluss
  type: template
  template: vzlogger
  id: 2
  usage: grid
  host: 192.168.179.57
  port: 8082
  uuid: "08232ee0-a4ab-11ec-8d36-211f6b082dc8"
----------------
```

### Log details

```text
[main  ] INFO 2023/11/11 13:30:27 evcc 0.122.0
[main  ] INFO 2023/11/11 13:30:27 using config file: /etc/evcc.yaml
[main  ] INFO 2023/11/11 13:30:27 starting ui and api at :7070
[db    ] INFO 2023/11/11 13:30:28 using sqlite database: /home/raspi/.evcc/evcc.db
[site  ] INFO 2023/11/11 13:30:28 site config:
[site  ] INFO 2023/11/11 13:30:28   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/11/11 13:30:28     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/11 13:30:28     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/11/11 13:30:28   vehicles:
[site  ] INFO 2023/11/11 13:30:28     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/11/11 13:30:28 loadpoint 1:
[lp-1  ] INFO 2023/11/11 13:30:28   mode:        pv
[lp-1  ] INFO 2023/11/11 13:30:28   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/11/11 13:30:28   meters:      charge ✓
[lp-1  ] INFO 2023/11/11 13:30:28     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/11/11 13:30:28 phase timer inactive
[lp-1  ] DEBUG 2023/11/11 13:30:28 pv timer inactive
[lp-1  ] DEBUG 2023/11/11 13:30:28 guard timer inactive
[lp-1  ] INFO 2023/11/11 13:30:28 vehicle updated: unknown -> NIXE
[site  ] DEBUG 2023/11/11 13:30:28 ----
[lp-1  ] DEBUG 2023/11/11 13:30:28 charge power: 0W
[site  ] DEBUG 2023/11/11 13:30:28 pv power: 250W
[main  ] INFO 2023/11/11 13:30:28 new version available: 0.122.1
[site  ] ERROR 2023/11/11 13:30:29 grid meter: jq: empty result
[site  ] ERROR 2023/11/11 13:30:29 grid meter: jq: empty result
```

### What type of operating system are you running?

Linux

### Version

0.122.0

andig, 2023-11-12:
Bitte Beispiel JSON.

andig, 2023-11-12:
@frandess ich kann das erstmal nicht nachvollziehen- das braucht eine Debugsession. Könntest Du für info@evcc.io bitte ein Portweiterleitung einrichten? Wir machen hier wieder auf wenn ich Zugang habe um das debuggen zu können. Dank!

frandess, 2023-11-13:
Das ist, was vzlogger liefert, wenn ich über den index gehe. Die letzten beiden channels liefern den gleichen Zählerwert (aus dem Kopf: 16.7.0), nur mit verschiedenen uuids, zum Debuggen:

 ```{
  "version": "0.8.3",
  "generator": "vzlogger",
  "data": [
    {
      "uuid": "a4934be0-a36e-11ec-b1ee-c50b409b7365",
      "last": 1699871408170,
      "interval": -1,
      "protocol": "sml",
      "tuples": [
        [
          1699871408170,
          19926002.400000002
        ]
      ]
    },
    {
      "uuid": "dd4bda60-a36e-11ec-b8a1-85dd8597439e",
      "last": 1699871408170,
      "interval": -1,
      "protocol": "sml",
      "tuples": [
        [
          1699871408170,
          29950198.200000003
        ]
      ]
    },
    {
      "uuid": "a8232ee0-a4ab-11ec-8d36-211f6b082dc8",
      "last": 1699871408170,
      "interval": -1,
      "protocol": "sml",
      "tuples": [
        [
          1699871408170,
          414
        ]
      ]
    },
    {
      "uuid": "08232ee0-a4ab-11ec-8d36-211f6b082dc8",
      "last": 1699871408170,
      "interval": -1,
      "protocol": "sml",
      "tuples": [
        [
          1699871408170,
          414
        ]
      ]
    }
  ]
}

andig, 2023-11-13:
> Könntest Du für [info@evcc.io](mailto:info@evcc.io) bitte ein Portweiterleitung einrichten?

frandess, 2023-11-13:
Vielen Dank, dass du dir das anschauen willst! 

Du meinst, unsere Firewall öffnen, damit du von außen drauf kommst? Könnten wir stattdessen auch ne Debuggingsession auf meinem Rechner machen? Mit Screenshare? 
Oder was meinst du mit Portweiterleitung?

andig, 2023-11-13:
Dass ich über einen Port genau auf den vzlogger komme.

frandess, 2023-11-14:
Ungern. Aber du könntest [das hier](https://github.com/frandess/vzlmock) benutzen, um vzlogger zu mocken. Passt das?

andig, 2023-11-14:
Das ist absolut perfekt- danke!

closed 2023-11-14 by andig

frandess, 2023-11-14:
Wow! Direkt gefixt! Cool!

Super Ding übrigens, das ihr da gebaut habt! Kudos! 👏🏻

andig, 2023-11-14:
Starker Support bei der Fehleranalyse. Wenn du Lust hast: wie suchen immer qualifizierte Unterstützung. Meld Dich gerne bei Slack 👍🏻